MarlinFirmware/Marlin #19018: Pt100 displays temperature of 1 degree on SKR Pro board

yangwenxiong, 2020-08-14:
### Bug Description
Pt100 displays temperature of 1 degree on SKR Pro board , but it appears normal on the SKR 1.3 board. 

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5072395/Marlin.zip)

### Steps to Reproduce

1. #define TEMP_SENSOR_0 21
2. Power is 3.3V, Pt100 connection INA826,
[image]

  3.I also tried the IN826 module's power connection +5V，but the normal temperature still cannot be displayed

**Expected behavior:** [What you expect to happen]
Expect it to show a temperature close to room temperature

**Actual behavior:** [What actually happens]
The temperature measured in PT100 is not correct

#### Additional Information
This is SKR 1.3 board + PT100, showing room temperature,
[image]

ellensp, 2020-08-14:
On the skr 1.3 your using 5v...

yangwenxiong, 2020-08-14:
> On the skr 1.3 your using 5v...
yeah

AnHardt, 2020-08-14:
Looks as is the INA-board is connected to a *normal* T input. If the 4.7K pullup-resistor is not removed that can't give correct readings. Usually the better option is to use a *plain* ADC-input.

yangwenxiong, 2020-08-14:
> Looks as if the INA-board is connected to a _normal_ T input. If the 4.7K pullup-resistor is not removed that can't give correct readings. Usually the better option is to use a _plain_ ADC-input.

Thank you! I tried to remove the pull up resistance of 4.7K, which can be displayed normally on the SKr Pro board, but in the same way, the pull up resistance is removed on the GTR board, and the display is still 1 degree

Portzal, 2020-08-15:
Just a suggestion but a simpler option is to use a high temperature thermistor. https://www.sliceengineering.com/products/thermistor-high-temperature

ellensp, 2020-08-15:
I would like you to swap extruder and bed thermistors over (electrically and in firmware) on the skr pro and see if the load temp reading moves or stays.

Bob-the-Kuhn, 2020-08-16:
Some PT100 amplifiers will run at 3.3V, some won't.

An alternative is to abandon the PT100 amplifier and just attach the PT100 sensor directly to a standard analog input.  You loose dynamic range (lower resolution) but that shouldn't be a problem.

yangwenxiong, 2020-08-17:
> I would like you to swap extruder and bed thermistors over (electrically and in firmware) on the skr pro and see if the low temp reading moves or stays.

Thermistor readings are relatively stable

yangwenxiong, 2020-08-17:
> Some PT100 amplifiers will run at 3.3V, some won't.
> 
> An alternative is to abandon the PT100 amplifier and just attach the PT100 sensor directly to a standard analog input. You loose dynamic range (lower resolution) but that shouldn't be a problem.

Yes, Thank you. I got it

yangwenxiong, 2020-08-31:
The problem has been found and solved

closed 2020-08-31 by yangwenxiong

AnHardt, 2020-08-31:
What has been the reason?
How looked looked your fix?

yangwenxiong, 2020-08-31:
> What has been the reason?
> How looked your fix?
> 
> (Just for updating my in-head database for PT100)

Some expansions with ADC are available on SKR PRO V1.1 or GTR V1.0. Note that the IN826 is stable at 5 volts