MarlinFirmware/Marlin #19021: [BUG] (short description)

bryanhunwardsen, 2020-08-14:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-08-14:
F300 is 300 mm/minute, so only 5mm/sec. That sounds slow, but is still about 12 cubic millimeters a second, which is probably more that your printer can sustain.

Have you actually measured the speed to see a mismatch, or are assuming this is a problem due to the skipped steps?

I suspect there is actually nothing wrong with the extrusion rate, and it is skipping because it can’t melt plastic fast enough. If you are actually seeing it move at 300 mm/s, then you have problems with other settings.

sjasonsmith, 2020-08-14:
It is also worth mentioning that the default 50mm/s as far faster than any normal printer could extrude at. That speed is only useful for loading/unloading filament. You will usually limit your speed to a proper printing rate in the slicer.

I use the volumetric speed limit in PrusaSlicer, and I have to configure my Ender 3 down around 7 cubic mm/s to avoid problems. I get up in the 12+ range on another machine with a volcano hotend.

AnHardt, 2020-08-14:
With a limit of `M203 E 50` in mm/second and a ordered feedrate 'G1 E150 F300' in mm/minute (= 5mm/second) you cant expect the feedrate to be limited. It's only 10% of the, with M203, allowed limit.

Please close.

boelle, 2020-08-14:
no title and no config

and following the recommendation of @AnHardt

closed 2020-08-14 by boelle

bryanhunwardsen, 2020-08-14:
@AnHardt  Thanks for pointing out the unit differences, I was not aware and otherwise overlooked it while reading documentation during my troubleshooting. I was about to reference your comment and reopen the issue request with more verbose bug description/troubleshooting, observations, data and well as full configs when I realized that my Esteps where not off on the SKR board (I had changed them but seems I forgot to save them and power cycled during testing). I corrected them nd retested and everything is working as expected now. (This confirms my true original issue that my Creality E3V2 v4 board went defective after two weeks overreporting temps so it was trying to print 10-20% lower than the reported temp. Is there a reason for the unit diferences between M203 and F, it seems otherwise problematic not not keep them in the same units? I ask because if there is no reason, I might consider doing a feature request to unify them or at least ask for M503 to report the current units rather than the not great "(units/s)" for feedrate. It might save future churn here for similar issue. 

@boelle Apologies sir, I have no clue how this issue request got double posted, the second one without a title(???)

AnHardt, 2020-08-14:
You are the one able to change the title.
About unified units please also see https://github.com/MarlinFirmware/Marlin/pull/18997#issuecomment-674258721 (written with this issue in mind).