MarlinFirmware/Marlin #19023: M81 automatically turns off neopixel LEDs, causes issues with FYSTEC Mini 12864 v2.1 LCDs

LittleNyanCat, 2020-08-14:
I'm running Marlin 2.0.4.4 (I have also tested 2.0.x bugfix branch with the same results), on a SKR 1.4 Turbo with a FYSTEC Mini 12864 v2.1 LCD

### "Bug" Description

I am using PSU_CONTROL to monitor externally the state of my printer (if it is printing or not) and everything was fine until I noticed a quite annoying issue. It seems that when M81 is executed, the firmware automatically disables all LEDs and NeoPixel LEDs connected to the mainboard after a few seconds of encoder inactivity, which kinda makes sense and _usually_ is not an issue. However, this causes a issues with the FYSTEC MINI 12864 v2.1 LCD, as it uses NeoPixel LEDs for its backlight, meaning that if you send the printer a M81 command, after a few seconds it turns off the LCD backlight and you cannot see anything unless you press the encoder wheel, which is undesirable for my case. I need the LCD working as normal, even with the power supply "off" (as far as Marlin knows). I refrain from calling this a bug, as it looks like it could be intended behavior, but with unforeseen consequences for setups like mine. 

### My Configuration
Skr 1.4 Turbo with FYSTEC Mini 12864 v2.1 LCD
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5074559/Marlin.zip)

### Steps to Reproduce

1. Set up Marlin with a FYSTEC Mini 12864 **v2.1** LCD (it's very important that it is the 2.1 version specifically, as other versions do not use NeoPixel LEDs, and thus may not suffer from this issue) 
2. Turn off the power supply from the LCD menu (or just wait if PSU_DEFAULT_OFF is enabled)
3. After a few seconds, the backlight turns off, until you press the encoder, which turns the backlight back on momentarily. 

**Expected behavior:** I could simply turn the power supply on and off with no extra fuss

**Actual behavior:** If I turn the power supply off, it also turns my LCD's backlight off, which is undesirable in my case

#### Additional Information

I am very aware that this is an **edge case** caused by me not using the PSU_CONTROL feature as "intended".

This could be easily addressed with a configuration that prevents M81 from turning of the LEDs, and I'm sure it could be useful for someone else eventually. I have managed a temporary workaround, but it is a little hacky and has some issues itself. 

**If someone could point me to a few lines of code that I can comment out to prevent M81 from disabling the LEDs, I'd greatly appreciate it!**

LittleNyanCat, 2020-08-14:
Just a little more context of what I am actually trying to do and why this is a problem:

I am using a separate Arduino to actually control the printers 12V PSU with a relay **the mainboard IS powered by this psu as well**, while the arduino has it's own, permanent 5v psu. I am using this arduino to control the printer's psu so I can some features of my own, the most relevant here being "shut off after print is finished". The way I'm monitoring if the printer is currently printing is with PSU_CONTROL. The idea is have PSU_DEFAULT_OFF and then to add a M80 (turn psu on) line to my start gcode and M81 (turn psu off) to my end gcode, this way, PS_ON_PIN always reports whether the printer is currently printing or not (and then this information is later used by the arduino to decide when to cut off power to the printer). I need my LCD to be on at all times, even when the psu is "off" on Marlin's perspective, as "off" simple means "not currently printing" to my arduino. This is why turning the LEDs off is an issue.

My current workaround is to use M42 to turn on and off the pin, but this has given me a few issues as well:
When the printer boots, the pin is in a sort of limbo state, where my arduino cannot reliably read it's state, so I have to set up a auto0.g script to turn the pin off right when Marlin boots. This is bad because this is a bit of a hacky implementation, and depends directly on the presence of an SD card at bootup. If there is no SD card present, the printer does not run the startup script, which causes the pin to remain in this limbo state and causes me to be unable to shut off the printer as normal.
For some reason my printer waits for me to "click to resume..." when running any gcode, including auto0.g, which is very annoying.

GMagician, 2020-08-14:
Always try new bugfix before posting bugs on an old version, many thing are always fixed meantime

LittleNyanCat, 2020-08-14:
@GMagician As I stated in the first line, I tried the 2.0.x bugfix branch and I still have the issue.

GMagician, 2020-08-14:
@LittleNyanCat noticed too late...I immediately erased my comment but web is faster. I'm cheking code to see if I can address it

LittleNyanCat, 2020-08-14:
Alright no worries, thank you

GMagician, 2020-08-14:
This is a really intentional behaviour. For a fast fix you may undef `LED_BACKLIGHT_TIMEOUT` in `Conditionals_LCD.h` line 164 (bugfix).

LittleNyanCat, 2020-08-14:
It works! Thank you so much!

It's interesting that they added this "feature" to versions v2.0 and v2.1 especifically, and no other version or LCD. I thought Marlin  automatically disabled led lighting with the M81 but I find it a little perplexing why they added this specifically to these two LCD models. I would be nice to add a little heads up where the Fysetc LCDs are defined to avoid confusing people like me and perhaps the option to disable this without a little fix like this. Anyways, thank you again!

boelle, 2020-08-14:
@LittleNyanCat is solved please click close

closed 2020-08-14 by LittleNyanCat