MarlinFirmware/Marlin #19027: Extruder don’t extrude while print (Extrude normaly manualy)

numidark, 2020-08-14:
Hello,

Following the update from Marlin 2.0 to Marlin 2.0 bugfix (to resolve the G29 bug) my extruder no longer extrudes during printing (or at least in a very strange way), I would like to tell you that I can extrude manually without any problem in the lcd menu.
I noticed that the extruder during printing repeatedly backs and forths, then extrudes very slowly, then resumes front and back.
I have no idea what's going on and admit I'm overwhelmed.

I would quickly put you a video showing the problem.

Thank you in advance for your help

boelle, 2020-08-14:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

numidark, 2020-08-14:
Sorry i miss to join my config ^^’

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5077755/Marlin.zip)

And GIF on my extruder :

[image]

I test quickly the bugfix-2.0.x branch and I give you a comeback.

Thank !

numidark, 2020-08-15:
After few hours of inspect and test, the problem was not the new version but a pin of runout sensor, i have a ramps 1.3 and is not possible to use ps_on and servo3 simultaly (used for runout sensor).
I "forget..." to reaffect pin in pin_ramps.h.
It’s finaly not a bug or a bad configuration.
Sorry for the inconvenience.
I close this issue

closed 2020-08-15 by numidark