MarlinFirmware/Marlin #19028: [BUG] DELAY_US counting wrong on STM32F103ZET6

TheNitek, 2020-08-14:
### Bug Description

I noticed that M672 did not work for my smart effector connected to a trigorilla pro. After further investigation I realized that DELAY_US seems to be counting wrong. The DELAY_US(1000) in M672 pauses for ~1.5ms instead of 1.0ms
Replacing DELAY_US(1000) by delay(1) fixed my problem, but only hides the underlying issue.

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5076938/Marlin.zip)

Anycubic Predator using a Trigorilla Pro and Smart Effector V1.3

### Steps to Reproduce

Send M672 R

**Expected behavior:** Smart Effector LED blinks

**Actual behavior:** Smart Effector LED does not blink

sjasonsmith, 2020-08-17:
If this is an issue I think it would apply to all STM32F103 MCUs. I think they are all the same speed, and use the same code fro delay calculations.

TheNitek, 2020-08-17:
I guess so, yes. I was a little bit confused because in Delay.h there is a case for `__CORTEX_M == 7` which has a comment

> Cortex-M3 through M7 can use the cycle counter of the DWT unit

so I would guess this should work for the STM32 as well.

sjasonsmith, 2020-08-17:
I noticed that mismatch between the code and comment as well. I didn’t explore further to determine whether that could be enabled for more MCUs.

rhapsodyv, 2020-08-17:
@TheNitek  could you try this define this and test again?
```
#define ARCH_PIPELINE_RELOAD_CYCLES 2
```

rhapsodyv, 2020-08-17:
Seems pipeline reload for cortex m3 is >= 2 cycles. So, that extra NOP could not be added, because it will increase the time:
https://electronics.stackexchange.com/questions/16424/cycle-counting-with-modern-cpus-e-g-arm

rhapsodyv, 2020-08-17:
It is better to add to the build_flags
```
build_flags=-DARCH_PIPELINE_RELOAD_CYCLES=2
```

TheNitek, 2020-08-18:
Awesome, thanks for debugging/fixing. I‘ll give it a try tomorrow.

TheNitek, 2020-08-18:
Doesn't seem to work. I set  -DARCH_PIPELINE_RELOAD_CYCLES=2 and according my VS Code the NOP is skipped, but the M672 still does not work. Look like I have to setup my oscilloscope once more

TheNitek, 2020-08-18:
I checked: Still getting ~1.7ms pulses instead of ~1ms

sjasonsmith, 2020-08-18:
@TheNitek there is a whole debate happening inside the Pull Request above about the correct way to fix this. I'm sure it will be resolved soon.

TheNitek, 2020-08-18:
Just saw it after testing. Thanks for all this!

For me it's not a problem right now, I'm just sticking with my delay(1);

boelle, 2020-09-11:
i assume the debate is still going on and that this one kind of depends on what happens?

sjasonsmith, 2020-09-11:
It is still in progress. There are a lot of factors to consider regarding a proper way to fix this, especially since this code is shared with a broad array of ARM MCUs.

sjasonsmith, 2020-11-12:
We still need to come back and finish this.

GadgetAngel, 2020-12-02:
@sjasonsmith Would this not also apply to DELAY_NS which is used heavily in doing temperature reading with the MAX6675 board and other SPI temperature sensor boards. See #20359.

sjasonsmith, 2020-12-03:
Yes it does. Is this causing problems for you? We haven’t prioritized finishing it because the direct impact seems small so far.

sjasonsmith, 2020-12-03:
I got lost reading your new report. There is a lot of information there. Have you de the ruined this is the cause of problems, or is it just an idea right now?

GadgetAngel, 2020-12-03:
I know that my problem with the MAX31865 board on the LPC1768 was due to timing issue.

I am currently having “0” values show up on the MAX6675 board with LPC1768. So there is a timing issue with that also.

I also having issues with the MAX31855 on LPC1768. The MAX31855 in Software SPI mode on LPC1768 does not work.

I do have work-around for ALLL three problems. 

If you want me to submit all the work-arounds so that Marlin can use them until a proper fix can be made, I will submit the work-arounds.  Just let me know what you all decide.  I want to do more testing on all these solutions. 

Do you want me submit a PR that will fix all these issues?

thinkyhead, 2020-12-03:
Wow, good work! I'd love to have a PR and it'll get reviewed quickly.

sjasonsmith, 2020-12-03:
I am surprised this delay issue would be a problem for a SPI device, since SPI is not usually pick about timing.

We have a PR in progress for the DELAY code. We should be getting back to it soon. PRs for your other improvements are of course welcome.

sjasonsmith, 2020-12-03:
I am also not sure that this bug impacts LPC176x based boards. It might, but we’ve not tested to verify.

rhapsodyv, 2020-12-03:
@GadgetAngel this PR helps with your issue: #19059 ?

GadgetAngel, 2020-12-03:
@rhapsodyv i will look to see if #19059 fixes the issue with all the errors I have found testing SPI temperature sensors this past month. If it fixes the problem great..... but I am BIG on actually testing. I have even purchased a RAMPS 1.4 board to test out my solution on. The ramps board should be here on my desk tomorrow.  But thanks, I will read to #19059 and TEST. If it solves my problems I will close the BUG report.

rhapsodyv, 2020-12-03:
> @rhapsodyv i will look to see if #19059 fixes the issue with all the errors I have found testing SPI temperature sensors this past month. If it fixes the problem great..... but I am BIG on actually [...]

Probably it will not cover everything you found. But we can join efforts to fix it! :-)

GadgetAngel, 2020-12-05:
@rhapsodyv I am still testing. What I see is that when I use the empty driver socket pins (i.e. E1 driver socket) to run a MAX31855 in software SPI mode I **can not use the free GPIO pins** when they are connected to an empty (not used) driver socket!!

Can some one explain this??   

Take for example, if I have one extruder on my printer and I am using an SKR V1.4 Turbo board, then E1 driver socket is empty.  Now if my SKR V1.4 Turbo board has a mix of UART and SPI drivers for X,Y,Z,E0 then the only free GPIO pins on E1 are E1 CS_PIN, E1 STEP_PIN and E1_DIR_PIN.  E1 has the UART jumper in place but Marlin is not using it because a TMC driver is NOT defined for it in configuration.h.  Or is this assumption wrong?  

With the current version of Marlin Bugfix branch as of today, I can not get E1's STEP_PIN or E1's CS_PIN or E1's DIR_PIN to act as the MAX6675_DO_PIN on SKR V1.4 or SKR V1.4 Turbo boards.  The MAX31855 reads all zero values.  

With the current bug-fix branch as of today, I can **only use one** of these PINS in software SPI mode (E1's STEP_PIN, E1's CS_PIN or E1's DIR_PIN).  

I can get software SPI to work (with the bugfix branch), if I use three pins from EXP1 but that means I have to use a TFT display which I do not like to use.

I should be able to use a MAX31855 in software SPI mode in the current bug-fix-branch **using ONLY** the free E1's STEP_PIN, DIR_PIN and E1 's CS_PIN (as long as the UART jumper is in place on E1).  

Here is another fact: I **can use** E1's CS_PIN, E1's STEP_PIN and E1 DIR_PIN, **if I use Adafruit MAX31855 library for the Software SPI call!!**  But there still is a problem even with using the Adafruit MAX31855 library call.  With the Adafruit MAX31855 library call for software SPI I can not use E1's DIR_PIN as MAX6675_DO_PIN.  So as long as I make E1's DIR_PIN act as MAX6675_SS_PIN or MAX6675_SCK_PIN I can get a MAX31855 hooked up using Adafruit MAX31855 library call in software SPI mode on the SKR V1.4 or SKR V1.4 Turbo (yes, I have both boards and tested both of them).   

Is this a timing issue?  If not, then why when I use the Marlin-bugfix branch (as of today) does it **not allow me to use E1's STEP_PIN as MAX6675_DO_PIN** BUT when I switch to using the Adafruit MAX31865 library call for software SPI then the same MCU board will indicate that **it can not use E1's DIR_PIN** as MAX6675_DO_PIN but it **can now use the E1's STEP_PIN as MAX6675_DO_PIN**?  

The only difference is a compilation (in one case it uses the bugfix-branch's software SPI called **max6675_spi.receive()** and the other case it is using **Adafruit_MAX31855 max855ref.readCelsius()** call?  

I put in a compile directive so that I can turn on the marlin-bugfix call or use the Adafruit library call.  I do not change the PINS file or the code, except for turning on or off a compiler directive.  

To me when I see this type of behavior on three different boards (SKR V1.3, SKR V1.4 and SKR V1.4 Turbo boards) that the real issue is due to timing problem or **I screwed up**.  In case it is "me screwing up", I am currently testing this issue on the SKR PRO V1.1 board and the GTR V1.0 board just to see if I see the same problem with these boards.  

I am hoping that I do not see the problem because the GTR V1.0 board has a MAX31855 built into the board and since the built-in MAX31855 of the GTR V1.0 board is NOT tied to the default hardware SPI bus the GTR board uses Software SPI to implement the MAX31855 call.  Has anyone tested the MAX31855 out on the GTR V1.0 board? 

I am assuming that I will NOT see the issue.  If I do see the issue then I will submit the PR which uses the Adafruit MAX31855 library for Software SPI for your consideration.  

If **I do not see the issue** (using a MAX31855 in Software SPI mode) with the SKR PRO or GTR boards  then  I will go back and try and figure out what I am doing wrong on the SKR V1.3, SKR V1.4 and SKR V1.4 Turbo boards.  By the way I am getting a eight channel logic analyzer for Christmas 👍🏻.

sjasonsmith, 2020-12-05:
Please do not pollute this report about a DELAY issue with debugging for something different.

closed 2021-02-26 by rhapsodyv

thinkyhead, 2021-08-31:
If this is still a problem with the current `bugfix-2.0.x` then please open a new issue.