MarlinFirmware/Marlin #19029: Ender 3 v1 with 4.2.7 board

badouuuuuu, 2020-08-15:
I received a creality 4.2.7 motherboard instead of a 1.1.5, but I cannot update marlin 2.0.6.
When I use the Ender 3 Pro 1.5 / Ender 3 v2 configs files, the axes do not work.
And it is impossible to make a home, I have the error "_Homing failed_" ...
I tried several times to build new firmware, but still not working...

Board Creality v4.2.7

[image]

[image]

I have no problem with this firmware: https://mega.nz/file/rNdh2ADZ#DTeOwwJ9YI7EPeQVcyvRzh_EUBuP3T8qa7RcXB8yxqc

[image]

ellensp, 2020-08-15:
Try this config. its a standard Ender 3 V1 modified for BOARD_CREALITY_V4 and TMC2208_STANDALONE drivers. 
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5078253/Config.zip)

Make sure default_envs = STM32F103RET6_creality in platformio.ini

Deimos85, 2020-08-15:
Hi,
i got also the new board for my Ender 3 Pro an get the same error. Can you please describe how to bring the firmware and the config on my printer? Is there a tutorial for this? 

This would be greate! 
Thank you!

badouuuuuu, 2020-08-15:
> Try this config. its a standard Ender 3 V1 modified for BOARD_CREALITY_V4 and TMC2208_STANDALONE drivers.
> [Config.zip](https://github.com/MarlinFirmware/Marlin/files/5078253/Config.zip)
> 
> Make sure default_envs = STM32F103RET6_creality in platformio.ini
> 
> NB config is for bugfix

@ellensp It's the same, still got "Homing failed" & no axes moves. Built with Marlin-bugfix-2.0.x

> Hi,
> i got also the new board for my Ender 3 Pro an get the same error. Can you please describe how to bring the firmware and the config on my printer? Is there a tutorial for this?
> 
> This would be greate!
> Thank you!

@Deimos85 Put [this files](https://mega.nz/file/rNdh2ADZ#DTeOwwJ9YI7EPeQVcyvRzh_EUBuP3T8qa7RcXB8yxqc) in the the root of your sdcard, and boot on your printer.

ellensp, 2020-08-15:
Can you connect pronterface? do some testing
Output of M119? for eg?
what does M115 say?  (confirming you actulay updated the firmware) 
Also try some basic movement tests. Ie try and move each axis 10 positive then 10 mm -ve 
Do they move at all, do they move in correct direction.??

badouuuuuu, 2020-08-15:
I tested with LCD Menu _Motion > Moves Axis_ moves
When you moves axis in LCD Menu, i hear a micro noise and nothing move at all, tried with  10mm, 1mm, 0.1mm.
With pronterface, nothing happens when i moved each one (check video below)

https://youtu.be/AYhkWQw7UpQ

**Chip STM32F103**

  [image]

**M119**
```
Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: open
```

**M115**
```
FIRMWARE_NAME:Marlin bugfix-2.0.x (Aug 15 2020 15:10:38) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
```

**Homing Failed error :** 
```
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```

ellensp, 2020-08-15:
Well your definitely installing firmware without issues.  
Same MPU as expected.

I was just watching a old 3DMN video of Ender 3 v2 and the motherboard was V4.2.2 
https://youtu.be/QFYegSEeGcE?t=1041

I'm wondering if they have change the pins on us 

With the working firmware if you do a M43 does it provide an output (I would be very surprised if it did as that would make things easy!)

badouuuuuu, 2020-08-15:
@ellensp ```echo:Unknown command: "M43"```
What can I do if the pins have been changed?
I am curious to learn, if you know any technical tips to fix board issues

ellensp, 2020-08-15:
Someone need to make a new pin mapping.
setup a spreadsheet with 64 rows. these are all the pins. you have
You can  grab this pins names from table 5 column  LQFP64  in https://www.st.com/resource/en/datasheet/stm32f103re.pdf
We can ignore any pins that aren't IO pins.
I would add a second column that is the pin fuction on a Crealty 4 
and a 3rd would be this board.

we know the screen works. so we can fill in those pins from the Crealty 4 column to your board column 
we know that serial 1 works as you using it to talk to pronterface so thats the same as  Crealty 4

So check other basic functionality
Does the board show room-ish temperature on lcd? if so then those thermistor pins as same as Crealty 4   

I would also look under the board as pins may be labeled there.

Since you can upload and talk to the board that is a plus

You can enable pins_debugging and with that you can use gcode M43 T
This lets you set a pin a toggling and watch for the pin to change on a meter or scope.

Just is be careful doing this, if you turn on a hot end, you may not notice till you burn yourself.

ellensp, 2020-08-15:
Other tricks, steppers will energize when the driver is enabled. so the stepper will lock in place so you cant easily turn it... 
Once energized you can find step and direction. presuming it has a step and direction pins!

jamespowell3, 2020-08-16:
I'm having the same issue.  Ender 3 pro with 4.2.7 board.  I did notice the INVERT_X_DIR, INVERT_Y_DIR, INVERT_Z_DIR, INVERT_E0_DIR in the ender 3 pro 1.5 config are opposite from the standard ender 3 prod config.  I tried reversing them but still no motor movement.

thisiskeithb, 2020-08-16:
It'd be nice if Creality actually followed the GPL and released their 32-bit firmware instead of posting [pre-compiled bins on their website](https://www.creality3dofficial.com/pages/firmware-download).

I'm surprised they are _**still**_ doing this despite all the backlash they received when they pulled this a while ago.

badouuuuuu, 2020-08-16:
@thisiskeithb Check here : https://www.creality.com/download/32
They're some new source code for v2 & Bin from 14 august 

@ellensp I understand what you say, thank you, but I think I am missing some basic for mapping, so I leave that for an expert

sjasonsmith, 2020-08-17:
@badouuuuuu none of those files seem to have source code in them for the 32-bit board. If you found it somewhere please provide a direct link.

ellensp, 2020-08-17:
what we know so far...  https://docs.google.com/spreadsheets/d/1DYhh9fwLnvZzuNvMoBen9Dl68KN-4TWkbO94lJ4iwe4/edit?usp=sharing

ellensp, 2020-08-17:
If someone with this board could make a high res scan of both sides of this board  (as high res as you can do) that might be useful. Flat bed scanners are often good for this.

badouuuuuu, 2020-08-17:
@sjasonsmith it's only source code from v2 32 bits, so the 4.2.2, my bad 

@ellensp Someone kindly took these pictures, I don't know if the quality is enough ? Because it didn't seem to work with the flat scanner : https://drive.google.com/drive/folders/1YluaMIV9-2FMR61OT8Ohzdmpn-4rUckn?usp=sharing
Tell me if it's ok for you

ellensp, 2020-08-17:
They are quiet hi res... some traces are quite visible. This should be useful (given time)

chasetb, 2020-08-17:
Just found this thread.
@ellensp Here are some photos of the board I received last week (though it's labeled as the Ender 3 Pro): https://drive.google.com/drive/folders/17jgYn8ZcDWk9XEm4wCCCQPzByd1ytRZd?usp=sharing

aylahdri, 2020-08-18:
[image]
[image]

ellensp, 2020-08-19:
I thought it might be just the stepper enable pin (since it is shared on all axis) but images and comparing disassembly from current marlin and the bin that works...seems to imply they are the same pins..  (cant be 100% sure though not all traces are visible and its hard going trough the assembly code...  (only 76,000 lines))

Flawioo, 2020-08-19:
Just for you know, Creality have posted some instructions here: https://www.creality.com/blog-detail/ender-3-motherboard-change-notification. I have sent a email to cs@creality.com to get some info.

ellensp, 2020-08-20:
Just been disassembling the code around X-Step and X-Dir pins.
They seem to be swapped from the 4.2.2 
Can someone test this for me? make sure i'm not chasing ghosts? 
Use this Config
[Ender 3 V1 modified for BOARD_CREALITY_V4 and TMC2208_STANDALONE drivers.zip](https://github.com/MarlinFirmware/Marlin/files/5100384/Ender.3.V1.modified.for.BOARD_CREALITY_V4.and.TMC2208_STANDALONE.drivers.zip)
But edit Marlin/src/pins/stm32f1/pins_CREALITY_V4.h
Change 
#define X_STEP_PIN                        PB9
#define X_DIR_PIN                           PC2

Compile and upload.  Then try some simple X moves.

chasetb, 2020-08-20:
FWIW, I complained to Creality and asked for a backup copy of the firmware for this board. What they sent indicates that the  stepper drivers are TMC2225. This is in contrast to the TMC2208 stepper drivers listed in their latest firmware on creality.com.

ellensp, 2020-08-20:
@chasetb the working firmware I've been disassembling linked by badouuuuuu https://mega.nz/file/rNdh2ADZ#DTeOwwJ9YI7EPeQVcyvRzh_EUBuP3T8qa7RcXB8yxqc is called "Ender-3- Marlin2.0.1 - V0.0.6 - TMC2225.bin"  also implying a TMC2225

chasetb, 2020-08-20:
@ellensp Perfect. There is likely not much difference then but I updated my previous Google Drive link with the stock firmware I was sent. That link again is: https://drive.google.com/drive/folders/17jgYn8ZcDWk9XEm4wCCCQPzByd1ytRZd?usp=sharing

badouuuuuu, 2020-08-20:
@ellensp It's working !
But steppers reversed, so I disabled:

```
#define INVERT_X_DIR false
#define INVERT_Y_DIR false 
```

And axies X & Y are ok !

Also work for the Z, but Extruder is not working :
```
#define Z_STEP_PIN                         PB5
#define Z_DIR_PIN                          PB6

#define INVERT_Z_DIR true
```
Zip test : [Marlin-2.0.x-test.zip](https://github.com/MarlinFirmware/Marlin/files/5102336/Marlin-2.0.x-test.zip)

badouuuuuu, 2020-08-20:
Finally, everythins is ok !

For the extruder reversed i've disabled 

```
#define INVERT_E0_DIR false
```

Working 4.2.7 : 
[Marlin-2.0.x-test-final.zip](https://github.com/MarlinFirmware/Marlin/files/5102644/Marlin-2.0.x-test-final.zip)

ellensp, 2020-08-20:
Nice, yes I confirmed all axis had step and direction lines swapped, but didn't get to updating the comment.

ellensp, 2020-08-20:
ok ill work on a PR for the new pins.. 
Are the stepper plugs keyed? ie can only go in one way? as that is what sets the need to invert or not...

badouuuuuu, 2020-08-20:
> Nice, yes I confirmed all axis had step and direction lines swapped, but didn't get to updating the comment.

badouuuuuu, 2020-08-20:
> ok ill work on a PR for the new pins..
> Are the stepper plugs keyed? ie can only go in one way? as that is what sets the need to invert or not...

I'm starting a calibration cube and check if everything is fine

Sorry for the mistake***

badouuuuuu, 2020-08-20:
> Are the stepper plugs keyed? ie can only go in one way? as that is what sets the need to invert or not...

For the extruder, i can go in negative, and XYZ are locked

ellensp, 2020-08-20:
PR submitted, I've asked @thisiskeithb if he wants to make an official Ender 3 V1 with V4.2.7 controller example config, from the information provided here.  (as he seems to like doing that sort of thing)

badouuuuuu, 2020-08-20:
> PR submitted, I've asked @thisiskeithb if he wants to make an official Ender 3 V1 with V4.2.7 controller example config, from the information provided here. (as he seems to like doing that sort of thing)

Thanks a lot for your help !

About the BLTouch is not responding by default, i'm talking with someone, he's trying, i'll back to you

Flawioo, 2020-08-20:
Hi! I've tested the configs Marlin-2.0.x-test-final.zip that @badouuuuuu said above and it worked. I've enabled BLTouch but it didn't work. It power on, the lights are ok and the inicial self test after power on the printer seems ok but no command responses.

eduardoluism, 2020-08-20:
> Hi! I've tested the configs Marlin-2.0.x-test-final.zip that @badouuuuuu said above and it worked (thanks all). I've enabled BLTouch but it didn't work. It power on, the lights are ok and the inicial self test after power on the printer seems ok but no command responses.

files @badouuuuuu sent does not have BLTOUCH enabled, check for BLTOUCH in Configuration.h, build and test again.

Flawioo, 2020-08-20:
> files @badouuuuuu sent does not have BLTOUCH enabled, check for BLTOUCH in Configuration.h, build and test again.

Sorry my friend, I forgot to inform that I've enabled the configs that BLTOUCH needs though.

fe4162, 2020-08-20:
good friends could someone be so kind to make a little video or explain how to install the marlin for the 4.2.7 board, since I'm new and I don't understand much, thank you very much

ellensp, 2020-08-21:
@fe4162, 'we' only got this mostly working yesterday. there are no documents or videos for this.

ellensp, 2020-08-21:
Guys with actual boards lets test the servo pin used to control the bltouch. 
Please do the following.
Enable PINS_DEBUGGING in Configuration_adv.h
Find In the file pins_CREALITY_V4.h the line "#define SERVO0_PIN                          PB0   // BLTouch OUT" and change PB0 to -1
disable BLTOUCH (any and bed leveling you have) 
Compile and upload

Firstly check with M43 that nothing in marlin is using PB0, so it free for us to play with.
Stick a multi meter on the PB0 pin on dc volts.

Manually control the PB0 pin with M42   (this should be pin 16)

Eg 
M42 P16 S0;   off  (ov
M42 P16 S255;  on (3.3v)
M42 P16 S128;  PWM 50% (1.65v approximately)

If this doesn't work, they may have change this pin also.

ellensp, 2020-08-21:
Also check your bltouch servo plug is wired correctly.   needs to be gnd, Vcc, data, some apparently are not wired this way.

link1107, 2020-08-26:
I've also bought this board and now cant make bltouch 3.1 to work. Tried all possible fixes that i've found across the internet, but still no luck. Selftest - ok, red light - ok, but when autohoming by z axis - just goes down with the sensor being retracted...

mcunanan19, 2020-08-26:
What pin did you use on the board? I fixed mine by using the 3 pin in the
default BL_T the G,V, IN. But I did not use the G, OUT pin. I use the old
PIN in where the previous bltouch is connected. Then in the firmware, I
still indicate SERVO0_PIN 27. It worked for me.

ellensp, 2020-08-26:
@mcunanan19  thats PB11    and isn't in V4 or V427 pins file ...   Where is this "old pin" you used. (Diagram please)

mcunanan19, 2020-08-26:
I only used those pin. in the firmware, you have to declared "SERVO0_PIN 27"

[image: image.png]

link1107, 2020-08-26:
> What pin did you use on the board? I fixed mine by using the 3 pin in the default BL_T the G,V, IN. But I did not use the G, OUT pin. I use the old PIN in where the previous bltouch is connected. Then in the firmware, I still indicate SERVO0_PIN 27. It worked for me.
> […](#)

I've already fixed the problem with the method you've described :)
Thanks for the reply.

mcunanan19, 2020-08-26:
Your welcome. It's nice to help someone.😀

Flawioo, 2020-08-26:
Marlin Firmware to get Creality v4.2.7 boards working perfectly

Steps to help anybody that needs to get this board working ok

Source of this how to: https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/blob/master/README.md
Visual help: https://youtu.be/OJmE9xwPk1U

Download bugfix from git hub https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip

Download Configuration exemples: https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip

In
Change: default_envs = mega2560
To: default_envs = STM32F103RET6_creality

In Configuration.h
Change: #define MOTHERBOARD BOARD_CREALITY_V4
To: #define MOTHERBOARD BOARD_CREALITY_V427

Change: #define CUSTOM_MACHINE_NAME "Ender-3 Pro V1.5"
To: #define CUSTOM_MACHINE_NAME "Ender-3 32bits" // Or any name you'd like to

Change: //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
To: #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // Enable BLTouch probe pins (white and black wires) to be connected to Z-
or
Change: //#define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS default
To: #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)

Pinout on board (5 pins named as BL_T) and respective connections

Board-----BLTouch wires
G---------Green----Servo
V---------Red------Servo
IN--------Yellow---Servo
G---------White----Probe (if it doesn't work, try to swap with Black)
OUT-------Black----Probe (if you are using Z_MIN_PROBE_PIN 17, if you don't, please just uncomment Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN as above)
Change: //#define BLTOUCH
To: #define BLTOUCH

Change: #define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }
To: #define NOZZLE_TO_PROBE_OFFSET { -40, -10, 0 } // Using this thingfile as mount, measure correctly using your mounted BLTouch https://www.thingiverse.com/thing:3003725

Change: #define PROBING_MARGIN 10
To: #define PROBING_MARGIN 15 // If you'd want stay away from the edges (higher values goes to center of the bed)

Change: #define MIN_SOFTWARE_ENDSTOP_Z
To: //#define MIN_SOFTWARE_ENDSTOP_Z // To allow to set Z offset values in negative

Change: //#define AUTO_BED_LEVELING_BILINEAR
To: #define AUTO_BED_LEVELING_BILINEAR

Change: //#define LCD_BED_LEVELING
To: #define LCD_BED_LEVELING

Change: //#define Z_SAFE_HOMING
To: #define Z_SAFE_HOMING

In my case using runout_sensor
Change: //#define FILAMENT_RUNOUT_SENSOR
To: #define FILAMENT_RUNOUT_SENSOR

Change: //#define NOZZLE_PARK_FEATURE
To: #define NOZZLE_PARK_FEATURE // To change filament

Change: #define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }
To: #define NOZZLE_PARK_POINT { (X_MAX_POS - 10), (Y_MIN_POS + 10), 20 } // To change the position of the hotend in front on the right of the bed

In Configuration_adv.h
Change: //#define BLTOUCH_DELAY 500
To: #define BLTOUCH_DELAY 500

Change: //#define ADVANCED_PAUSE_FEATURE
To: #define ADVANCED_PAUSE_FEATURE

Change: //#define PARK_HEAD_ON_PAUSE // Park the nozzle during pause and filament change.
To: #define PARK_HEAD_ON_PAUSE // Park the nozzle during pause and filament change.

Change: //#define PINS_DEBUGGING // M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe To: #define PINS_DEBUGGING // Only if you'd like to discover new features, if you don't, don't let it as it is

Compile
In Auto Build Marlin at Visual Studio, click on Build botton and wait for the green line\ 1 succeeded in 00:00:26.355

Where is the firmware I've compilled
In my case: C:\creality-v4.2.7-marlin-2.0.6-bugfix\Marlin-bugfix-2.0.x.pio\build\STM32F103RET6_creality
In your case: somedir.pio\build\STM32F103RET6_creality

thinkyhead, 2020-08-27:
Sorry about that. The `Z_MIN_PROBE_PIN` for the V4 board is now fixed.

Flawioo, 2020-08-27:
> Sorry about that. The `Z_MIN_PROBE_PIN` for the V4 board is now fixed.

Thanks!

Emile58-F, 2020-08-28:
Bonjour. Excusez moi, c'est bien la bonne connexion, pour installé  un bl touch. Parce que ici j'ai acheté sur le site Amazon.fr mais les branchements avec 5 broches, c'est la même carte mère creality v4.2.7 vous pouvez m'aider s'il vous plaît...
Bonne soirée 
Emile58-F

Flawioo, 2020-08-28:
> Bonjour. Excusez moi, c'est bien la bonne connexion, pour installé un bl touch. Parce que ici j'ai acheté sur le site Amazon.fr mais les branchements avec 5 broches, c'est la même carte mère creality v4.2.7 vous pouvez m'aider s'il vous plaît...
> Bonne soirée
> Emile58-F

Hi please could you translate to english first to all of us? We'll really appreciate.

mweirath, 2020-08-29:
@Flawioo I wanted to thank you for your documentation, although I am still having an issue after following the instructions.  

I think my problems are with the following lines: 

```
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // Enable BLTouch probe pins (white and black wires) to be connected to Z-
// --- or ---
#define Z_MIN_PROBE_PIN PB1 // Creality V4.2.7 BLTouch OUT (written on the board)
```
I have tried both lines, although I initially used your other documentation for the second line that I believe had a value of 17.  I haven't gone back and tried the other version, although it isn't clear which line I need to use.  Right now I am trying the first line, as I am using the 5 pin connection on the board.  My current issue is that when autohoming the printer won't lower on the Z axis.  In fact, now I can't even get the Z axis to go down, only up.

Hotel-Delta, 2020-08-29:
@Flawioo thank you very much for your efforts and your documention. I am using your compiled firmware [https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/blob/master/firmware-20200825-161146-BLT-and_Runnout-filament.bin](url) and it works perfectly.

The last little issue is the loose connection between the 3-pole BL-touch connector and the 5-pin connector on the board, obviously both connectors don't fit very well together. What is name of this type of connector housing I need to buy to make it fit better?
[image]

Flawioo, 2020-08-29:
> @Flawioo thank you very much for your efforts and your documention. I am using your compiled firmware [...]

Hi @Hotel-Delta 
I think the name is "jst connector female 5 pin"

Flawioo, 2020-08-29:
> @Flawioo I wanted to thank you for your documentation, although I am still having an issue after following the instructions.
> 
> I think my problems are with the following lines:
> 
> ```
> #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // Enable BLTouch probe pins (white and black wires) to be connected to Z-
> // --- or ---
> #define Z_MIN_PROBE_PIN PB1 // Creality V4.2.7 BLTouch OUT (written on the board)
> ```
> 
> I have tried both lines, although I initially used your other documentation for the second line that I believe had a value of 17. I haven't gone back and tried the other version, although it isn't clear which line I need to use. Right now I am trying the first line, as I am using the 5 pin connection on the board. My current issue is that when autohoming the printer won't lower on the Z axis. In fact, now I can't even get the Z axis to go down, only up.
> 
> My goal is to use the 5 pin connector and disable the Z Stop.

Hi @mweirath, to achieve your goal:

The previous "how to" was updated by our friends that have done a great work, but let's go to your problem.

The two lines below do exactly the same thing, pin PB1=17. But don't worry because it ins't needed anymore (nowadays).
```
#define Z_MIN_PROBE_PIN PB1
// and
#define Z_MIN_PROBE_PIN 17
```

Use the 5 pins BL_T and use the common Z-endstop switch (both at the same time)
or
enable this line 

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

and be happy using probe pins of BLTouch connected on Z-

mweirath, 2020-08-29:
@Flawioo I tried the updates.  Right now I have both lines of code enabled and my Z Axis end stop is unplugged.  

```
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define Z_MIN_PROBE_PIN PB1 // Pin 32 is the RAMPS default
```

I am still having a problem that my ender 3 will no longer descend.  If I increase the X-axis it will go up, but it will not go down.

Flawioo, 2020-08-29:
> @Flawioo I tried the updates. Right now I have both lines of code enabled and my Z Axis end stop is unplugged. Only the BLTouch is plugged in using the 5 Pin Connector.
> 
> ```
> #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
> 
> #define Z_MIN_PROBE_PIN PB1 // Pin 32 is the RAMPS default
> ```
> 
> I am still having a problem that my ender 3 will no longer descend. If I increase the X-axis it will go up, but it will not go down.

Ok, try this simple firmware pre compiled. It doesn't have BLTouch or Runnout filament sensor enabled but you can test if the problem is the firmware or some kind of connection.
https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/blob/master/firmware-20200829-011646-Simplest.bin

mweirath, 2020-08-29:
@Flawioo I tried with the file you shared and I am able to get the Z-Axis to work, although I can't go below 0 (currently the Ender is sitting around +80, which it thinks is 0).

mweirath, 2020-08-29:
I found another recommendation that I am trying.  It seems to have addressed the known issues, but I still need to validate all the settings are correct and I can correctly print.  

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING

//#define Z_MIN_PROBE_PIN PB1 // Pin 32 is the RAMPS default
#define Z_MIN_PROBE_PIN Z_PROBE_PIN // Pin 32 is the RAMPS default

[9 more replies]