MarlinFirmware/Marlin #19033: [FR] Add the possibility to reheat the extruder far away from the print on filament change

Gerporgl, 2020-08-15:
#11497 # Description

Add the possibility to reheat the extruder far away from the print on filament change resume position when doing an ADVANCED_PAUSE_FEATURE on filament change and using HOTEND_IDLE_TIMEOUT so that the hotend does not reheat while touching the print.

### Feature Workflow

1. After the nozzle has been primed and the user select continue from the ADVANCED_PAUSE_FEATURE if the hotend is not at the right temperature that it was before it should allow it to reheat but before moving to the resume position so that it does not stay hot and melt the print in progress.

#### Additional Information

I am using octoprint and the filament runout detection is connected to the Marlin board. The filament change process seems to works well even when using the host (octoprint) to print but it seems the hotend is resuming its reheating process on the print, which leads to some artifacts at the print resume location.
I am not entirely sure this is something that octoprint does or if this is really caused by the HOTEND_IDLE_TIMEOUT like I am assuming.

thinkyhead, 2020-08-15:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2020-08-16:
This behavior was changed to reheat the nozzle before resuming in https://github.com/MarlinFirmware/Marlin/pull/19009 which is now merged.

Gerporgl, 2020-08-16:
I tried the latest bugfix-2.0.x branch and it works for me! Thanks!

closed 2020-08-16 by thisiskeithb