MarlinFirmware/Marlin #19036: [BUG] UBL G29 P1 does not complete

logiclrd, 2020-08-16:
### Bug Description

I have Marlin configured to generate a 15x15 UBL mesh with a `PROBING_MARGIN` of 20. When I issue `G29 P1`, it starts probing points, but when it gets to point 211/255, it moves the probe to the centre of the bed, emits a chirp and stops with no message. No further probing. Has done it 3 times in a row now, with a reset before the last attempt.

### My Configurations

[MarlinConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/5079864/MarlinConfiguration.zip)

Printer is a CoreYX design with a 400x400 bed.

### Steps to Reproduce

1. Issue `G29 P1`

2. Wait.

**Expected behavior:** After 225 probes, a mesh is generated.

**Actual behavior:** When it reaches point 211/225, it aborts the mesh generation with a chirp but no message.

#### Additional Information

Last few lines of the log of the serial connection after `G29 P1 V4`:

```
Bed X: 328.57 Y: 20.00 Z: -0.36
Probing mesh point 208/225.
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 354.29 Y: 45.71 Z: -0.14
Probing mesh point 209/225.
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 354.29 Y: 20.00 Z: -0.34
Probing mesh point 210/225.
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 354.29 Y: 380.00 Z: 0.03
Probing mesh point 211/225.
echo:busy: processing
>>> M420 S1
SENDING:M420 S1
echo:Invalid mesh.
echo:Bed Leveling OFF
echo:Fade Height OFF
```

ellensp, 2020-08-16:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

logiclrd, 2020-08-16:
The Git commit my firmware is built from (f7ff3f0) was HEAD of `bugfix-2.0.x` as of 7:30 PM yesterday.

Here is the complete log of G28 and G29 P1 as shown by Pronterface's log window, with `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` enabled in the build:

* [Printrun.log](https://github.com/MarlinFirmware/Marlin/files/5081242/Printrun.log)

It stopped at the exact same place as the previous 3 test runs, at mesh point 211/225.

I used M111 S246 instead of S247, because the 1 bit is "ECHO", and the log already contains the exact text Pronterface is sending (twice for each command). Pronterface logs its serial I/O writes with lines prefixed with `SENDING:`, but Pronterface also polls thermistor readings, and with M111 bit 1 set, I saw those polling calls get echoed as well, even though Pronterface elides them on its end.

olafnew, 2020-08-18:
Excatly the same issue! But on the Delta printer

Vertabreak, 2020-08-18:
this isn't a issue its a feature the skipped area is considered unreachable and you are expected to fill it with a smart fill g29 p3 or a manual g29 p4. your describing UBL working as intended to be clear this is exactly how UBL works and not in any way a bug or issue.

olafnew, 2020-08-18:
> this isn't a issue its a feature the skipped area is considered unreachable and you are expected to fill it with a smart fill g29 p3 or a manual g29 p4. your describing UBL working as intended to be clear this is exactly how UBL works and not in any way a bug or issue.

In my case(https://github.com/MarlinFirmware/Marlin/issues/19061) - it probes HALF the bed. It does not probe the other half. It's simply not bed leveling, when the system does not now about HALF a bed.

logiclrd, 2020-08-18:
Okay, I think I have an understanding of what is going on.

I was running on a version of the firmware from April, and it created a mesh just fine. I have a copy of the mesh from April in comments on #18977.

Based on what you're saying, though, it sounds like the April version had a bug where it was probing points that it _shouldn't_ have been probing. Turns out they're within reach with my printer just fine, so I didn't notice any problem, but the latest version of the code is paying proper attention to some configuration piece somewhere that is telling it that it shouldn't be able to move the head there.

I wrote a little program to visualize which points got probed, and this is the result:

[image]

The 15 points that it skipped are all along the right edge. My probe is attached to the left side of the print head, so that the print head has to move 40mm further to the right than the location it actually wants to probe. Factoring in the 20mm margin on all sides, that rightmost column needs the probe to be at X380, which means the print head's true position needs to be X420. With the old code, it just blithely moved to X420 and probed, and fortunately nothing went wrong. With the latest code, then, if I'm understanding correctly, something in the configuration is telling it that X420 _isn't actually a possible location_, so it doesn't try to probe any more.

Does that sound about right?

I saw this as a bug, because the April build just did the right thing, and the new build suddenly stopped.

If so, then, how do I tell Marlin that the range of motion is larger than the bed size and X420 isn't printable, but _is_ reachable?

I also think there is still a bug here of a cosmetic nature. The UI Marlin displays doesn't say, "UBL complete, all reachable points captured, 15/225 points unreachable" or anything of the sort. It simply halts the operation leaving the text `Probing mesh point 211/225` on the screen. This is a bug in _appearances_, even if the underlying logic is correct.

sjasonsmith, 2020-08-18:
The delta probing logic treats the probing area as a circle concentric to the bed. For an offset probe that effectively results in a "probing margin" of either the PROBING_MARGIN, or the hypotenuse of the X/Y offset.

I think to override the safe probing radius you will have to increase `DELTA_PRINTABLE_RADIUS` to reflect the probe offset. 

On Cartesian printers you would modify the constants such as `X_MIN_POS` to tell Marlin that the nozzle can travel over an area, but there is no bed beneath it. I don't think there is a way to do this for a delta right now. It seems that `DELTA_PRINTABLE_RADIUS` is used in many places in the code that calculate travel limits.

Somebody else may come along with a better way, this is just what I have been able to figure out reading the code.

It seems like ideally you would have two values: `DELTA_PRINTABLE_RADIUS` to declare the size of the bed, then something like `DELTA_MOVEMENT_RADIUS` to define the total range of movement.

logiclrd, 2020-08-18:
So in my case, I need to disable software endstops and then carefully test just how far right X can go, then take 1 or 2 mm off it just to be safe and stuff that into `X_MAX_POS`. Assuming the value is >= 420, it'll then be able to probe the points at X=380.

sjasonsmith, 2020-08-18:
Sorry, I got confused between this and the issue opened by @olafnew, which is about delta.

boelle, 2020-09-11:
@logiclrd that is how i did it on a big printer i have build

i just incresed the bed area in small steps so i was close to a crash at X and Y max, 

side story is that i did not allow for the bl touch probe in the cad drawings and i lost about 20mm in each direction where the nozzle cant reach

but did you figure things out?

logiclrd, 2020-09-11:
Yeah, my _old_ behaviour was actually a bug. It shouldn't have been doing it, as far as I can tell. It probed a 15x15 grid where some of those points required the head to go beyond `X_MAX_POS` to get the probe to the right place. After updating Marlin, it started following the rules, which means that the rightmost column of samples was simply skipped -- not possible to do automatically. I thought _that_ was the bug because I had seen it do a 15x15 grid before. But, following the discussion here, I adjusted my settings, carefully measuring the maximum positions (turns out that after replacing my hot end, I actually can't even reach the full 400mm of the bed any more in the Y direction! there's no endstop in the positive direction, but the gantry crashes around Y 392), and also adjusting the margin for the UBL grid. After doing that, it now measures all 15x15 points.

closed 2020-09-11 by logiclrd