MarlinFirmware/Marlin #19037: [BUG] (Y Motor getting too hot 2.0.6)

codyburns59, 2020-08-16:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Marlin-2.0.6 first attempt.zip](https://github.com/MarlinFirmware/Marlin/files/5079877/Marlin-2.0.6.first.attempt.zip)

[Marlin-2.0.6 attempt 2.zip](https://github.com/MarlinFirmware/Marlin/files/5079879/Marlin-2.0.6.attempt.2.zip)

### Steps to Reproduce

use anet a8 example template in marlin, make changes to include btt skr mini e3 v2 and tmc2209 drivers, with the factory motors (0.9a) I came up with the current of:
0.9a/1.41=0.6382_rms
vref=0.6382*2.5=1.59574/1.77=0.9015*.90
=0.8114

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
motors be slightly warm to the touch and function properly

**Actual behavior:** [What actually happens]
motors are warm enough to cook an egg

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
does not demonstrate the issue but here is the setup if it helps: https://www.thingiverse.com/thing:4565399/files

boelle, 2020-09-11:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. 

you can use a tool like winmerge to make sure settings are copied over correctly

sjasonsmith, 2020-09-11:
You do not have the slave addresses configured in your Configuration_adv.h. This board uses hardware serial and TMC addresses to talk to each driver. You are currently configuring one of the drivers four times, rather than configuring all four drivers.

While not configured for your printer, you can look at one of the Ender3 examples which use this board as an example.
https://github.com/MarlinFirmware/Configurations/blob/c7bb5c5d230044ca2cf9195ab5453df0cd5a799d/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration_adv.h#L2393

I am closing this issue, since it can be easily explained from your configuration files.

If you need further help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-11 by sjasonsmith