MarlinFirmware/Marlin #19039: (Bug) BTT SKR 1.4 Turbo-  Fan1 through E1, TH0 and E0det Not turning on, at Bootup - When X-Stop unplugged, after update to 2.0.6 through Latest Bugfix 2.0.X

elektrofonic1, 2020-08-16:
This is my first post, so please excuse if I missed a posting on this issue or if I placed this in the wrong place. I have searched what seems to me to be all the internets to no avail. I feel as if I don't know what I should be typing to find what I am looking for. Seeing as this is my first post, it can  be assumed that I am not one to post unless I am absolutely stuck, as I hate bothering anyone.

### Bug Description

On my BTT SKR 1.4 Turbo, my Fan1 on E1, TH0 and E0det is not turning on, at bootup - When X-Stop unplugged. When i reboot, with X-Stop plugged through P1_29 and GND, Everything runs, but the Fan1 which should be running only when hotend reaches 50C, runs no matter what.

Everything was working perfectly until I updated past 2.0.53 (latest bugfix right before 2.0.6)

### My Configurations

[Configuration AND adv.zip](https://github.com/MarlinFirmware/Marlin/files/5080189/Configuration.AND.adv.zip)

### Steps to Reproduce

Load latest bugfix
Unplug X-Stop and Reboot to get nothing working right.
Also, Plug in and reboot to get the mostly working setup

**Expected behavior:** [What you expect to happen]

Fan0 should be off until hotend reaches 50C
Filament sensor LED should be on and the sensor should function.
Thermistor on TH0 should not be reading -19

**Actual behavior:** [What actually happens]

When X-stop is plugged in, Fan0 spins as soon as device is booted, Filament sensor works and Thermistor reads room temperature.

When X-stop is unplugged, Fan0 is off as soon as device is booted, Filament sensor and LED is off, and Thermistor reads -19

#### Additional Information
BTT SKR V1.4 Turbo
Sensorless Homing is ON
TMC2209 Drivers on X,Y,Z1,Z2,E0

I apologize if I missed anything important. If I did, I'll be happy to include after. Thanks!

Tino

thisiskeithb, 2020-08-16:
So if you flash back to `2.0.5.3`, these issues don’t occur?

It sounds more like a board/wiring issue than a Marlin bug.

elektrofonic1, 2020-08-16:
I've tried replacing the board with another V1.4 and it still didn't work out.

I'll try reflashing now.

ellensp, 2020-08-16:
Description is not clear... but for a auto fan for keeping hot end cool you need to set E0_AUTO_FAN_PIN to the correct pin. Provided Configs says -1 ie not enabled.

elektrofonic1, 2020-08-16:
Just so I'm clear, even if my pins file contains:

#if HOTENDS == 1
  #ifndef FAN1_PIN
    #define FAN1_PIN                       P2_04
  #endif
#else
  #ifndef HEATER_1_PIN
    #define HEATER_1_PIN                   P2_04
  #endif
#endif

which is in fact the correct pin associated with E1, I need to set E0_AUTO_FAN_PIN to FAN1_PIN?

I just attempted the change, and I'm still getting the same reaction.

The oddest thing is, I cannot seem to track down why I'm getting this reaction.

Filament sensor is in E0Det [P1_26 and 5V]
Thermistor is in TH0 [P0_24 and GND]
The Hotend Cooling Fan is in HE1 [P2_04 and 24V]

Im Using Stallguard and the diag pins are cut on Z, E0, E1 (driver in use as Z2)

My 2 endstops on X and Y are unplugged and thats the issue.

Y-stop unplugged is perfectly fine. X-stop, when unplugged causes Fan1 to not spin as it shouldn't. The filament sensor is off and my thermistor is registering sub temps and unless i have x-stop plugged in, they continue to not register.

Thank you for the suggestions thus far.

thisiskeithb, 2020-08-17:
> I need to set E0_AUTO_FAN_PIN to FAN1_PIN?

Comment out the `E0_AUTO_FAN_PIN` line in your config since you have it set to `-1` (which undefines it), or set it to `FAN1_PIN`.

elektrofonic1, 2020-08-17:
For future reference, if someone converts a CR-10S Pro to a BTT SKR 1.4 Turbo motherboard, and they choose to break out the ribbon cable on the board's side, If they decide to go to Sensorless Homing (TMC Stallguard), they need to make sure to:

**On PINS 11 & 12 from the ribbon** [X- Endstop] 
Keep Pin 11 [GND X-/X&Z-] plugged in to GND on X_MIN (on SKR)

**On PINS 3 & 4 from the ribbon** [X+ Filament Sensor]
Keep Pin 4 [5V] plugged in to 5V on E0Det (on SKR)

Y- can be completely unplugged because it's not part of the ribbon, but when unplugging the endstops, the ribbon still needs to supply the breakout board on the X gantry with 5V and a GND from somewhere. Without these pins plugged in, the following issues occur:

The filament sensor LED isn't supplied the voltage it needs to function.
The Thermistor doesn't function and the hotend fan doesn't run. 

I'm assuming it has to do with the GND not reaching the breakout board, but full disclosure obligates me to say I haven't verified this assumption.

I apologize now if this information falls out of the scope of the Firmware itself, but I feel if someone else converts their CR-10S Pro, and turns their sensorless homing on, they'll run into this issue and considering this information is nowhere else, It may come in handy. If it belongs elsewhere, please point me in the right direction and I'll be happy to move it. Thanks for all the suggestions!

closed 2020-08-17 by elektrofonic1