MarlinFirmware/Marlin #19040: [FR] SD-Card Menu for Ender 3 V2 like in predecessors models

Semmi121, 2020-08-16:
Is it possible to add the sdcard menu from the predecessors models of the Ender 3 line also on the new V2 Marlin Firmware?
I use an external SD-Card Reader and the internal sdcard of the printer has a pin that triggers whether the memory card must be refreshed. This doesn't work for the external reader that doesn't show new content of the memory card even you restart the printer. 
I hope it is possible cause of the new display of the Ender 3 V2.

Thanks

Semmi

Semmi121, 2020-08-25:
Could somebody write few words whether it would be a big issue to add this functionality or not? As I wrote in my beginning question I suppose that this could be difficult cause of the new display. I saw the problem with the "Store configuration" and I suppose that it would be more difficult to add a complete menu. Maybe I'm wrong. 

A lot of people use an external reader and maybe somebody could explain a workaround for this problem, even if the menu isn't possible in near future. Thanks a lot in advance for your help.

sjasonsmith, 2020-12-02:
@Semmi121 please provide more details about what you are looking for. Specifically:

1. What hardware are you using the external reader, and how are you connecting it? Please include links and pictures as needed to describe it.

2. Zip up and attach your configuration files which enable the external SD. Also include your pins file and any other files you have modified.

3. Provide _exact_ steps to reproduce the problems you are facing, that you think this feature request will solve.

jojolll, 2020-12-07:
> please provide more details about what you are looking for. Specifically:
> 
> 1.

I encountered the same problem on my Ender3V2
It will happen no matter which external reader is used or which version of marlin: the microSD port of the Creality 4.2.2 card only updates the content when the `SD_DETECT_PIN` is activated -> so you have to physically unplug the microSD extension and plug it back in to update the content display, or restart the printer which is still quite annoying.

On the E3V2 there is no original option to refresh the card as on the E3.
> Provide exact steps to reproduce the problems you are facing, that you think this feature request will solve.
Since the microSD port is more fragile than a regular SD, many users will plug in a MicroSD -> SD extension.

@Semmi121 
I found an ugly but functional solution for those who have a BLTOUCH while leaving the Z endstop plugged. 
In `Marlin-2.0.x\Marlin\src\pins\stm32f1\pins_CREALITY_V4.h`, I exchanged the address of the `Z_STOP_PIN` and that of `SD_DETECT_PIN`.

--> Now, when I press the Z limit switch (which I have shifted so that it is only activated manually) the content of the SD card disappears (NO MEDIA), and when I release it it refreshes.

Semmi121, 2020-12-17:
@jojolll 
I have no BLTOUCH and so I cannot use your workaround. It's a little bit tricky, but if it works, why not. Maybe there is a different method to solve the problem also when the Z-Endstop.

@sjasonsmith 
The issue was good described by jojolll. I have no idea whether it's possible to create an extra menu for the sdcard, like for old E3 devices, or there is an opportunity to bypass the SD_DETECT_PIN flag. As mentioned by jojolll the microSD port is not the stablest and so it would be fine if we can use the external reader. I hope that somebody can find a solution for that issue.

Joho-x, 2020-12-29:
I had a similar problem trying to use a microSD to SD extension on my Ender 3 with a 32bit board (V4.2.2). But after some digging I found that all the functions are already in Marlin, but they are disabled when a SD_DETECT_PIN is configured. 
Therefore all I had to do is comment the following line in /Marlin/src/pins/stm32f1/pins_CREALITY_V4.h (or the appropriate config file for your printer) and it works how I wanted it to be.
`//#define SD_DETECT_PIN                       PC7`

What happens is, that if not defined SD_DETECT_PIN gets set to -1 (false) in pins.h. As a result automatic SD-card detection is disabled and you get a menu item “Attach Media” or “Release Media” in the main menu. So you lose some comfort but can use a SD-card you won't accidentally breath in ;-).

Hope it helps.

Semmi121, 2020-12-31:
@Joho-x 
I guess that the problem could be, that the menu items concernig the sd-card aren't there for the Ender 3 V2, cause of the another display. Maybe someone could say a few words about this solution whether it would be practicable or not for the E3V2.

jojolll, 2021-01-10:
> @Joho-x
> I guess that the problem could be, that the menu items concernig the sd-card aren't there for the Ender 3 V2, cause of the another display. Maybe someone could say a few words about this solution whether it would be practicable or not for the E3V2.

I've tried @Joho-x solution before looking for mine, but as can be expected nothing appears in the menu on the E3V2 screen. The only function removed is the detection of the MicroSD, so we are forced to turn off the machine in 100% of the cases even without extension.

Joho-x, 2021-01-10:
@jojolll
I don't have access to a V2 screen, so I can't test it, but do the custom user menus work with the V2 Display?
Then I would suggest to change the CUSTOM_USER_MENUS code in the Configuration_adv.h to something like the following:

``` C++
/**
 * User-defined menu items that execute custom GCode
 */
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
	#define CUSTOM_USER_MENU_TITLE "SD card"
 	#define USER_SCRIPT_DONE "M117 User Script Done"
	#define USER_SCRIPT_AUDIBLE_FEEDBACK
	//#define USER_SCRIPT_RETURN  // Return to status screen after a script

	#define USER_DESC_1 "Release SD card"
	#define USER_GCODE_1 "M22"

	#define USER_DESC_2 "Init SD card"
	#define USER_GCODE_2 "M21"

#endif
```