MarlinFirmware/Marlin #19041: PlatformIO Core 4.4 compatibility

ivankravets, 2020-08-16:
Hi all,

We have refactored our package management system in PlatformIO Core 4.4. See upcoming changes https://github.com/platformio/platformio-core/blob/develop/HISTORY.rst
This is the biggest upgrade to package manager since the 2014 year. We didn't affect CLI  or other public APIs. Nevertheless, internal modules are changes.

Please ping me here if you need any help with adapting your "extra scripts" for any PIO Core version.

Regards, 
Your friends at PlatformIO.

rhapsodyv, 2020-08-16:
I just made the package manager / spec compatibility:
#19034

But some builds were failing with "missing package" error. But it just happens in the PIO dev version. But I will test again today.

ivankravets, 2020-08-17:
> But some builds were failing with "missing package" error. 

Could you provide more details? Thanks!

ivankravets, 2020-08-17:
We uploaded a package with the Arduino framework for LPC1768. Please re-test the latest development version of PIO Core.

ivankravets, 2020-09-03:
Yes, [PlatformIO Core 5.0](https://github.com/platformio/platformio-core/releases/tag/v5.0.0) is out! Thanks for your reports!

closed 2020-09-03 by ivankravets