MarlinFirmware/Marlin #19046: [FR] G34 Adjust 4 point Levelling alignment process

Celcius1, 2020-08-16:
### Description
Adjust the method of alignment in the 4 point levelling

### Feature Workflow

On the weekend I implement the ABL with 4 Z steppers for the most part it worked OK, but the method in the way it levelled was incorrect, current implementation is to probe the 4 points in this sequence SW, NW, NE, SE, and then move all 4 motors, I find from observing the printer behaviour,  This is not the correct method for levelling a bed when using 4 points, not to mention the printer can get the bed out of whack pretty fast.

In practicality the levelling process should be first obtain your home position in the centre of the bed.

Then probe SW and NE and adjust bed tilt, then check again. Once level to within an accuracy of say 0.3 then move onto the next diagonal.
Then probe SE and NW and adjust bed tilt, then check again.  Once level to within an accuracy of say 0.3 then check all 4 points.

This process at any one time is only moving to steppers and adjusting on the diagonal.  This is how I level my bed manually and also how I do it in with many other things.  I would code the change myself but am having a hard time, trying to find where I make the changes in the code.  To change the levelling behaviour.  So if I can change the levelling behaviour to the process mentioned above this should get the routine working quite well.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-08-16:
@InsanityAutomation may be interested in this. He added the 4-point support, but I'm not sure if he actually had a printer to try it with at the time.

Celcius1, 2020-08-17:
> @InsanityAutomation may be interested in this. He added the 4-point support, but I'm not sure if he actually had a printer to try it with at the time.

Oh ok I'm trying to understand the code to see if I can change it but I might be getting in over my head as my speciality is more hardware and mechanical design than programming, but I'll do my best but if he see's the request, I'll be happy to work with him to optimise the 4 point levelling routine.

InsanityAutomation, 2020-08-17:
This has been on my list to revisit and the method sounds simpler than what I had in mind. Day job has me booked solid for the next 3 weeks though.

Celcius1, 2020-09-13:
> This has been on my list to revisit and the method sounds simpler than what I had in mind. Day job has me booked solid for the next 3 weeks though.

Its been nearly a month since you responded, have you made any progress with the 4 point levelling routine, and if you have could you send me the relevant files so I can test the changes.

Regards

Celcius1, 2020-09-26:
> This has been on my list to revisit and the method sounds simpler than what I had in mind. Day job has me booked solid for the next 3 weeks though.

@InsanityAutomation have you even managed to get to work on this, its been more than 3 weeks now, an update would be considered polite!

InsanityAutomation, 2020-09-26:
I have not, day job has been 60+ hours a week so ive been struggling to keep up with the commitments to people who provide hardware. This one is a bit down the list as its a project for a machine I want to build but havnt had time to yet.

Celcius1, 2020-10-14:
@InsanityAutomation I understand you are quite busy and want to test this code, on your own project but I'm more than happy to test this code otherwise could you provide details as to where I will find the routines related to this code so I can attempt to make the changes myself

Celcius1, 2021-04-12:
Ok I get the impression I'm being ignored here, I've told you how to fix the leveling routine, and I currently use it in its limited capacity, and I even said I would test the code, but you have done nothing regarding this issue I raised, I REPEAT nothing for 6 months, would you get your finger out of your backside.  And do the needed code changes, Marlin is a much bigger code package than GRBL, so not as easy to make changes to unlike GRBL.

InsanityAutomation, 2021-04-12:
You want to pay commercial rates for me to work on things ive volunteered my time for to put it ahead of the people im currently working 50+ hours a week for? My rate is $110ph. You want to throw insults around, well I will now promise without commercial sponsorship this will never be touched.

rhapsodyv, 2021-04-12:
@Celcius1 read this to understand how open source works: https://github.com/MarlinFirmware/Marlin/issues/18429#issuecomment-810573722