MarlinFirmware/Marlin #19048: Autoload filament with filament presence sensor

BsCmOD, 2020-08-17:
Hi to every one, I don't know if there were someone other who has done such request, if there was I apologize for doubled request, however I think it could be a cool feature by adding the possibility into configuration.h file not only the enable option but also to choose the lenght of filament to load, at which speed and with how much delay. Thank's for attention and many thanks for the hard job done day by day by the developers!

sjasonsmith, 2020-08-17:
The name is not very obvious, but I think that `ADVANCED_PAUSE_FEATURE` in `Configuration_adv.h` is what you are looking for.

I don't use it myself so can't answer questions about it, but you can visit one of the various Marlin support forums if you have questions about it.

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-08-17 by sjasonsmith

thisiskeithb, 2020-08-17:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/9195