MarlinFirmware/Marlin #19049: [BUG] (MKS Robin Nano 1.2 TFT 3.5: New UI crashes selecting Babysteps

Einhorn-1802, 2020-08-17:
I am using the MKS Robin Nano Board V1.2 together with the 3.5 inch MKS touch display and I have tried different user interfaces.

In my configuration I have the option to use the old "standard" Marlin UI. 
For that UI I am selecting 
#define FSMC_GRAPHICAL_TFT
and everything works fine.

Using the new interface selected with
#define TFT_480x320
I see a problem with the menu item Babystep.
Touching that item in the menu, the screen gets dark (nothing visible) and the printer stops immediately.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5083149/Marlin.zip)

### Steps to Reproduce

1. Flash the firmware
2. Start a print job
3. During print operation: Press Settings ->Settings -> Tune -> Babystep

Nothing appears on the display. The display is empty (black) and the printer stops
I have seen facebook posts desribing the same issue. So it's not only linked to my printer or setup.

jeor112, 2020-08-30:
I have the same issue, but with latest release 2.0.6.1

sjasonsmith, 2020-09-11:
A sanity check was added which should prevent you from enabling this unsupported feature on that display.
#19284

sjasonsmith, 2020-09-11:
I am presuming this resolves this issue, and will close this issue. Please reply if you still have issues in the current bugfix branch.

sjasonsmith, 2020-09-11:
Duplicate of #19233

closed 2020-09-11 by sjasonsmith