MarlinFirmware/Marlin #19050: [BUG] TMC drivers retain old hybrid chopper settings after firmware update.

looxonline, 2020-08-17:
### Bug Description

It seems as if the hybrid switching settings are only sent to the TMC drivers when hybrid switching is enabled and not when hybrid switching is disabled.

This means that if I reprogram the firmware and turn off hybrid switching it is not actually turned off after the board reboots since the drivers can maintain their power and volatile memory through a board reset. To get hybrid switching to turn off requires a full power cycle of the whole printer which is not a huge issue but since the code to configure hybrid switching is already there it may make sense to add a #ifndef check which actively turns it off instead of just assuming that it is off.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1.) Enable hybrid switching on a board of your choice.
2.) Disable hybrid switching using a programming method that does not require a full power cycle of the board.
3.) Hybrid switching is still enabled as the settings are retained within the volatile memory of the drivers.

**Expected behavior:** [What you expect to happen]

After a reprogram of the board it is natural to expect the machine to behave according to the programmed config. In this case, to have hybrid switching disabled.

**Actual behavior:** [What actually happens]

Since only the board is reset but power is retained to the drivers, volatile memory on the drivers is retained and furthermore, since the board does not actively update disabled features they may still remain enabled until the next power cycle.

#### Additional Information

N/A
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5083703/config.zip)

Hyrules, 2020-08-24:
I have reproduced this issue as well. I'm and a SKR 1.4 Trubo with TMC2209 drivers. I have to completely turn off the printer to get the new settings to apply.

phazei, 2020-09-05:
I've also been having this issue.  Need to unplug the whole board between firmware updates when doing lots of hybrid testing.

thisiskeithb, 2020-09-05:
Does Initializing EEPROM through the menu or by sending `M502` followed by `M500` use the updated values?

looxonline, 2020-09-05:
> Does Initializing EEPROM through the menu or by sending `M502` followed by `M500` use the updated values?

No because if you follow the TMC init flow you will see that the hybrid settings are not configured of its turned off. Therefore the volatile values will remain within the drivers until a power cycle. Easy fix though since the code is already there to configure them. It's really just a conditional statement. If I have time this weekend I'll try to implement and test.

looxonline, 2020-10-06:
Let's keep it active bot. Still an issue. Just no time to dig deep into it yet.

looxonline, 2020-11-07:
Still an issue. Still no time to fix it myself yet.

looxonline, 2021-01-10:
Really not sure that the bot should add that label for confirmed bugs...

thinkyhead, 2021-04-15:
> Really not sure that the bot should add that label for confirmed bugs...

Label exempted.

thinkyhead, 2021-04-15:
Currently Marlin removes *all* code related to the TMC driver feature if `HYBRID_THRESHOLD` is disabled. So, first Marlin will have to be altered to always include code for `HYBRID_THRESHOLD` when threshold-capable drivers are present. Then, in `MarlinSettings::postprocess` code will need to be added for the case where `HYBRID_THRESHOLD` is disabled which calls `set_pwm_thrs` on all the threshold-capable drivers to turn off the feature.

thinkyhead, 2021-04-15:
Q: What value can be set with `set_pwm_thrs` that disables it? It looks like modifying `tmc_init` may be enough.

thinkyhead, 2021-04-15:
Please test the linked PR to see if it does the right thing with your setup. Sorry this took so long for me to address. Typically most issues take less than an hour to fix, but if we don't attack them right away they tend to get lost in the noise.

thisiskeithb, 2021-04-16:
I spoke with @looxonline and they are comfortable with me closing this. It can always be reopened if the fix from https://github.com/MarlinFirmware/Marlin/pull/21623 doesn't work.

thisiskeithb, 2021-04-17:
Reopening since the proposed fix ended up causing any TMC-based config to lock up on boot.

thinkyhead, 2021-04-17:
So as it turns out, we can't just initialize drivers with the hybrid threshold feature to a threshold of 0 when the feature is turned off in Marlin. But, perhaps we can add a warning to Marlin to inform users that they are building Marlin in a way that will leave TMC stepper drivers in an 'undefined state.' Even if users never bother to set a hybrid threshold, at least the firmware and `TMCStepper` library will know about it and keep the value set to a functional range.