MarlinFirmware/Marlin #19051: [BUG] ultralcd.cpp screen_click() bug

andreibobirica, 2020-08-17:
<### Bug Description

### My Configurations

Now my configuration with the problem resolved:
https://github.com/andreibobirica/Marlin/blob/4755003463bc5026358ae8ef5cf1cb69fe7f374b/Marlin/src/lcd/ultralcd.cpp#L1572

### Steps to Reproduce

Use Marlin 2.0.x or buxfix 2.0.x with an TFT screen with FSMC_GRAPHICAL_TFT actived and classic UI.
This bug is noticeable only in the EDIT screen.
However, it also creates numerous problems in mesh bed leveling screens.

**Expected behavior:** [What you expect to happen]
1 click, 1 encoder action

**Actual behavior:** [What actually happens]
1 click, hundreds of action (one every clock of the screen)

#### Additional Information
To solve the problem, just encapsulate the function with a timer.
void MarlinUI::screen_click(const uint8_t row, const uint8_t col, const uint8_t, const uint8_t) {
      const millis_t now = millis();
      if (ELAPSED(now, next_button_update_ms)) {
       ......
      }
}

sjasonsmith, 2020-09-11:
Please attach your Marlin configuration files, as requested by the issue template. Marlin behavior is extremely configurable, and it is very difficult to debug any issue without knowing how the firmware is configured.

I see a link to a source file, but it isn't clear that also contains the configurations which should be used.

@andreibobirica if you already have a solution for this, I suggest you create your own Pull Request to Marlin with the fix, rather than waiting for someone else to figure out how to re-apply your change. Many of us do not have the hardware needed to reproduce the issue, so are not comfortable attempting to implement your change.

sjasonsmith, 2020-10-12:
@rhapsodyv is this the TFT UI you have been working with? If so, do you think this is still an issue or already resolved?

thinkyhead, 2020-10-12:
Thanks for the code sample. It seems sensible. I'll roll a PR from it and post it shortly.

closed 2020-10-12 by thinkyhead

andreibobirica, 2020-10-12:
Thank you very much for working on this issue, it was one of the few remaining issues on the Classic UI.