MarlinFirmware/Marlin #19055: [BUG] Stepper motor dont move in fast Velocity

carto1985, 2020-08-17:
Hi Guys,
I am having a problem with my 3d printer project.
I am using Mega2560 / RAMPS1.4/srv8825 with motor sanyo denki 103H546,
I configured drivers on fullsteps, When I Put the step/mm 200 on marlin V2.0.x
DEFAULT_AXIS_STEPS_PER_UNIT   { 200, 200, 200, 500 )
whe I drive My motors in 3000mm/min on pronterface the motor dont moves, just do a high pitch noise, but when I slow down the speed to 300mm/min it moves correctly, I would like to work with 1/16 step, but this configuration dont work too.
There is another configuration that Im missing?
Thank you for your help.

AnHardt, 2020-08-17:
Let's calculate.
300mm/min = 50mm/s. With 200steps/mm we get 10,000 steps/s. With a 8bit AVR at 16MHz that's about the maximum interrupt rate we can get - likely a bit more. For that Marlin does switch into double-step-mode, taking two steps per interrupt. In fullstep-mode of the drivers that will not work because no moment is produced.
Switching to 1/16 step-mode can't help. Then you have to do 16,000steps/s, so Marlin will do 32steps per interrupt at that speed.
Either move slower, or reduce the gearing to get less steps/mm or take a faster processor.

boelle, 2020-08-17:
@carto1985  is this a question or bug?

carto1985, 2020-08-17:
Hi, thank you for your explanation,
But How Can I reduce the speed in Marlin? And what are the average speed on 3dprinters? I dont have this data. 300mm/min its a good speed for printing? Should use 1/2 or 1/4 step?
Thank you again
I have a leadscrew of 1mm per turn,

boelle, 2020-08-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-17 by boelle