MarlinFirmware/Marlin #19056: [BUG] BED PID Autotune and PSU auto power-on

pietchaki, 2020-08-17:
### Bug Description

Trying to run PID AutoTune for the bed, on a printer that is using the PS_ON pin to automatically power on the main power suply, results in a failure.

### My Configurations

I'm using an ATX power supply, with it's PS-ON cable (green-wire) connected to RAMPS PS-ON pin. 
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/5085000/ConfigFiles.zip)

### Steps to Reproduce

1. Use some kind of auto power on feature on the main PSU controlled by Marlin.
2. Set marlin configurations:
```
#define PIDTEMPBED
#define PSU_CONTROL
#define PSU_ACTIVE_HIGH false
#define PSU_DEFAULT_OFF 
#define AUTO_POWER_CONTROL
```
3. Run Bed PID AutoTune: M303 C10 D1 E-1 S95 U1
4. receive error:
```
Error:Heating failed, system stopped! Heater_ID: bed
Error:Printer halted. kill() called!
```

**Expected behavior:**
Turn PSU On, then run PID tune.

**Actual behavior:** [What actually happens]
Marlin keeps PSU OFF, and try to heat hotend or bed. This results in thermal runaway, as it is trying to heat but gets no increase in temperature read.

#### Additional Information
With my current configuration (using the auto power-on feature), I can set temperatures manually, and can print without any problems. Only the PID tune fails.
When I forced the PSU to power-on, disconnecting the PS-ON cable from RAMPS and connecting it to GROUND, then PID Tune was successful.
When running PID tune, the "target temperature" on the display, Pronterface and Octoprint stays at 0 (zero). I believe the target temperature is what causes the PSU to Power ON when I set the printer to pre-heat, configurable by `"#define AUTO_POWER_E_TEMP".`

### Possible Solutions:
One approach could be setting the "target temperature" thing, as this would cause the PSU to power on.
Another way would be to, before starting the PID tune procedures, check if PSU_CONTROL is enabled and then set the PS_ON pin.

GMagician, 2020-08-18:
Just a fast code change...if compile you may test it

pietchaki, 2020-08-19:
Hi, sorry for the delay,
I just got the latest Bugfix_2.0.x, builded it and uploaded to my board.
Now when I send M303, the PSU is turned ON. BUT the heater mosfet stays OFF.
Both for the extruder and bed. The Mosfet LEDs on the RAMPS board stays off, also the one in my bed (mk2b).
Again, I can heat them by sending M104 or M140 and all works normal.

I believe I correctly merged my settings into the new file, anyway heres the one I used in this test:
[Configuration_bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/5099889/Configuration_bugfix.zip)

Also, thanks for the update!

GMagician, 2020-08-20:
This sound really weird. Once you have power you should have the mosfet working. Code change is not "config" dependent so I doubt the problem is there.

boelle, 2020-08-20:
@pietchaki PR was merged just 2 days ago, did you download again after it was merged?

pietchaki, 2020-08-23:
Great!
It's working correctly now! 
Thank you!

boelle, 2020-08-23:
> It's working correctly now

when the PR is merged then click close if it still works

boelle, 2020-08-24:
@pietchaki does it still work? PR was merged

closed 2020-08-24 by thisiskeithb

pietchaki, 2020-08-24:
Sorry for the delay, I was testing the merged version.
All working fine here!

PR #19066 (2020-08-19): Fix PSU on when pid autotune
Should fix #19056
PR #19119 (2020-08-24): Move PSU on before setting heating pwm (PID autotune)
This should definitively fix #19056