MarlinFirmware/Marlin #19057: [BUG] (2.0.6) FILAMENT_RUNOUT_SENSOR/NOZZLE_PARK_FEATURE issue

wolfgangmauer, 2020-08-17:
### Bug Description

With a filament change, filament is unloaded and loaded, move back do original position and start nozzle heating.
This will ruin the print !!!

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5086146/Marlin.zip)
### Steps to Reproduce

1. Filament runout

**Expected behavior:** [What you expect to happen]
Go to park position, unload, load, **nozzle heating, return to print position**, continue printing.

**Actual behavior:** [What actually happens]
Go to park position, unload, load, **return to print position, nozzle heating**, continue printing.

thisiskeithb, 2020-08-17:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19033

thisiskeithb, 2020-08-17:
This behavior was changed in the latest `bugfix-2.0.x`.

closed 2020-08-17 by thisiskeithb