MarlinFirmware/Marlin #19060: [BUG] (short description)The mks_robin_pro cannot use AT24C32 eeprom, and DUAL_X_CARRIAGE power off continues to print abnormally

makerbase-mks, 2020-08-18:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-08-18:
Please don't mix two issues in one report. I suggest you split this so each issue has its own report.

It is very important to follow the issue template. Both of these issues need to include configuration files and steps to reproduce.

boelle, 2020-08-18:
> ### Steps to Reproduce
> 1. [First Step]
> 2. [Second Step]
> 3. [and so on...]

boelle, 2020-09-15:
template not followed

no response

closing

closed 2020-09-15 by boelle