MarlinFirmware/Marlin #19061: Delta printer UBL on round bed not probing all the bed [BUG?]

olafnew, 2020-08-18:
Good day. Have being fighting a problem for a while now.

I have BLTouch configured and it has really great repeatability setup, and have being using 3-point leveling since the beginning. I have decided to move to UBL, and have being fighting a problem ever since.

My setup: 
- correctly tuned and calibrated Delta, using BLTouch as my probe. 
- Probe offsets are correct and nozzle to table distance are 0.1mm at Z0 X0 Y0, but due to thermals, when the bed is heated i have height inconsistency within all printable radius, and thus i was hoping to resolve this with UBL.
- Latest Marlin Bugfix-2.0.x.

Below is my config for UBL:

Configuration.h
`
#define NOZZLE_TO_PROBE_OFFSET { -15.0, -10.0, -4.32 } //-17.0, -12.0

// Most probes should stay away from the edges of the bed, but
// with NOZZLE_AS_PROBE this can be negative for a wider probing area.
#define PROBING_MARGIN 15

// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 2500 // 80 * 80 8000

// Feedrate (mm/m) for the first approach when double-probing (MULTIPLE_PROBING == 2)
#define Z_PROBE_SPEED_FAST 60 * 80 //HOMING_FEEDRATE_Z

// Feedrate (mm/m) for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)

/**
 * Multiple Probing
 *
 * You may get improved results by probing 2 or more times.
 * With EXTRA_PROBING the more atypical reading(s) will be disregarded.
 *
 * A total of 2 does fast/slow probes with a weighted average.
 * A total of 3 or more adds more slow probes, taking the average.
 */
#define MULTIPLE_PROBING 5

// The size of the print bed
#define X_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)
#define Y_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)

#define AUTO_BED_LEVELING_UBL

#define RESTORE_LEVELING_AFTER_G28

#if ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_BILINEAR, AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT

  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)

  /**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  #define G26_MESH_VALIDATION
  #if ENABLED(G26_MESH_VALIDATION)
    #define MESH_TEST_NOZZLE_SIZE    0.4  // (mm) Diameter of primary nozzle.
    #define MESH_TEST_LAYER_HEIGHT   0.2  // (mm) Default layer height for the G26 Mesh Validation Tool.
    #define MESH_TEST_HOTEND_TEMP  190    // (°C) Default nozzle temperature for the G26 Mesh Validation Tool.
    #define MESH_TEST_BED_TEMP      60    // (°C) Default bed temperature for the G26 Mesh Validation Tool.
    #define G26_XY_FEEDRATE         20    // (mm/s) Feedrate for XY Moves for the G26 Mesh Validation Tool.
    #define G26_RETRACT_MULTIPLIER   1.0  // G26 Q (retraction) used by default between mesh test elements.
  #endif

#elif ENABLED(AUTO_BED_LEVELING_UBL)

//===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  //#define MESH_EDIT_GFX_OVERLAY   // Display a graphics overlay while editing the mesh

  #define MESH_INSET 5              // Set Mesh bounds as an inset region of the bed
  #define GRID_MAX_POINTS_X 6      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
  #define UBL_SAVE_ACTIVE_ON_M500   // Save the currently active mesh in the current slot on M500

  //#define UBL_Z_RAISE_WHEN_OFF_MESH 2.5 // When the nozzle is off the mesh, this value is used
                                          // as the Z-Height correction value.

`

Configuration_adv.h
`
// @section leveling

/**
 * Points to probe for all 3-point Leveling procedures.
 * Override if the automatically selected points are inadequate.
 */
#if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
  //#define PROBE_PT_1_X 15
  //#define PROBE_PT_1_Y 110
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 110
  //#define PROBE_PT_3_Y 20
#endif

/**
 * Probing Margins
 *
 * Override PROBING_MARGIN for each side of the build plate
 * Useful to get probe points to exact positions on targets or
 * to allow leveling to avoid plate clamps on only specific
 * sides of the bed. With NOZZLE_AS_PROBE negative values are
 * allowed, to permit probing outside the bed.
 *
 * If you are replacing the prior *_PROBE_BED_POSITION options,
 * LEFT and FRONT values in most cases will map directly over
 * RIGHT and REAR would be the inverse such as
 * (X/Y_BED_SIZE - RIGHT/BACK_PROBE_BED_POSITION)
 *
 * This will allow all positions to match at compilation, however
 * should the probe position be modified with M851XY then the
 * probe points will follow. This prevents any change from causing
 * the probe to be unable to reach any points.
 */
#if PROBE_SELECTED && !IS_KINEMATIC
  //#define PROBING_MARGIN_LEFT PROBING_MARGIN
  //#define PROBING_MARGIN_RIGHT PROBING_MARGIN
  //#define PROBING_MARGIN_FRONT PROBING_MARGIN
  //#define PROBING_MARGIN_BACK PROBING_MARGIN
#endif

#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X -155 // MESH_INSET
  //#define MESH_MIN_Y -155 // MESH_INSET
  //#define MESH_MAX_X 155 // X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y 155 // Y_BED_SIZE - (MESH_INSET)
#endif
`

**Problem:**
I then try to bed level: 
G28
G29 P1
And here is the main problem, please see video below. 
Probing starts, and i get "probing mesh point 1/36", and so on, till it hits point "Probing mesh point 17/36", which results in stopping of probing, and gives the following in terminal:
Recv: X:0.00 Y:0.00 Z:5.00 E:0.00 Count A:120340 B:120340 C:120340
Recv: ok
Send: M113 S2
Recv: ok
[...]

I've tried enlarging the mesh - and still i get a stop in the process of probing.

The main problem - IT DOES NOT probe all of the round bed. It probes just a small section of the bed, and stops. No matter what i do, set bed to probe offsets, try to tinker with probing margins - the result is consistently the same, the system probes just a section of the bed. In my opinion it tries to probe a **SQUARE** space in XY dimension, and as soon as it hits restricted space(diameter defined by DELTA_PRINTABLE_RADIUS var) it simply stops. And the logic behind mesh area and probing is designed only for square bed's. 

[https://youtu.be/GjOu2jR0vkk](url)

Please advise - how to properly configure UBL on delta printers with round beds...

olafnew, 2020-08-18:
Here is the debug output of M111 S247 (the problematic section)

Recv: deploy: 1
Recv: >>> Probe::run_z_probe  X84.00 Y-59.00 Z9.09
Recv: >>> Probe::probe_down_to_z  X84.00 Y-59.00 Z9.09
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z9.09
Recv: >  X84.00 Y-59.00 Z-1.68
Recv:   destination= X84.00 Y-59.00 Z9.09 : destination = current_position
Recv:   current_position= X84.00 Y-59.00 Z9.09 : xy move
Recv:   destination= X84.00 Y-59.00 Z-1.68 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z-1.68 : z lower move
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z-1.68
Recv:   current_position= X84.00 Y-59.00 Z4.03 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X84.00 Y-59.00 Z4.03
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z4.03
Recv: >  X84.00 Y-59.00 Z9.03
Recv:   destination= X84.00 Y-59.00 Z4.03 : destination = current_position
Recv:   destination= X84.00 Y-59.00 Z9.03 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z9.03 : z raise move
Recv:   current_position= X84.00 Y-59.00 Z9.03 : xy move
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z9.03
Recv: >>> Probe::probe_down_to_z  X84.00 Y-59.00 Z9.03
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z9.03
Recv: >  X84.00 Y-59.00 Z-1.68
Recv:   destination= X84.00 Y-59.00 Z9.03 : destination = current_position
Recv:   current_position= X84.00 Y-59.00 Z9.03 : xy move
Recv:   destination= X84.00 Y-59.00 Z-1.68 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z-1.68 : z lower move
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z-1.68
Recv:   current_position= X84.00 Y-59.00 Z4.07 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X84.00 Y-59.00 Z4.07
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z4.07
Recv: >  X84.00 Y-59.00 Z9.07
Recv:   destination= X84.00 Y-59.00 Z4.07 : destination = current_position
Recv:   destination= X84.00 Y-59.00 Z9.07 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z9.07 : z raise move
Recv:   current_position= X84.00 Y-59.00 Z9.07 : xy move
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z9.07
Recv: >>> Probe::probe_down_to_z  X84.00 Y-59.00 Z9.07
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z9.07
Recv: >  X84.00 Y-59.00 Z-1.68
Recv:   destination= X84.00 Y-59.00 Z9.07 : destination = current_position
Recv:   current_position= X84.00 Y-59.00 Z9.07 : xy move
Recv:   destination= X84.00 Y-59.00 Z-1.68 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z-1.68 : z lower move
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z-1.68
Recv:   current_position= X84.00 Y-59.00 Z4.08 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X84.00 Y-59.00 Z4.08
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z4.08
Recv: >  X84.00 Y-59.00 Z9.08
Recv:   destination= X84.00 Y-59.00 Z4.08 : destination = current_position
Recv:   destination= X84.00 Y-59.00 Z9.08 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z9.08 : z raise move
Recv:   current_position= X84.00 Y-59.00 Z9.08 : xy move
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z9.08
Recv: >>> Probe::probe_down_to_z  X84.00 Y-59.00 Z9.08
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z9.08
Recv: >  X84.00 Y-59.00 Z-1.68
Recv:   destination= X84.00 Y-59.00 Z9.08 : destination = current_position
Recv:   current_position= X84.00 Y-59.00 Z9.08 : xy move
Recv:   destination= X84.00 Y-59.00 Z-1.68 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y-59.00 Z-1.68 : z lower move
[...]
Recv: echo:busy: processing
Recv: <<< do_blocking_move_to  X84.00 Y-59.00 Z-1.68
Recv:   current_position= X84.00 Y-59.00 Z4.04 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X84.00 Y-59.00 Z4.04
Recv: >>> do_blocking_move_to  X84.00 Y-59.00 Z4.04
Recv: >  X84.00 Y-59.00 Z9.04
Recv:   destination= X84.00 Y-59.00 Z4.04 : destination = current_position
[... 138 lines cut ...]
Recv:   destination= X84.00 Y79.00 Z9.23 : destination = current_position
Recv:   destination= X84.00 Y79.00 Z19.32 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y79.00 Z19.32 : z raise move
Recv:   current_position= X84.00 Y79.00 Z19.32 : xy move
Recv: <<< do_blocking_move_to  X84.00 Y79.00 Z19.32
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv: >>> do_blocking_move_to  X84.00 Y79.00 Z19.32
Recv: >  X84.00 Y79.00 Z19.32
Recv:   destination= X84.00 Y79.00 Z19.32 : destination = current_position
Recv:   current_position= X84.00 Y79.00 Z19.32 : xy move
Recv: <<< do_blocking_move_to  X84.00 Y79.00 Z19.32
Recv: >>> do_blocking_move_to  X84.00 Y79.00 Z19.32
Recv: >  X84.00 Y79.00 Z5.00
Recv:   destination= X84.00 Y79.00 Z19.32 : destination = current_position
Recv:   current_position= X84.00 Y79.00 Z19.32 : xy move
Recv:   destination= X84.00 Y79.00 Z5.00 : prepare_fast_move_to_destination
Recv:   current_position= X84.00 Y79.00 Z5.00 : z lower move
Recv: echo:busy: processing
[...]
Recv: <<< do_blocking_move_to  X84.00 Y79.00 Z5.00
Recv: >>> do_blocking_move_to  X84.00 Y79.00 Z5.00
Recv: >  X0.00 Y0.00 Z5.00
Recv:   destination= X84.00 Y79.00 Z5.00 : destination = current_position

Vertabreak, 2020-08-18:
Based on the above noted configurations you have told marlin to block a total of 20mm on all sides of your bed reducing the entire probe area by 40mm this is a miss configuration.

olafnew, 2020-08-18:
> Based on the above noted configurations you have told marlin to block a total of 20mm on all sides of your bed reducing the entire probe area by 40mm this is a miss configuration.

Why 20mm? In my case it's #define PROBING_MARGIN 15, so 15mm and 30mm in total. It's just one of the settings i've tried during my search of remedying the situation. My bed is totaly ok with 3(!)mm margin.

My worst problem is that it probes HALF the bed. It does not probe the other half of the bed.

Vertabreak, 2020-08-18:
change #define PROBING_MARGIN 0 & #define MESH_INSET 0 possibly 1 if its way to close to the edge will allow it to probe the entire probable area a g29 p3 or p4 is still required for the unreachable area should get your moving in the correct direction.

olafnew, 2020-08-18:
> change #define PROBING_MARGIN 0 & #define MESH_INSET 0 possibly 1 if its way to close to the edge will allow it to probe the entire probable area a g29 p3 or p4 is still required for the unreachable area, this should get you moving in the correct direction. Based on the little data i have available at this time i would assume there are many more invalid settings present as well.

I did.. And it still stops at 17/36. Please see video - it's self explanatory (https://youtu.be/402xxAF_WGs)

Is there a way to re-define mesh generation into a round mesh? Using P4 is extremely inconvenient when interchanging printing surfaces :( It's way too labour intensive in real usage scenario

Vertabreak, 2020-08-18:
depending on the probe mount the larger the unprobable area will be.  
EXAMPLE" https://marlinfw.org/assets/images/features/bed_probe_areas.jpg

olafnew, 2020-08-18:
> depending on the probe mount the larger the unprobable area will be.
> EXAMPLE" https://marlinfw.org/assets/images/features/bed_probe_areas.jpg

That i understand, but my kinematic allows full reach of probe of all the bed. Though i do not really understand how to configure asymmetrical offset...

I still ask the question - is there a way to create circular mesh for probing?

Vertabreak, 2020-08-18:
there is a way to force UBL to probe all points that involves adding the probe offsets.
example:
#define MESH_MAX_X X_BED_SIZE - (MESH_INSET) - 40 //PROBE OFFSET X

it would be extremely unlikely for a feature to be entirely redone based on a users idea of how they think it should work without that user re-coding it and submitting the change fore review and even then it may not be accepted i have a bunch of code that was refused this is what cloning a repo is for.

olafnew, 2020-08-18:
> there is a way to force UBL to probe all points that involves adding the probe offsets.
> example:
> #define MESH_MAX_X X_BED_SIZE - (MESH_INSET) - 40 //PROBE OFFSET X
> 
> it would be extremely unlikely for a feature to be entirely redone based on a users idea of how they think it should work without that user re-coding it and submitting the change for review and even then it may not be accepted. i have a bunch of code that was refused this is what cloning a repo is for.

Yes, I understand that redoing a feature is highly unlikely.

define MESH_MAX_X X_BED_SIZE - (MESH_INSET) - 40 --- this mesh define really is not working well for circular buildplates. Could you advise where to dive in further for re-coding the algorithm for circular bed's?

Vertabreak, 2020-08-18:
"points you in the direction of /Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp"

boelle, 2020-09-11:
@olafnew tried what @Vertabreak suggested?

closed 2020-09-23 by github-actions[bot]