MarlinFirmware/Marlin #19065: SKR Pro v1.1 Pins file

Dsalo119, 2020-08-18:
Marlin 2.0.6 breaks sensorless homing for BigTreeTech SKR Pro v1.1.

**Expected behavior**
Sensorless homing should work

**Observed behavior**
Sensorless homing does not work

**Steps to reproduce**
Compiloe v2.0.6 for BTT SKR Pro v1.1 with TMC2209 drivers and sensorless homing

**Steps to work around**
From 2.0.5.4 to 2.0.6, BTT SKR Pro v1.2 was added, so now includes a common pins file:
"pins_BTT_SKR_PRO_Common.h"

If you replace this file with "pins_BTT_SKR_PRO_V1_1" from 2.0.5.4 and rename it to "pins_BTT_SKR_PRO_Common.h" then compile, you receive several warnings about pins already being defined, however sensorless homing works again.

thisiskeithb, 2020-08-18:
Please follow the issue template and attach your configs.

thisiskeithb, 2020-08-18:
I just built a basic SKR Pro 1.1 config with sensorless homing based on the latest `bugfix-2.0.x` & diag pins are defined correctly:

[image]

Dsalo119, 2020-08-18:
> Please follow the issue template and attach your configs.

files added in original message

thisiskeithb, 2020-08-18:
Your X min maps to `PB10` and Y min map to `PE12` in both configs, so sensorless homing is configured to use the correct pins.

Can you clarify what you mean by "Sensorless homing does not work"?

Also, after updating your firmware, did you Initialize EEPROM either through the menu or by sending `M502`, followed by `M500`?

Dsalo119, 2020-08-18:
EEPROM initialization fixed this.  I wasn't aware that pins data would be stored there.  Why would changing the pins file back to 2.0.5.4 version without initializing the EEPROM fix this if they are supposed to be using the same pins?

I'll close

closed 2020-08-18 by Dsalo119

thisiskeithb, 2020-08-18:
Pin mappings aren't kept in EEPROM, but other settings like `*_STALL_SENSITIVITY` are.

It's good to get in the habit of initializing EEPROM after each firmware update to ensure you're always pulling values from your new firmware & not old EEPROM settings.