MarlinFirmware/Marlin #19067: Filament change

redgod, 2020-08-18:
If when I activate the change of filament, it raises the nozzle it goes to a corner then it returns low but not to the layers that it should and is 2 above .......... marlin does not remember the correct layer height

boelle, 2020-08-18:
did you read the template? you are missing a lot of info

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

redgod, 2020-08-19:
sorry it's my first time posting

I upload the configuration I have with marlin 2.0.6, I think I have the configuration right, the filament change is done but at the moment of printing it does it 2 layers above

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5093466/Marlin.zip)

boelle, 2020-08-19:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

redgod, 2020-09-21:
I still can't get the filament switch to work, doesn't remember z height and ends up printing on air

qwewer0, 2020-09-21:
@redgod Check these two https://github.com/MarlinFirmware/Marlin/issues/19455, https://github.com/MarlinFirmware/Marlin/issues/18191 issues. Not a solution, but you could participate in those if you have the same problems.

redgod, 2020-11-28:
Sigo sin poder hacer funcionar el interruptor de filamento, no recuerdo la altura z y termina imprimiendo en el aire

qwewer0, 2020-11-28:
@redgod Could you provide detailed steps to reproduce the issue?

redgod, 2020-11-28:
Description
Steps to reproduce
change of filament during printing
retraction and bleeding of the filament well
does not remember the apposition of origin
Expected behavior
I expected a perfect filament change

Actual behavior
After purging, the nozzle does not remember the correct height and position, since it begins to print more in the center and leaving the external wall exposed without filament

[image]

[image]

qwewer0, 2020-11-28:
Change filament from the LCD or within the gcode with M600?

redgod, 2020-11-28:
from the LCD

qwewer0, 2020-11-28:
Try change filament (LCD or M600) two times in the same print.

My results:
**Left**: LCD change filament at the bottom and top, and two M600 in between. The first filament change got shifted in the Y axis, to the max direction.
**Right**: Only one M600, which also got shifted in the Y axis, to the max direction.
[image]

sjasonsmith, 2020-12-04:
@qwewer0 are you working on fixing this, or just testing to help verify there is a problem?

qwewer0, 2020-12-04:
@sjasonsmith Just tested it out to see if I can verify the problem.

Still waiting for @redgod to do two filament change in one print to see if I got the same problem in a different "outfit" or not.

ysoline, 2020-12-13:
Hello, I also have a problem with Z shifting when resuming printing. The X and Y are fine. But the Z I have at least 1min of gap between the initial layer and the new layer.

qwewer0, 2020-12-31:
Can't reproduce it anymore on the latest bugfix-2.0.x

@rhapsodyv Not sure what was my issue that caused my layer shift, so if I was the reason to add "`Bug: Confirmed !`" label, then I think that should be removed until further tests and information from @redgod and maybe @ysoline too.

redgod, 2021-01-31:
I still have the same problems, none of my ender 3 fields stick

Sebazzz, 2021-01-31:
To me it seems that sometimes the Z-offset is applied after the filament change. That is the only thing I can think of (CR-6 SE).

bartoszwas, 2021-02-13:
I got similar issue. I have installed filament runout sensor to my raspberry pi orctoprint. It seems to work fine but after filament change it starts to print a bit higher... Printer reports the same z value but still its a bit too high. I have also installed bed leveling sensor. Printer is Ender 3, Marlin 2.0.7.2.

[image]

tommymonk, 2021-03-17:
I'm also experiencing an occasional 1-2 layer discrepancy when it returns to print causing issues similar to above, it doesn't happen with every print though.
The z height numerically is correct but physically is wrong.

bartoszwas, 2021-03-18:
Check rolls on the z-axis, mine were fixed too thigh. This plus a bit loose on the trapezoid nut gives that result as you can see on picture above.

zemuro, 2021-03-25:
The same issue, filament broke, the head parked, went throuh filament change prcedure, printing resumed in the exact place, but 2-3 layers higher. Was so frustrated, forgot to notice exact heihgt %(

tommymonk, 2021-04-04:
Thanks @bartoszwas  you were spot on, adjusting the z axis roller nut has solved it

closed 2021-06-15 by github-actions[bot]