MarlinFirmware/Marlin #19068: [BUG] DIGIPOT_MCP4018 not working

caradas, 2020-08-18:
### Bug Description ###
M907 S0 doesn't reduce the motor current to 0

### My Configurations ###
Example Configs for Flashforge https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/FlashForge/Creator%202X

### Steps to Reproduce ###
Use Serial connection to send "M907 S0" and "G28"

**Expected behavior:** 
Homing failure/No Stepper movement

**Actual behavior:**
Normal movement

**Additional Information**
I tried to sniff the i2c bus with a logic analizer and found no signs of activity
I also temporary switched the DIGIPOTS_I2C_SCL to pin 13 to see if the Debug led is flashing when i send the m907 command but also nothing to see

boelle, 2020-08-18:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

caradas, 2020-08-19:
**[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5094527/Configuration.zip)** 
Sorry i thought it would be enough to post a link to the configs i used

boelle, 2020-08-19:
are you sure that you did not change just a little bit compared to the example config?

caradas, 2020-08-19:
Yes i am. 
Just to confirm it again: i downloaded marlin from github, put it in a new folder, added the example configs, opened it with vs code, compiled it and flashed it on the 2560 with atmel studio and an avr ice debugger.

closed 2020-09-26 by github-actions[bot]