MarlinFirmware/Marlin #19069: [BUG] Marlin 2.0.6 with Power Loss Recovery with Marlin_Dev_Mode

andreibobirica, 2020-08-19:
riinizia la stampa senza fare l'homing e quindi crede di essere già a 0/0, facendo questo stampa da tutt'altra parte, molte delle volte andando fuori dal piatto.

### My Configurations
https://github.com/andreibobirica/Marlin

### Steps to Reproduce
Activate Power Loss recovery
Activate Marlin_Dev_Mode 
lose power.
restart with power loss.

**Expected behavior:** [What you expect to happen]
Homing x/y, purge a little bit, restart printing from the last point.

**Actual behavior:** [What actually happens]
Start printing from another point.

#### Additional Information
This problem is due to the fact that Marlin_Dev_Mode, in addition to providing the possibility of giving additional commands, is not compatible with power loss recovery.
Going deeper into the code, I noticed that the G28 command with the Marlin_Dev_Mode has one more option, that is G28 S.
This option is automatically used in the power loss resume function.
In practice, homing is not done at the beginning of the print.
It is invoked:
TERN_ (MARLIN_DEV_MODE, "S") // Simulated Homing
Homing is simulated, without doing it.
Which makes it impossible to resume printing.

GMagician, 2020-08-19:
Maybe MARLIN_DEV_MODE is intented to be used during development stage and so "add" some weird things.
Why are you keeping it active?

andreibobirica, 2020-08-19:
> _DEV_MODE is intented to be used during development stage and so "add" some weird things.

I enabled it to try it out and to use one of the available commands, but I didn't think that enabling it would change pre-existing functions.

thinkyhead, 2020-08-26:
Marlin Dev Mode is not intended to be used for any real working printer situation. It is a "free space" for developers to do whatever is required to explore features that are under study. This is the first issue connected to the option that we have seen, but I may add a sanity-check to alert users that the option is enabled.