MarlinFirmware/Marlin #19071: [FR] Allow filament runout distance above 10mm through menus

Taomyn, 2020-08-19:
### Description

Using the menus I cannot set the filament runout distance greater than 10mm. I require it at 25mm so I am forced to use the "undocumented" M412 D25 command

### Feature Workflow

1. Got to advanced configuration in the menus, for Filament
2. Go up to higher value

#### Additional Information

I appreciate there needs to be a limit, perhaps expose this as a new configuration setting in the code? Also, someone should update the documentation of the M412 command.

[image]

GMagician, 2020-08-19:
Missing your configuration files so can't be sure but what is FILAMENT_RUNOUT_DISTANCE_MM value? edit seems to let you change from 1mm to 1.5 times such value.

Taomyn, 2020-08-20:
> Missing your configuration files so can't be sure but what is FILAMENT_RUNOUT_DISTANCE_MM value? edit seems to let you change from 1mm to 1.5 times such value.

No config files as it's not part of the PR template, but my current firmware value is 7mm as that is what BTT state is the best for their sensor, but in reality for my setup it works more reliably at 25mm - I think it's because I place my sensor before the extruder rather than their example of hanging on the tubing afterwards. I've changed my config now to 25 but haven't had the time to flash the update yet, but thought as I could not see how to set a maximum plus that it gets ignored by the M412 command, that it would be useful if you could.

GMagician, 2020-08-20:
> No config files as it's not part of the PR template

This is not a PR, more similar to a bug report, here because of cnf files.
However 7mm * 1,5 is 10,5 here is your edit limit.

> but thought as I could not see how to set a maximum plus that it gets ignored by the M412 command, that it would be useful if you could.

Not really sure to translate this correctly, but in gcode there is no limit

Taomyn, 2020-08-20:
> > No config files as it's not part of the PR template
> 
> This is not a PR, more similar to a bug report, here because of cnf files.

Sorry I miss-typed, now corrected, this is a feature request FR, and the template does not request adding any config files.

> However 7mm * 1,5 is 10,5 here is your edit limit.
> 
> > but thought as I could not see how to set a maximum plus that it gets ignored by the M412 command, that it would be useful if you could.
> 
> Not really sure to translate this correctly, but in gcode there is no limit

Yes, undocumented gcode, I am requesting something in the config that allows a limit to be set that the menu will use and not an arbitrary calculation like 1.5x which isn't useful.

GMagician, 2020-08-20:
FILAMENT_RUNOUT_DISTANCE_MM is the parameter you need to set to correctly handle runout, I think that 1.5 times such value is goot for adjustments. Why do you config a value that is wrong for your machine?

Taomyn, 2020-08-20:
I know that's the parameter I never asked for that, but 1.5x is not good for adjustments my real life example proves this and I was not the only one as this was how someone else ended up getting their sensor to work. I set the firmware parameter to 7mm because that is what the OEM of the sensor recommends. Yet when I tested my situation, it was far too short and so was the maximum 10mm in the menu. I have to either reflash the custom firmware, which others may not be able to do so easily, or use an undocumented command to override the maximum. Not a great end-user experience.

So I suggest either have a new config value that can override the useless 1.5x guess, or make it a more sensible value like 100 or better still simply the maximum the variable can allow depending on it's definition.

GMagician, 2020-08-20:
> I set the firmware parameter to 7mm because that is what the OEM of the sensor recommends

This is a little weird, it seems the OEM doesn't know how his sensor should work. If you and others need a bigger value then I suspect the suggested one is wrong

> So I suggest either have a new config value that can override the useless 1.5x

Then you ask add a new parameter to be modified in cnf that requires a new reflash? If a user is unable to change the current available parameter why should he be able to change the new one?

In my opinion the correct solution is to set FILAMENT_RUNOUT_DISTANCE_MM to the correct value. Once this is correct you may edit it in a range that is for sure valid. But I'm not a judge and not a "super" expert, let's see what they say...

Taomyn, 2020-08-20:
Yeah, the unfortunate thing is that OEM recommendation is for a stock extruder and for the sensor to be placed after the extruder on its output - and it just hangs in the air which I don't think is a good idea.

[image]

Whereas I have an aftermarket dual-gear extruder and I more sensibly properly fixed the sensor before the input of the extruder.

[image]

Archmonger, 2020-08-25:
I agree with OP's sentiment. Max value in the Marlin OSD should not be capped.

slowbro, 2021-01-19:
I put in a PR for updated docs for M412 here: https://github.com/MarlinFirmware/MarlinDocumentation/pull/368

For this, I think the easiest thing to do would be to set the max to `999` (max for float3). Since there is no limit applied to `M412` I don't see why one should be applied to the menu. I'll work on a PR for that tomorrow.

slowbro, 2021-01-21:
PR #20828 was merged into bugfix-2.0.x which should resolve this.

Taomyn, 2021-01-23:
Thank-you, compiled and flashed to my printer and confirm works.

closed 2021-01-23 by Taomyn