MarlinFirmware/Marlin #19072: [FR] AUX Leveling / Z Offset Tool

PhilippMolitor, 2020-08-19:
### Description

I really would like to see AUX-Leveling and Z-Offset submenus (in my case, for the Ender 3 v2) such as with the modded Marlin firmware by Smith3D: [https://smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/](https://smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/
)
Seems like they are not sharing their changes to the Marlin firmware in source code form.. (isn't this a GPL 3 license violation?)

### Feature Workflow

See attached Link for screenshots and more info.

javierpal, 2020-09-01:
it would be good to have that kind of Feature, it's soo good in my ender 3.

xcasxcursex, 2020-12-05:
https://github.com/MarlinFirmware/Marlin/issues/18096 

Will make this possible via configuration files (Just use the gcode from the link on the smith3d page above)

This link shows a method to do it otherwise: https://3dprinting.stackexchange.com/questions/7676/how-to-add-menu-options-to-the-marlin-firmware-lcd-menu

peta, 2021-01-17:
@PhilippMolitor For sure, Smith3d is sharing their custom version in source code form. We can find their DWIN modifications here: https://github.com/smith3d/Marlin/tree/bugfix-2.0.x-Smith3D/Marlin/src/lcd/dwin/e3v2

xcasxcursex, 2021-01-18:
Why thumbs down? I've given you a workaround and pointed you to the fix.

LekeFly, 2021-02-16:
Would love to see this function, right now i can't figure out how to set my Z-offset without it. It's in the - range.

xcasxcursex, 2021-02-17:
>  right now i can't figure out how to set my Z-offset without it.

Look at the GCode above

slowbro, 2021-07-03:
This was solved by #18177, as xcasxcursex mentioned there, and can be closed.

closed 2021-07-03 by ellensp