MarlinFirmware/Marlin #19073: [BUG] Filament runout when cold extruder

GMagician, 2020-08-19:
I faced a situation where printer is starting to print, filament is missing, but hotend is still cold.

Runout script is properly executed but it's aborted with a "hotend cold" error message, this error stop runout handling but the printer start to print.

Cause is 'filament_ran_out' variable, it is set once script commands are enqueued but none check if they are successful or not, then, in this case, M600 is aborted and never executed anymore...

My first approach may be to call a "reset" if M600 is aborted but another is to raise event only when extruder is really moving.

Any hint about the best solution?

boelle, 2020-08-19:
i would call this a bug, 

it should heat up first and then do the runout script, but that is just my common sense logic, not sure if others do agree

GMagician, 2020-08-19:
Yes but it's a bug to me too, the problem is that M600 is enqueued but tagged as executed. Since action is aborted it's never executed anymore. Question is what's the best solution?

I think that runout may be senseless if extruder is not "moving", but I prefer to listen to "big" developpers thinking

GMagician, 2020-08-19:
@boelle and yes, another solution may be to raise the event only when extruder "hot" enough. In such case action is not aborted and everything will be ok

qwewer0, 2020-09-07:
https://github.com/MarlinFirmware/Marlin/pull/19298 solves my problem in https://github.com/MarlinFirmware/Marlin/issues/15425 issue, but here are some observation that I made while testing it:

If there is no bed heating or the bed is already on the desired temp before nozzle heating, then there won't be any `M600: Too Cold` notification, but if there is bed heating before nozzle heating, then there is `M600: Too Cold` notification.
And there is still a non working FilamentRunout before the actual working FilamentRunout that the fix introduces.
<details>
  <summary>No bed heating before nozzle heating</summary>

```
READ: //action:notification Ender-3 Ready.
READ: echo:Now fresh file: shape-~1.gco
Now fresh file: shape-~1.gco
READ: File opened: shape-~1.gco Size: 38183
READ: File selected
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:resume
READ: //action:prompt_end
READ: //action:prompt_begin Resuming SD
READ: //action:prompt_button Dismiss
READ: //action:prompt_show
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused filament_runout 0
READ: X:162.2000 Y:119.5000 Z:7.1700 E:0.0000 Count X:12976 Y:9560 Z:2868
READ: //action:paused
...
```
</details>

<details>
  <summary>With bed heating before nozzle heating</summary>

```
READ: //action:notification Ender-3 Ready.
READ: echo:Now fresh file: shape-~1.gco
Now fresh file: shape-~1.gco
READ: File opened: shape-~1.gco Size: 38182
READ: File selected
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:resume
READ: //action:prompt_end
READ: //action:prompt_begin Resuming SD
READ: //action:prompt_button Dismiss
READ: //action:prompt_show
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Bed Heating...
READ:  T:111.28 /0.0000 B:32.92 /40.0000 @:0 B@:0 W:?
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused filament_runout 0
READ:  T:110.56 /0.0000 B:32.92 /40.0000 @:0 B@:0 W:?
...
READ:  T:82.43 /0.0000 B:40.88 /40.0000 @:0 B@:115 W:0
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:paused
READ: //action:notification M600: Too Cold
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused filament_runout 0
READ: X:162.2000 Y:119.5000 Z:7.1700 E:0.0000 Count X:12976 Y:9560 Z:2868
READ: //action:paused
...
```
</details>

thinkyhead, 2020-09-07:
The correct solution would be for `M600` to see that the temperature is too low and wait for the nozzle to heat up. It should also do this in general usage, so for debugging I assume we can just start up the machine, home with `G28` and send `M600` while the nozzle is cold, and keep refining until it works properly.

thinkyhead, 2020-09-08:
Ok, I combined both approaches. It's only sensible that the first step of `M600` should be to heat up the nozzle, and as long as it puts up the usual prompts it shouldn't be a surprise to anyone when it does.

qwewer0, 2020-09-08:
Still get two FilamentRunout, with `M600: Too Cold` on the second one if there is bed heating before the nozzle heating, but without bed heating or the bed being at the right temperature or over it, now only gives a single FilamentRunout, so that is an improvement.

<details>
  <summary>with bed heating before nozzle heating</summary>

```
READ: //action:notification Ender-3 Ready.
READ: echo:Now fresh file: shape-~1.gco
Now fresh file: shape-~1.gco
READ: File opened: shape-~1.gco Size: 38182
READ: File selected
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:resume
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Bed Heating...
READ:  T:20.62 /0.0000 B:22.50 /40.0000 @:0 B@:0 W:?
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused filament_runout 0
READ:  T:20.62 /0.0000 B:22.19 /40.0000 @:0 B@:127 W:?
...
READ:  T:20.62 /0.0000 B:41.03 /40.0000 @:0 B@:114 W:0
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:paused
READ: //action:notification M600: Too Cold
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused filament_runout 0
...
```
</details>

GMagician, 2020-09-09:
@qwewer0 what is the issue when 2 runout are raised? Have you tryed merged code?

qwewer0, 2020-09-09:
My last respond was with the new merged modifications.

qwewer0, 2020-09-09:
Just so that I understand it correctly.
Isn't the ultimate goal is to have the filament runout script trigger on the first appearance? (highlighted in bold)

> **READ: //action:prompt_end
> READ: //action:prompt_begin FilamentRunout T0
> READ: //action:prompt_show
> READ: //action:paused filament_runout 0**
> READ:  T:20.62 /0.0000 B:22.19 /40.0000 @:0 B@:127 W:? [...]

GMagician, 2020-09-09:
My original goal was not to miss it, even if raised more than once due to cold extruder. @thinkyhead changed it to let it heat extruder in such case but I didn't tested it. I don't know why you get failing when heating bed. Bed should not be considered in this case

GMagician, 2020-09-09:
I checkd code. I suspect @thinkyhead  is not working as expected. safe temp is done on "load" but cold is raised on "unload"

GMagician, 2020-09-10:
I checked code. Cold error is raised inside `pause_print` function, while added code works in unload and load filament...I'll post a fix

GMagician, 2020-09-10:
@qwewer0 may you test new PR?

qwewer0, 2020-09-10:
> My original goal was not to miss it, even if raised more than once due to cold extruder. @thinkyhead changed it to let it heat extruder in such case but I didn't tested it. I don't know why you get failing when heating bed. Bed should not be considered in this case

I think I missed a few information. The fix did not cause the first filament runout script to be missed, because that bug was already in marlin https://github.com/MarlinFirmware/Marlin/issues/15425
Will test the new changes in a min.

qwewer0, 2020-09-10:
Using the new changes, there is no `M600: Too Cold`  no matter if there is bed heating or not before the nozzle heating.
And there is now only one filament runout notification. Nice job!
<details>
  <summary>No bed heating before nozzle heating</summary>

```
READ: //action:notification Ender-3 Ready.
READ: echo:Now fresh file: shape-~1.gco
Now fresh file: shape-~1.gco
READ: File opened: shape-~1.gco Size: 38182
READ: File selected
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:resume
READ: //action:prompt_end
READ: //action:prompt_begin Resuming SD
READ: //action:prompt_button Dismiss
READ: //action:prompt_show
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Bed Cooling...
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused//action:paused
```
</details>

<details>
  <summary>With bed heating before nozzle heating</summary>

```
READ: //action:notification Settings Stored
READ: //action:notification Ender-3 Ready.
READ: echo:Now fresh file: shape-~1.gco
Now fresh file: shape-~1.gco
READ: File opened: shape-~1.gco Size: 38182
READ: File selected
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:resume
READ: //action:prompt_end
READ: //action:prompt_begin Resuming SD
READ: //action:prompt_button Dismiss
READ: //action:prompt_show
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:notification Bed Heating...
READ:  T:20.62 /0.0000 B:20.62 /40.0000 @:0 B@:0 W:?
READ: //action:prompt_end
READ: //action:prompt_begin FilamentRunout T0
READ: //action:prompt_show
READ: //action:paused filament_runout 0
READ:  T:20.62 /0.0000 B:20.62 /40.0000 @:0 B@:127 W:?
...
READ:  T:20.62 /0.0000 B:40.88 /40.0000 @:0 B@:115 W:0
READ: //action:notification Shape-Cylinder_7-12.40.gcode
READ: //action:paused
```
</details>

qwewer0, 2020-09-10:
Should we be concerned about the that one with bed heating has `//action:paused filament_runout 0`, but not the other?

GMagician, 2020-09-10:
@qwewer0 can you post first lines of your gcode (with bed heating and not) so I can do tests.
What I don't like is

> READ: //action:paused//action:paused

qwewer0, 2020-09-10:
> can you post first lines of your gcode (with bed heating and not) so I can do tests.

For no bed heating I got the same results (filament runout script count) if the bed temp was already at that temp or I removed it from the gcode, but if I removed the bed heating from the gcode, then before the filament runout script it homes the printer, so it runs the next line after the nozzle heating gcode, and this doesn't happen if there is a bed heating before the nozzle heating. (Forgot to mention this before)
M75 did not make any difference, if I placed it at the start of my start gcode.

<details>
  <summary>Gcode</summary>

; generated by SuperSlicer 2.2.52 on 2020-09-07 at 10:40:01 UTC

; 

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.40mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.40mm

; object:{"name":"Shape-Cylinder","id":"Shape-Cylinder id:0 copy 0","object_center":[116.000000,115.000000,0.000000],"boundingbox_center":[116.000000,115.000000,0.100000],"boundingbox_size":[35.000000,35.000000,0.200000]}

M73 P0 R2
M107
; Ender 3 PrusaSlicer Custom Start G-code
M412 S1	; Enable Filament Runout Sensor
M220 S100	; Reset Feedrate
M221 S100	; Reset Flowrate
M107	; Turn-off fan
M190 S40	; Set Heated Bed temperature, and wait for heat up
M104 S200		; Set Extruder temperature, but do not wait for heat up
G28	; Home all axes
</details>

> What I don't like is
> 
> > READ: //action:paused//action:paused

Yeah, I saw that too, I think https://github.com/MarlinFirmware/Marlin/issues/18854 is the issue for that.

GMagician, 2020-09-10:
@qwewer0 

> For no bed heating I got the same results (filament runout script count) if the bed temp was already at that temp or I removed it from the gcode, but if I removed the bed heating from the gcode, [...]

not sure to understand that, I think posted gcode is too short to test behaviour, also I don't have "action host" active so I'm not sure I can test it.

What I now fixed in my PR is something I saw on my machine. PID is calibrated at 270°C then when heated at 175 (min temp) it has some overshoot/undershoot. In undershoot case it doesn't load filament.

GMagician, 2020-09-10:
> Yeah, I saw that too, I think #18854 is the issue for that.

Yes, same issue, then I don't care of it here

qwewer0, 2020-09-10:
> I think posted gcode is too short to test behaviour

@GMagician Don't know if it helps anymore, but here is the full gcode.
[Shape-Cylinder_7-12.40.zip](https://github.com/MarlinFirmware/Marlin/files/5203535/Shape-Cylinder_7-12.40.zip)

GMagician, 2020-09-10:
@qwewer0 

> Should we be concerned about that the one with bed heating has //action:paused filament_runout 0

no, this is raised on filament runout detection (code below), we must check why you don't have it when bed is not heating

```c++
  #if ENABLED(HOST_ACTION_COMMANDS)
    if (run_runout_script
      && ( strstr(FILAMENT_RUNOUT_SCRIPT, "M600")
        || strstr(FILAMENT_RUNOUT_SCRIPT, "M125")
        #if ENABLED(ADVANCED_PAUSE_FEATURE)
          || strstr(FILAMENT_RUNOUT_SCRIPT, "M25")
        #endif
      )
    ) {
      host_action_paused(false);
    }
    else {
      // Legacy Repetier command for use until newer version supports standard dialog
      // To be removed later when pause command also triggers dialog
      #ifdef ACTION_ON_FILAMENT_RUNOUT
        host_action(PSTR(ACTION_ON_FILAMENT_RUNOUT " T"), false);
        SERIAL_CHAR(tool);
        SERIAL_EOL();
      #endif

      host_action_pause(false);
    }
    SERIAL_ECHOPGM(" " ACTION_REASON_ON_FILAMENT_RUNOUT " ");
    SERIAL_CHAR(tool);
    SERIAL_EOL();
  #endif // HOST_ACTION_COMMANDS
```

GMagician, 2020-09-10:
@qwewer0 I tested my last PR with your gcode and with or without bed heating I see:

> 19:37:34.002 : //action:paused filament_runout 0
a delay (heating nozzle)
> 19:38:53.228 : //action:paused

qwewer0, 2020-09-10:
> we must check why you don't have it when bed is not heating

> I tested my last PR with your gcode and with or without bed heating I see:
>> 19:37:34.002 : //action:paused filament_runout 0
>> a delay (heating nozzle)
>> 19:38:53.228 : //action:paused

@GMagician I want to correct myself:

- With bed heating before nozzle heating I get `//action:paused filament_runout 0` without the printer being homed.
- Without bed heating (with M190 commented out) before nozzle heating I get `//action:paused filament_runout 0` with the printer being homed.
- With bed heating, but the bed temperature is higher than the set temp, so with `//action:notification Bed Cooling...` I don't get `//action:paused filament_runout 0` on the output and the printer isn't homed.

I'm sorry that I overlooked these things.

GMagician, 2020-09-10:
First two may be correct and I don't care of them. I didn't check who is raising runout event, importan thing is it is raised.
If home is not executed then filament change should be done on site (if I'm not wrong)
What happens with 3rd case? For sure you have to wait for bed colling before proceeding. What happens when bed is cooled down?

GMagician, 2020-09-10:
@qwewer0 when do you get the double //action:paused? I want to take a look at it

qwewer0, 2020-09-10:
> @qwewer0 when do you get the double //action:paused? I want to take a look at it

"With" `//action:notification Bed Cooling...` I get double `//action:paused`, but it still just parks and waits for nozzle heating as in the other cases. Don't know if it really called two times or if it is just a double notification.

GMagician, 2020-09-10:
Ok, then my PR fixes correctly al cases, only dount may be the double "paused" but may be covered by another PR

qwewer0, 2020-09-10:
Here is a new(?) problem and this might not be the right place to share it. If there is bed heating, then the filament unload is done at the `EXTRUDE_MINTEMP` 170°C, while on all the other cases it heats to the set (M104 S200) 200°C.

1. No bed heating -> Printer homed (G28 is the next line after M104) -> Nozzle parked -> Nozzle heating to the set 200°C -> Filament unloading -> `//action:prompt_begin Nozzle Parked`
2. Bed heating -> Bed heated to set temp. -> Nozzle heating to 170°C -> Filament unloading -> `//action:prompt_begin Nozzle Parked`
3. Bed "cooling" -> Nozzle heating -> Nozzle heating to the set 200°C -> Filament unloading -> `//action:prompt_begin Nozzle Parked`

In the 1. case if I switch out the next (`G28`) line after the nozzle heating to `M300 S2000 P5`, then there is no homing and parking, instead the M300 is played.

Shout we be concerned that no parking is done, while it says `//action:prompt_begin Nozzle Parked` after it unloaded the filament or is this normal as there was no homed axis?

GMagician, 2020-09-10:
> Shout we be concerned that no parking is done, while it says //action:prompt_begin Nozzle Parked after it unloaded the filament or is this normal as there was no homed axis?

When axis not home then filament change is done on site, that's what code does

> Nozzle heating to 170°C

if filament runout is detected before hotend heating, then mintemp is used, this is the fix @thinkyhead did.

I don't know now if mintemp is correct or not, I'm thinking at policarbonate etc, but in such cases maybe mintemp is configured higher

GMagician, 2020-09-10:
@qwewer0 I think all issues related to this issue have been fixed. I'm going to close it, do you agree?

qwewer0, 2020-09-10:
Is there any reason why would it heat to 170°C if the initial bed temperature is below the set temperature, but it heats to 200°C if the initial bed temperature is higher than the set bed temperature? And why would it execute the next gcode (G28 or M300 in my test) line when there is no bed heating? Does the two has to do something with each other?

If you say that this is way too far from this issues base problem, then maybe a new issue is needed for this.

qwewer0, 2020-09-10:
> I think all issues related to this "issue" have been fixed. I'm going to close it, do you agree?

If the problem in my last response isn't related to this issue, then I think this can be closed, and maybe https://github.com/MarlinFirmware/Marlin/issues/15425 too.

GMagician, 2020-09-10:
There is no relationship between bed temperature and filament change temp. All depends on when runout is detected and where M600 is put into queue. If this happens after M104/109 then filament change is done at programmed temp. If M600 is enqueued before then desired temp is unknown and Marlin uses "mintemp".
I think, but not checked, that when temp is below code that wait calls something that raises the "runout" event. When temp is above such call is not done and something else will raise event, maybe in extruder wait temp
If I'm right with above statement, if you change your "M190 S40" to "M190 R40" you will see that extruder is always heated at 170

qwewer0, 2020-09-10:
With `M190 R40`, sure enough, it is heating the nozzle to 170°C.

GMagician, 2020-09-10:
Then it's some "idle" call when waiting temp that probably raises "runout" event. Can I close?

qwewer0, 2020-09-10:
Yes, don't wait for me.

PR #19332 (2020-09-10): Fix too cold in m600
Followup recent fixes to fix #19073.
closed 2020-09-10 by GMagician