MarlinFirmware/Marlin #19074: [BUG] Laser sometimes stays on when issuing M112

Sebbb, 2020-08-19:
Hello,

I use LaserWeb4 to engrave. When I cancel a job, LaserWeb4 issues M112, all steppers stop instantly, the software halts, but sometimes the laser just stays on (SPINDLE_LASER_ENA_PIN, which I set to P2_07, normally used for HEATER_0, stays powered). It only shuts down when resetting the board.

I have a BTT SKR 1.4 Turbo board. 

When I power up the laser and issue M112 manually, it stops as expected.

I'm completely lost why that happens, the code looks okay. I even added the code to switch it off again in the kill function, but that does not help either.

```
diff --git a/Marlin/src/MarlinCore.cpp b/Marlin/src/MarlinCore.cpp
index 30b6fbe3d..a5b5b7ece 100644
--- a/Marlin/src/MarlinCore.cpp
+++ b/Marlin/src/MarlinCore.cpp
@@ -824,7 +824,11 @@ void minkill(const bool steppers_off/*=false*/) {
   #if HAS_KILL

     // Wait for kill to be released
-    while (kill_state()) watchdog_refresh();
+    while (kill_state()) {
+      watchdog_refresh();
+      cli();
+      WRITE(SPINDLE_LASER_ENA_PIN, !SPINDLE_LASER_ACTIVE_HIGH);
+    }

     // Wait for kill to be pressed
     while (!kill_state()) watchdog_refresh();
```

I run a recent checkout of bugfix-2.0.x.

Any ideas how to solve this? Is there anything that could prevent WRITE from shutting down the pin in this state?

Steps to reproduce:
- Engrave with LaserWeb4
- click "abort job"

Sebastian

GMagician, 2020-08-19:
> **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

GMagician, 2020-08-19:
Maybe you don't have HAS_KILL so your code is not executed

GMagician, 2020-08-19:
Don't know how many laser cnf are present and how they wrok but I think spindle enable pin must be reset inside `set_enabled` when `enable` argument is false.

GMagician, 2020-08-19:
Please test PR

Sebbb, 2020-08-19:
@GMagician 

HAS_KILL is set.

See 
https://github.com/Sebbb/MarlinCR10S/tree/bugfix-2.0.x-laser
for my configuration files and the complete software I just tested with.

It behaves quite strange: When I click "abort", the laser goes off for a few seconds, and then turns on again. The board did not reboot in the background, it stays like this until I reset it.

I created a video of it, there you see 2 green LEDs on the board: The left one is PWM (inverted), the right one is power for the laser.

https://youtu.be/LXE5x8CqMl4

You can see, as soon as I issue M112, the right LED goes off. And a few seconds later, it goes on again. Also the left LED still flickers, as if it still changes its values, but all interrupts should be disabled at this stage..

GMagician, 2020-08-19:
> It behaves quite strange: When I click "abort", the laser goes off for a few seconds, and then turns on again.

I read something about this here: https://marlinfw.org/docs/configuration/laser_spindle.html

> It is highly recommended that an external 1k-10k pullup resistor be connected to the SPINDLE_LASER_ENABLE_PIN. This will prevent the spindle/laser from powering on briefly during power up or when the controller is reset (which happens whenever you connect or disconnect from most controllers).

Maybe this is the cause?

Sebbb, 2020-08-19:
No. As said, I use the pin normally used for the heater, this also doesn't power up during boot. Also, after switching on, the laser stays off until I actively switch it on.

GMagician, 2020-08-20:
That's really weird. Interrupt are disabled and board may reset but it seems it doesn't, as I can see from your video seems pwm still working when laser is turned on again

Sebbb, 2020-08-20:
Indeed. I adjusted my debug code a bit:

```
  #if HAS_KILL
    cli();
    while(true) {
      WRITE(BEEPER_PIN, HIGH);
      delay(100);
      WRITE(BEEPER_PIN, LOW);
      delay(900);
      #if ENABLED(USE_WATCHDOG)
        watchdog_refresh();
      #endif
    }
...
```

So I know exactly where it is, that interrupts are switched of and nothing else could change the status except the watchdog refresh.

You can see it started beeping, and then the LED goes on again, and it continues beeping.

https://youtu.be/CXkAN7d1Ll4

So, the only code called is watchdog_refresh();

Then I disabled the watchdog, `//#define USE_WATCHDOG` in Configuration_adv.h, but this does not help either.

I disconnected the laser, so there is no external power that could lighten up the LED without control of the board.

BTW, that the laser stays on happens only every third or fourth try.

So, as long as the board doesn't support multitasking, I have no idea anymore.

GMagician, 2020-08-21:
> I have no idea anymore

Me too

thinkyhead, 2020-10-04:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-11-22:
@Sebbb did you what @thinkyhead suggested?

Sebbb, 2020-11-27:
@boelle not yet unfortunately, currently I don't use the laser.. As soon as take it out again, I'll test it...

If you don't want the issue to stay open it can be closed, I will reopen it then.

closed 2021-01-04 by github-actions[bot]