MarlinFirmware/Marlin #19075: #define Z_MIN_PROBE_REPEATABILITY_TEST does not work without Z_SAFE_HOMING

qystan, 2020-08-19:
### Bug Description

The Z_MIN_PROBE_REPEATABILITY_TEST does not run if #define Z_SAFE_HOMING is not activated.  
I am using an Allen-key probe.  In spite of the above, Bi-linear Bed-Levelling runs fine.

### Steps to Reproduce
Compile with #define Z_SAFE_HOMING commented out and run PROBE_REPEATABILITY_TEST on the menu.

**Expected behavior:** [What you expect to happen]
The probe should move to the center of the bed and perform the repeatability test (or any position over the bed but center should be easier to define).

**Actual behavior:** [What actually happens]
It will perform an Auto-Home to Xmin, Ymin and Zmin (front left corner) and move Z up to the park position and end.  There are no no status or error messages.

Additional information:
With Z-SAFE_HOMING active, the test runs auto-home to Xmin, Ymin, Zmin and then positions the nozzle at the center of the bed and performs the repeatability test.

Reason I prefer home to be at the front left, outside of the bed is to avoid filament drips on the bed while the hot-end is heating up.

Configuration files with (Z_SAFE_HOMING deactivated) is attached.  
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5097086/config.zip)

GMagician, 2020-08-19:
What if you home manually (G28) and then move to center and once there you ask for test?

qystan, 2020-08-19:
It will do an auto-home to the left front corner and stop. 

> What if you home manually (G28) and then move to center and once there you ask for test?

It will do the same. Auto-home to the left front corner and stop.

GMagician, 2020-08-19:
Are you using bugfix? M48 seems doesn't home automatically, and if axis are not homed it raise an error.
When called if does test without moving, so may raise an error if "position" is not reachable (not checked what this may really check)

GMagician, 2020-08-19:
I changed my cnf to remove Z_SAFE_HOME. I confirm you that M48 execute test where X & Y are. If G28 has not been done before then an error is sent on serial. How did you launch M48? What is the exact sequence you're doing?

qystan, 2020-08-19:
Your result matches M48.cpp documentation.

 I did M48 launched from the MOTION menu M48 PROBE TEST option. I have not tried sending M48 directly.

qystan, 2020-08-20:
Amended menu.cpp as follows did the trick:

From this:
  #if ENABLED(Z_MIN_PROBE_REPEATABILITY_TEST)
    GCODES_ITEM(MSG_M48_TEST, PSTR("G28\nM48 P10 X100 Y100"));
  #endif

To this which positions the nozzle/probe to the center of the bed:
  #if ENABLED(Z_MIN_PROBE_REPEATABILITY_TEST)
    GCODES_ITEM(MSG_M48_TEST, PSTR("G28\nG0 Z5\nM48 P10 X100 Y100"));
  #endif

GMagician, 2020-08-20:
In my source code lcd menu item is:
```c++
#if ENABLED(Z_MIN_PROBE_REPEATABILITY_TEST)
    GCODES_ITEM(MSG_M48_TEST, PSTR("G28\nM48 P10"));
#endif
```

With it M48 will be done in home position, but since probe is outside the real area I think it's aborted.

I think it's not correct as menu item. It should only be M48, but maybe you may open a new issue with it

qystan, 2020-08-20:
Sorry, it should read as:

Amend menu.cpp as follows:

From this:
#if ENABLED(Z_MIN_PROBE_REPEATABILITY_TEST)
GCODES_ITEM(MSG_M48_TEST, PSTR("G28\nM48 P10"));
#endif

To this which positions the nozzle/probe to the center of the bed:
#if ENABLED(Z_MIN_PROBE_REPEATABILITY_TEST)
GCODES_ITEM(MSG_M48_TEST, PSTR("G28\nG0 Z5\nM48 P10 X100 Y100"));
#endif

GMagician, 2020-08-20:
I think you may open an issue for such case. I think that a lot of printers with probe on nozzle left will suffer this and some sort of fix must be done

qystan, 2020-08-21:
Thanks for the discussion.  

It helped me find menu.cpp to understand the feature's implementation (my cpp is only rudimentary).

Will raise an issue.

closed 2020-08-21 by qystan