MarlinFirmware/Marlin #19077: [BUG] Executing T0 or T1 causes the bed to move and the tool head to change positions

philm001, 2020-08-19:
Hello all,

Here is the issue, when I execute a T0/T1 command, the bed raises itself and rams into the nozzles (I have a dual extruder). There have been some cases where the T0/T1 command causes the extruder head to move as it is touching the bed.

Instead of talking about my configuration, here is the GCode that causes it to happen:

```
;Basic settings: Layer height: {layer_height} Walls: {wall_thickness} Fill: {fill_density}
;Print time: {print_time}
;Filament used: {filament_amount}m {filament_weight}g
;Filament cost: {filament_cost}
M42 P6 S255 ; Turn on LEDs
G21                          ; metric values
G90                          ; absolute positioning
M82                          ; set extruder to absolute mode
M107                         ; start with the fan off
G28 F200                     ; home all
M109 T0 S240.000000         ; heat first nozzle for auto leveling
M109 T1 S240.000000         ; heat second nozzle for auto leveling
T0                             ; switch to first nozzle and retract filament
G1 F100
G92 E0
G1 E-20 F100
T1                            ; switch to second nozzle and retract filament
G92 E0
G1 E-20 F100
T0                           ; switch back to first nozzle for auto leveling
G12
M204 S300                    ; set accel for probing
;M203 Z4                      ; set z max speed for probing
G29                          ; autolevel
M204 S2500                   ; set accel back to normal
M203 Z20                     ; set z max back to limit
G1 Z5 F200                   ; raise nozzle up 5mm
;M104 T0 S0                  ; cool first nozzle after leveling
;M104 T1 S0                  ; cool second nozzle after leveling
T1                           ; extrude filament back into second nozzle
G92 E0
G1 E20 F100
G92 E0
T0                           ; extrude filament back into first nozzle
G92 E0
G1 E20 F100
G92 E0
M400                         ; clear buffer
G4 S1                        ; pause
G1 X20 Y200 F6000
G1 F{travel_speed}
M117 AXIOM Dual Primary Printing...        ;Put printing message on LCD screen
```

In case you are wondering, I am working with the Airwolf Axiom printers
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5097277/Configuration.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5097280/Configuration.zip)

Run the GCode in order

**Expected behavior:** I am expecting the bed to not move when a tool change occurs

**Actual behavior:** Bed and sometimes the toolhead moves

philm001, 2020-08-19:
Ok, so I found a define statement that was the culprit:

`TOOLCHANGE_NO_RETURN`

And the comment:

`// Never return to previous position on tool-change`

So the axiom printer does not have a magnetic toolchange or anything fancy like that. Maybe we can add in another comment or define it better because it was not intuitive when I read it. Previous position like position of when the head was printing? Where the last auto-home was? What is previous position.

If I am printing a piece that uses the dual extruders, will it still print properly? Will the tool head move to some random location when it is actually printing? The comment worries me because I think that the printer will not pick up right at the stop the T0/T1 command is issued.

boelle, 2020-09-11:
> Ok, so I found a define statement that was the culprit:
> 
> `TOOLCHANGE_NO_RETURN`
> 
> And the comment:
> 
> `// Never return to previous position on tool-change`

closing as it was a config error

can always reopen if its not the case

closed 2020-09-11 by boelle