MarlinFirmware/Marlin #19081: Incorrect Bed size

Styg1an, 2020-08-19:
### Bug Description

I setup my bed size to 400x400x400 but Marlin still handle it as 200x200x200

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5098307/Configs.zip).

### Steps to Reproduce

1. Download Marlin 2.0.6
2. Set X_BED_SIZE 400, Y_BED_SIZE 400, Z_MAX_POS_400
3. SEND G0 X200

**Expected behavior:** [What you expect to happen]
    - X axis goes to position 200, the middle of the bed   

**Actual behavior:** [What actually happens]
    - X axis goes to position 200, the end on the bed
#### Additional Information

GMagician, 2020-08-19:
m502 and m500?

GMagician, 2020-08-19:
This seems a bad DEFAULT_AXIS_STEPS_PER_UNIT configuration. 
If you changed from previous cnf do a M502 & M500 and check again

closed 2020-08-24 by ellensp