MarlinFirmware/Marlin #19083: LPC1768 Fails to set MAX6675_SS_PIN as output

capermike, 2020-08-20:
Working of configuring MKS-SGEN-L 1.0 board to use MAX31855 sensor for single hot end.

Configured for MAX31855, assign pin P0_02 to MAX_6675_SS 

Expected operation
P0_02 goes low when READ_MAX6675 is called.

Scoped pin with oscilloscope, stays high. 

If a M42 M1 P002 is issued from the console, P0_02 functions as expected, reboot and SS pin stops functioning until M42 command is issued (actually does not matter if pin mode is set as input or output, once the M42 command is sent for the pin it goes into output mode)
Issuing a M42 M0 P002 after pin is active as output, pin continues as output.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5099986/configs.zip)
[pins_MKS_SGEN_L.h.zip](https://github.com/MarlinFirmware/Marlin/files/5099987/pins_MKS_SGEN_L.h.zip)

ellensp, 2020-08-20:
Please enable PINS_DEBUGGING and send the machine a M43, Attach output here.

capermike, 2020-08-20:
Results of M43 12:48:15.586: N12 M43
12:48:15.587: PIN: 0.00 E1_DIR_PIN protected
12:48:15.588: PIN: 0.01 E1_STEP_PIN Input = 1
12:48:15.589: PIN: 0.02 (A 2) MAX6675_SS_PIN Output = 1
12:48:15.591: PIN: 0.03 (A 3) <unused/unknown> Analog in = 812 Input = 0
12:48:15.592: PIN: 0.04 TMC_SW_SCK Output = 1
12:48:15.593: PIN: 0.05 TMC_SW_MISO Input = 1
12:48:15.594: PIN: 0.06 <unused/unknown> Output = 1
12:48:15.595: PIN: 0.07 DOGLCD_SCK protected
12:48:15.596: . SCK_PIN protected
12:48:15.597: PIN: 0.08 MISO_PIN protected
12:48:15.598: PIN: 0.09 DOGLCD_MOSI protected
12:48:15.599: . MOSI_PIN protected
12:48:15.600: PIN: 0.10 E1_ENABLE_PIN Output = 1
12:48:15.601: PIN: 0.11 E0_DIR_PIN protected
12:48:15.602: PIN: 0.15 LCD_PINS_D4 Analog in = 0
12:48:15.603: PIN: 0.16 LCD_PINS_RS Output = 0
12:48:15.604: PIN: 0.17 LCD_PINS_D5 Input = 1
12:48:15.605: PIN: 0.18 LCD_PINS_ENABLE Output = 0
12:48:15.606: PIN: 0.19 Y_STEP_PIN protected
12:48:15.607: PIN: 0.20 Y_DIR_PIN protected
12:48:15.608: PIN: 0.21 Z_ENABLE_PIN protected
12:48:15.609: PIN: 0.22 Z_STEP_PIN protected
12:48:15.610: PIN: 0.23 (A23) <unused/unknown> Analog in = 0 Input = 0
12:48:15.612: PIN: 0.24 (A24) <unused/unknown> Analog in = 985 Input = 0
12:48:15.614: PIN: 0.25 (A25) <unused/unknown> Analog in = 1023 Input = 0
12:48:15.616: PIN: 0.26 (A26) <unused/unknown> Analog in = 858 Input = 0
12:48:15.617: PIN: 0.27 SD_DETECT_PIN Input = 0
12:48:15.618: PIN: 0.28 LCD_SDSS Output = 1
12:48:15.619: . SDSS Output = 1
12:48:15.620: . SS_PIN Output = 1
12:48:15.621: PIN: 0.29 <unused/unknown> Input = 1
12:48:15.622: PIN: 0.30 <unused/unknown> Input = 0
12:48:15.623: PIN: 1.00 LCD_PINS_D6 Analog in = 0
12:48:15.624: PIN: 1.01 X_CS_PIN protected
12:48:15.625: PIN: 1.04 <unused/unknown> Input = 1
12:48:15.626: PIN: 1.08 Y_CS_PIN protected
12:48:15.627: PIN: 1.09 <unused/unknown> Input = 1
12:48:15.628: PIN: 1.10 Z_CS_PIN protected
12:48:15.629: PIN: 1.14 <unused/unknown> Input = 1
12:48:15.630: PIN: 1.15 E0_CS_PIN protected
12:48:15.631: PIN: 1.16 <unused/unknown> Input = 1
12:48:15.632: PIN: 1.17 E1_CS_PIN Input = 1
12:48:15.633: PIN: 1.18 LED_PIN Output = 1
12:48:15.634: PIN: 1.19 <unused/unknown> Output = 0
12:48:15.635: PIN: 1.20 <unused/unknown> Output = 0
12:48:15.636: PIN: 1.21 <unused/unknown> Output = 0
12:48:15.637: PIN: 1.22 LCD_PINS_D7 Input = 1
12:48:15.638: PIN: 1.23 CASE_LIGHT_PIN Output = 1
12:48:15.639: . SERVO0_PIN Output = 0
12:48:15.640: PIN: 1.24 <unused/unknown> Input = 1
12:48:15.641: PIN: 1.25 Z_MIN_PIN protected
12:48:15.642: . Z_DIAG_PIN protected
12:48:15.643: . Z_STOP_PIN protected
12:48:15.644: PIN: 1.26 E1_DIAG_PIN Input = 1
12:48:15.645: PIN: 1.27 Y_MIN_PIN protected
12:48:15.646: . Y_DIAG_PIN protected
12:48:15.647: . Y_STOP_PIN protected
12:48:15.648: PIN: 1.28 E0_DIAG_PIN Input = 1
12:48:15.649: PIN: 1.29 X_MIN_PIN protected
12:48:15.650: . X_DIAG_PIN protected
12:48:15.651: . X_STOP_PIN protected
12:48:15.653: PIN: 1.30 (A62) BTN_ENC Analog in = 782
12:48:15.654: PIN: 1.31 (A63) BEEPER_PIN Output = 0
12:48:15.655: PIN: 2.00 CONTROLLER_FAN_PIN protected
12:48:15.656: . SERVO1_PIN protected
12:48:15.657: PIN: 2.01 X_ENABLE_PIN protected
12:48:15.658: PIN: 2.02 X_STEP_PIN protected
12:48:15.659: PIN: 2.03 X_DIR_PIN protected
12:48:15.660: PIN: 2.04 FAN_PIN protected
12:48:15.661: PIN: 2.05 HEATER_BED_PIN protected
12:48:15.662: PIN: 2.06 E0_AUTO_FAN_PIN protected
12:48:15.663: . FAN1_PIN protected
12:48:15.664: PIN: 2.07 HEATER_0_PIN protected
12:48:15.665: PIN: 2.08 Y_ENABLE_PIN protected
12:48:15.666: PIN: 2.09 <unused/unknown> Input = 0
12:48:15.668: PIN: 2.10 <unused/unknown> Input = 1
12:48:15.669: PIN: 2.11 Z_DIR_PIN protected
12:48:15.670: PIN: 2.12 E0_ENABLE_PIN protected
12:48:15.671: PIN: 2.13 E0_STEP_PIN protected
12:48:15.672: PIN: 3.25 BTN_EN1 Input = 1
12:48:15.673: PIN: 3.26 BTN_EN2 Input = 1
12:48:15.674: PIN: 4.28 TMC_SW_MOSI Input = 0
12:48:15.675: PIN: 4.29 <unused/unknown> Input = 1

capermike, 2020-08-21:
It seem the issue may be related to the fastio.h, the temperature.ccp calls the fastio.h SET_OUTPUT versus the M42.cpp that calls the GPio.h pinmode. I'm not seeing where the pin direction is getting set via the fastio.h

thinkyhead, 2020-09-28:
Looks like there are no occurrences of `SET_OUTPUT(MAX_SS_PIN)` in Marlin anymore. Closing as solved.

closed 2020-09-28 by thinkyhead