MarlinFirmware/Marlin #19084: PlatformIO build failed

19504643, 2020-08-20:
### Bug Description
Variadic Macros errors in "macros.h"

### My Configurations

origin macros.h without modification

### Steps to Reproduce

I use RUMBA32 board, so do some changes in Configuration.h
1. install VSCode IDE and PlatformIO IDE
2. in file "Configuration.h", change two lines:
```
L107:  #define SERIAL_PORT -1
L131:  #define MOTHERBOARD BOARD_RUMBA32_V1_1
```
3. Open the PlatformIO panel, and locate the env:rumba32 menu, click "Build", and error occurs.
those are all the variable macros errors like this:
// Macros to support option testing
#define _CAT(a,V...) a##V

**Actual behavior:** 
[image]

ellensp, 2020-08-20:
These are Intellisense display issue, nothing to do with compiling.. 
Click on the terminal tab  for any actual errors

19504643, 2020-08-20:
@ellensp 
after I clicked the 'Build', it shows: Processing rumba32 (platform: ststm32@~8.0.0; board: rumba32_f446ve; framework: arduino).
but I have to wait for several hours untill the terminal gives the error without details
[image]

ellensp, 2020-08-20:
Something wrong with your build environment, same test for me completes without issue

ellensp, 2020-08-24:
Tried on windows, compiled in about 2 minutes. no issues.
Delete your .platformio directory and let platformio reinstall and try again. 
Also ensure your machine is clean of malware. Install malwarebytes scan your machine etc, Then uninstall it again (it nags to much if you don't register)

closed 2020-08-24 by ellensp