MarlinFirmware/Marlin #19085: [BUG] FILAMENT_RUNOUT_THRESHOLD not used if FILAMENT_RUNOUT_DISTANCE_MM is defined

wolfgangmauer, 2020-08-20:
### Bug Description
I have the problem that sometimes a filament change is carried out even though the filament is loaded.
I now wanted to increase the FILAMENT_RUNOUT_THRESHOLD value.
I noticed that this is not taken into account if FILAMENT_RUNOUT_DISTANCE_MM is defined.

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5102302/Marlin.zip)

GMagician, 2020-08-20:
AFAIK that's because the threshold is done by the lenght itself. Filament must miss for the whole lenght to throw the runout.

wolfgangmauer, 2020-08-20:
The comments say otherwise

 

  // After a runout is detected, continue printing this length of filament

  // before executing the runout script. Useful for a sensor at the end of

  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.

#define FILAMENT_RUNOUT_DISTANCE_MM 60

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

GMagician, 2020-08-20:
That's not what I read on code (if correctly read it) it seems that once sensor read ok it reset lenght.

you may enable FILAMENT_RUNOUT_SENSOR_DEBUG in `runout.h` to check what is really happening with triggering distance

wolfgangmauer, 2020-08-20:
Ok, i enable debug output…

It happens very seldom that the script is executed even though filament is still present.

 

https://www.avast.com/antivirus

wolfgangmauer, 2020-08-20:
> Yeah but you may force a runout and see what happens with distance counting, then try to put filament and see if it's reset

If i force a runout, everything works as expected. 

The mistake came from somewhere else, I think

I will watch it

https://www.avast.com/antivirus

boelle, 2020-09-11:
@wolfgangmauer what did you find out?

wolfgangmauer, 2020-09-11:
Sorry for delay reply.
I changed my sensor, changed the FIL_RUNOUT_STATE to LOW.
No issues since...

closed 2020-09-11 by wolfgangmauer