MarlinFirmware/Marlin #19086: [BUG] M122 unknown command

DanJunior78, 2020-08-20:
### Bug Description
Using a MKS Gen L V2.1 Board, MKS LCD12864A and MKS TFT35.
Sending M122 via pronterface ends in an unknown command. 
Compiled Marlin 2.0.5.4 and activated...
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5102477/config.zip)
 
#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE TMC2130

#define TMC_DEBUG

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5102477/config.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** Try to get the SPI stats of my TMC2130

**Actual behavior:** Nothing, it seems the config wasn´t recognized

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-08-20:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ellensp, 2020-08-20:
Provided Config is broken... can not compile marlin 2.0.5.4 with this as is.
#define MOTHERBOARD MKS_GEN_L_V2  is wrong needs to be   #define MOTHERBOARD BOARD_MKS_GEN_L_V2
#define LCD_INFO_SCREEN_STYLE 1 is set with no 2004 lcd defined. has #define MKS_MINI_12864
Auto Bed Leveling is enabled without any probe defined....

I suspect the configuration.h your looking at is not the config file your using.

Marlin uses Configuration.h file in same directory marlin/marlin as the Makefile and  Marlin.ino file.  
Eg this directory https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin

DanJunior78, 2020-08-20:
Funny thing. I was using the configuration.h of the config folder and it was at least looking like compiling it without any fault. I´ll test a bit. Platformio isn´t my friend the last days ;).

DanJunior78, 2020-08-20:
Thanks a lot...yes, was my bad. Really nice support

closed 2020-08-20 by DanJunior78