MarlinFirmware/Marlin #19090: [BUG] E-Motor does not decelerate when switching from G1 to G0

robin-gdwl, 2020-08-20:
### Bug Description
When going from a G1-Move with extrusion to a G0-Move the E-Axis does not appear to decelerate. Instead, it stops abruptly. This is probably not an issue on most printers, however, I am using an extruder with considerable momentum and abruptly stopping it causes a spike in current. 
This happens both with a `G92 E0` (reset extruder position to 0) between the G1 and G0 move and without.

### Steps to Reproduce
Example Gcode snippet: 
```
...
G92 E0
G1 X661.07   Y451.21  E0.87
G1 X661.27   Y450.46  E1.75
G1 X661.54   Y449.73  E2.62
G1 X661.89   Y449.04  E3.50
G1 X662.31   Y448.39  E4.37
;The switch and abrupt stop happens here
G0 F7200 X695.92 Y480.26
G0 X888.33 Y480.31
G0 X888.26 Y480.24
...
```

**Expected behavior:** [What you expect to happen]
When a G0 - Move comes up the planner properly decelerates the E-Stepper bofore the G0-Move is started

**Actual behavior:** [What actually happens]
E-stepper either stops abruptly or decelerates with the Acceleration Settings usually used for G0-moves. Not entirely sure which one but I would guess it is the former.

GMagician, 2020-08-20:
Don't know what does marlin but any other gcode machine will decelerate all XYE before executing G0

robin-gdwl, 2020-08-20:
> Don't know what does marlin but any other gcode machine will decelerate all XYE before executing G0

Yes I would expect this as well. I am not sure if it is something in my configuration that leads to this behaviour

boelle, 2020-09-11:
@robin-gdwl does this also happen with latest copy of 
 [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

robin-gdwl, 2020-09-11:
@boelle It will be a week until I can test it. Unfortunately I made quite many modifications to marlin so switching to a different branch is not always easy. 
I will get back to you once I test it

boelle, 2020-09-11:
no problemo,

closed 2020-10-19 by github-actions[bot]