MarlinFirmware/Marlin #19091: [FR] M33 long file names improvment

bitboy85, 2020-08-20:
In octoprint its currently not possibe to show files on the sd card with long filenames.
An implementation is refused because there is currently no way to request all long filenames with a single command. Each file needs to be requested seperatly.
See here: https://github.com/OctoPrint/OctoPrint/issues/1600

So i suggest adding a * parameter to the M33 command so
`M33 *`
will return all files stored on the sd card with their long file names.

thinkyhead, 2020-08-26:
The purpose of `M33` is to give hosts a way to pretty up their file listing once they have the file listing in DOS8.3 format. Surely host software developers understand how to do this. If it's very difficult and host developers get stuck on how to accomplish this, I will set aside some time to add the necessary code to OctoPrint as an example for others to follow.

bitboy85, 2020-08-26:
@thinkyhead Well if you have a look at my first post and the discussion, the main developer of octoprint knows exactly how to implement this but refuses to do so because she thinks its a bad way doing so and its up to the firmware developer to provide a better way to get all filenames with one command.

firepower9966, 2020-08-27:
Also ESP3D can not use long filenames because limit of marlin. 
https://github.com/luc-github/ESP3D/issues/326

AnHardt, 2020-08-27:
> most easy solution would be to fall back to a HTML encoding. 

That will not work. For that we'd have to know the kind of encoding on the card - but since that information is not on the card - ... .

> but refuses to do so because she thinks its a bad way doing so.

We think that to. For the exact same reasons we think it's not a good idea to read all long filenames when sending the directory. It potentially blocks Marlin for a looong time.
A more intelligent strategy is needed - maybe like requesting only that long filenames currently displayed.

bitboy85, 2020-08-27:
In the octoprint discussion linked above there were even suggestions to show only the selected file with long filenames or to use a caching method...was refused too.

Currently there is support for M33. How is encoding handled here? How does it work without knowing the encoding on the card?

slowbro, 2021-07-07:
#22271 was merged, this can be closed.

closed 2021-07-07 by thisiskeithb

thisiskeithb, 2021-07-07:
Implemented as [`M20 L`](https://marlinfw.org/docs/gcode/M020.html).