MarlinFirmware/Marlin #19093: [BUG] Unable to compile when probe is enabled and BED_CENTER_AT_0_0 is active

Arakon, 2020-08-20:
### Bug Description

My Mendel90 has 0,0 at bed center, with bed going from -100 to (+)100.
When trying to enable a servo-driven probe (BFPTouch), the compilation fails with multiple "PROBING_MARGIN is too large" errors. If I disable the probe defines OR the bed center define, it compiles fine. I also tested setting it up for the BLTouch probe with the same result.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5104867/Marlin.zip)

### Steps to Reproduce

- Set up the probe in whichever way needed.
- Make sure BED_CENTER_AT_0_0 is defined.
- Attempt to compile.

**Expected behavior:** [What you expect to happen]

Code should compile fine. There's no reason why it should fail because the bed center is defined as the 0 point.

**Actual behavior:** [What actually happens]

Compilation fails.

#### Additional Information

> In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1315:32: error: static assertion failed: PROBING_MARGIN is too large.
 1315 |   static_assert(PROBING_MARGIN < _MARGIN(X), "PROBING_MARGIN is too large.");
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1316:37: error: static assertion failed: PROBING_MARGIN_BACK is too large.
 1316 |   static_assert(PROBING_MARGIN_BACK < _MARGIN(Y), "PROBING_MARGIN_BACK is too large.");
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1317:38: error: static assertion failed: PROBING_MARGIN_FRONT is too large.
 1317 |   static_assert(PROBING_MARGIN_FRONT < _MARGIN(Y), "PROBING_MARGIN_FRONT is too large.");
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1318:37: error: static assertion failed: PROBING_MARGIN_LEFT is too large.
 1318 |   static_assert(PROBING_MARGIN_LEFT < _MARGIN(X), "PROBING_MARGIN_LEFT is too large.");
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1319:38: error: static assertion failed: PROBING_MARGIN_RIGHT is too large.
 1319 |   static_assert(PROBING_MARGIN_RIGHT < _MARGIN(X), "PROBING_MARGIN_RIGHT is too large.");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1315:32: error: static assertion failed: PROBING_MARGIN is too large.
 1315 |   static_assert(PROBING_MARGIN < _MARGIN(X), "PROBING_MARGIN is too large.");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1316:37: error: static assertion failed: PROBING_MARGIN_BACK is too large.
 1316 |   static_assert(PROBING_MARGIN_BACK < _MARGIN(Y), "PROBING_MARGIN_BACK is too large.");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1317:38: error: static assertion failed: PROBING_MARGIN_FRONT is too large.
 1317 |   static_assert(PROBING_MARGIN_FRONT < _MARGIN(Y), "PROBING_MARGIN_FRONT is too large.");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1318:37: error: static assertion failed: PROBING_MARGIN_LEFT is too large.
 1318 |   static_assert(PROBING_MARGIN_LEFT < _MARGIN(X), "PROBING_MARGIN_LEFT is too large.");
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1319:38: error: static assertion failed: PROBING_MARGIN_RIGHT is too large.
 1319 |   static_assert(PROBING_MARGIN_RIGHT < _MARGIN(X), "PROBING_MARGIN_RIGHT is too large.");

ellensp, 2020-08-21:
BED_CENTER_AT_0_0 is not what you think it is. it is for scara and delta printers.
you should not enable it on your printer.

Arakon, 2020-08-21:
Ah! Okay.. then that should probably be mentioned, since as it stands, it's in no way mentioned that this option isn't for cartesian printers.. where it's not that unusual for them to be centered at 0,0, for older models.
Thanks for the info.

closed 2020-08-21 by Arakon

thinkyhead, 2020-08-28:
The authors of those configs can speak better to their choices than I can. The configs themselves should work as they are if the authors haven't come back around to patch them. Of course, we rely on user feedback to know at any time what lesser-used combinations might have gotten broken by a recent change, and while we try to keep the configs close to the manufacturer's stock configuration, we appreciate user-submitted configs with some fine-tuning and beneficial adjustments.

I'd like to put together a CI test to test the Configurations repository and catch build errors early, but with so many configs I feel like I'd be pushing some unwritten usage limit. OTOH that build report would be something cool to see…