MarlinFirmware/Marlin #19096: [BUG] Manual Leveling Mesh is not restored after G28

dominik59, 2020-08-20:
### Bug Description

I have SKR MINI E3 1.2v board and i have prepared my firmware version using tutorial: https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/
I decided to prepare it from official Marlin because i have noticed that on SKR MINI E3 1.2v official firmware after G28 command Manual Leveling Mesh is not restored too. Firstly i suspected EEPROM. After reading plenty of issues and advices i have decided to comment out lines (pins_BTT_SKR_MINI_E3_common.h):
```
#define FLASH_EEPROM_EMULATION
#define EEPROM_PAGE_SIZE     (0x800U)           // 2KB
#define EEPROM_START_ADDRESS (0x8000000UL + (STM32_FLASH_SIZE) * 1024UL - (EEPROM_PAGE_SIZE) * 2UL)
#define MARLIN_EEPROM_SIZE    EEPROM_PAGE_SIZE  // 2KB
```
because i thought that there is some problem with emulated eeprom. Unfortunatelly everything seems to work the same:
1. I power up Ender 3 Pro
2. I go to Motion -> Bed level -> <Level all points> -> Store settings
3. Print fist layer test
4. See that everything looks perfect
5. Power off machine
6. Print the same first layer test 
7. See that there is no gap between nozzle and bed (manual level has not been loaded)

Later on i have found in documentation: https://marlinfw.org/docs/gcode/M420.html that with M420 S i could load saved mesh. I  opened gcode file of "first layer test" and inserted M420 S after G28. Then i restarted machine, run print and mesh has successfully load and print looked perfect. 

P.S. I have uncommented #define RESTORE_LEVELING_AFTER_G28 line in my firmware version.
### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5105352/Marlin.zip)

### Steps to Reproduce

1. I power up Ender 3 Pro
2. I go to Motion -> Bed level -> <Level all points> -> Store settings
3. Print fist layer test
4. See that everything looks perfect
5. Power off machine
6. Print the same first layer test 
7. See that there is no gap between nozzle and bed (manual level has not been loaded)

**Expected behavior:** When RESTORE_LEVELING_AFTER_G28 option is enabled in Configuration.h, after G28 command has been executed, leveling mesh should be loaded.

**Actual behavior:** RESTORE_LEVELING_AFTER_G28 option is enabled in Configuration.h and G28 doesn't load leveling mesh

GMagician, 2020-08-20:
You should set it to true I think `#define RESTORE_LEVELING_AFTER_G28 true`

andreibobirica, 2020-08-21:
Man, don't forgot M420 S1 in the start GCODE

dominik59, 2020-08-21:
I have probably found solution. 
All you need to do to load Manual Leveling Mesh at startup is in Configuration_adv.h change
```
//#define STARTUP_COMMANDS "M17 Z"
```
to
```
#define STARTUP_COMMANDS "M420 S"
```

It will load mesh at every startup. I know that the same can be achieved using UBL, but i have tried it and i missing Bed offset setting there. I am missing it because my leveling procedure look like that:
1. Start bed leveling
2. Go one by one point and set it just to barely touch piece of paper
3. Finish bed leveling
4. Store settings to EEPROM
5. Start printing one layer test print (https://www.thingiverse.com/thing:2177790)
6. When I see that there are gaps between lines I go to Tune -> Bed Z and decrease value until there will be no gaps
7. I stop print
8. Go to Motion -> Bed Level -> Store settings to EEPROM
9. Done

As i know i could not apply the same procedure using UBL, and UBL seems to calibrate points in strange way: It doesn't calibrate one by one point which is closest to previous, it just moving nozzle randomly on bed.

To Sum it up:
For me UBL is inconvenient when you have no probe (like BLTouch)
When you decide to use MESH_BED_LEVELING option in Configuration.h you need to change:
```
//#define STARTUP_COMMANDS "M17 Z"
```
to
```
#define STARTUP_COMMANDS "M420 S"
```

boelle, 2020-08-21:
if "problem" is solved then click close below