MarlinFirmware/Marlin #19103: [BUG] Stuttering on latest snapshot on Sbase

karabas2011, 2020-08-21:
Snapshot marlin2 from 20-aug-2020
delta/mks sbase

The same as in my old issue:
https://github.com/MarlinFirmware/Marlin/issues/17055#issue-574461208

Tried: 
#define MINIMUM_STEPPER_PULSE 5  as recommended for DRV8825 -  no help

#MIN_STEPS_PER_SEGMENT  1   //6   ( https://github.com/MarlinFirmware/Marlin/issues/14532# )
Problem seems to be gone (after wiping eeprom)! but probably it is incorrect solution.

boelle, 2020-08-23:
@karabas2011 did the suggestion from @TheNitek help you?

karabas2011, 2020-08-24:
Yes, I wrote about it in first message

karabas2011, 2020-08-24:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5117290/Marlin.zip)

boelle, 2020-08-24:
> Problem seems to be gone (after wiping eeprom)! but probably it is incorrect solution.

wipe of eeprom with M502 followed by M500 is what you should always do if updating

TheNitek, 2020-08-25:
I tried M502/M500 and while it changed (and improved) the behaviour, it didn‘t fully fix it. MIN_STEPS_PER_SEGMENT change was reqired.

TheNitek, 2020-08-27:
Today it happened again: While one print was fine, I started another one (without turning the printer of or changing any settings!) and it started stuttering again. A reset fixed it this time.

thinkyhead, 2020-08-27:
It may be that the slicer is being allowed to create too many tiny segments. There's a slicer setting that will allow you to specify a minimum segment length, and increasing this value may help to eliminate the issue. If printing from a host, it may also help to print from the SD card instead.

TheNitek, 2020-08-28:
I was printing from SD card. I tried files sliced  with Cura and Slic3r. After resetting the printer the same file printed fine.

TheNitek, 2020-09-06:
I feel like #MIN_STEPS_PER_SEGMENT 1 only make things slightly better, so the initial problem is not visible. Seems like the printer is still running at nearly 100% CPU usage resulting in #19186

sjasonsmith, 2020-09-06:
Are you seeing differing performance between multiple prints of the same gcode, or with different files?

When STL files are created, most software lets you choose how detailed they should be. If the detail is too high it creates extra work for the printer without actually improving quality.

If you are exporting your own STL files, try lowering the exported quality. You will see a corresponding decrease in file size when you do this. In Fusion 360 I usually export using “Medium” quality.

If you are using STLs from someone else, look for a setting in your slicer that specifies the minimum detail to slice. Play with that to try to find a value that gives you a gcode file size reduction without perceivable quality change.

closed 2020-10-08 by karabas2011