MarlinFirmware/Marlin #19124: [BUG] TMC 2209 Sensorless Homing on Z Axis not working

HorstHartholz, 2020-08-23:
Hi,
I've upgraded my Ender 3 with a BBT SKR v1.4 turbo and TMC 2209 and everything is working fine including the sensorless homing on x and y, except the sensorless homing on the z axis even when the sensitivity is at 250.
It might be that i missed something to set, because sensorless homing normaly only set up for x and y.

Hardware: Ender 3, BBT SKR v1.4 Turbo, TMC 2209 v1.2

Thanks for helping and sorry for my english it's not my nativ language

Here's my config files:

[Config-Files.zip](https://github.com/MarlinFirmware/Marlin/files/5114826/Config-Files.zip)

XDA-Bam, 2020-08-23:
As far as I know, sensorless homing on axes with lead screws is absolutely not recommended and often doesn't work at all. The stall torque on the motor side is just too low and you risk damaging your bed. This is not a bug, it's a technical limitation.

Compare [this post](https://github.com/MarlinFirmware/Marlin/issues/8781#issuecomment-351783373) by teemuatlut (author of the TMC stepper library).

Grogyan, 2020-08-23:
Line 889 is commented in config.h which is used for using the nozzle as the probe.

I would not recommend this option as you can end up damaging your nozzle/bed

boelle, 2020-08-23:
> As far as I know, sensorless homing on axes with lead screws

i tried it.... there are simply to much torque in screws that sensorless homing will work

So this is not a bug but a user misunderstanding

HorstHartholz, 2020-08-23:
ok thanks i didn't know that, but could I set a very high micro stepping so the torque goes down and it could work?

teemuatlut, 2020-08-24:
Microstepping doesn't affect torque. It's a common misunderstanding.

Don't use sensorless Z unless you've got a delta. Even then I wouldn't recommend it because an actual sensor will be more accurate.

boelle, 2020-08-24:
if anything you should lower the voltage on the Z steppers, but i doubt it would change anything

and as @teemuatlut  says real endstops are better as they are more precise

but its not a bug so will close this one

closed 2020-08-24 by boelle