MarlinFirmware/Marlin #19127: Marlin 2.0.6 Startup Command is now "echo:start" and not "start"

nathangridley, 2020-08-24:
### Bug Description

When connecting my Simplify3D to the freshly built and uploaded Marlin 2.0.6 to my Arduino with RAMPS controller, I was receiving communications errors until I modified the "Wait for Startup Command" option from "start" to "echo:start". I noticed that many of the communications responses have "echo:" in front of them, like a debugging setting was left on.

### My Configurations

Attached are my configuration files:
[Gridley Marlin 2.0.6 Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/5115388/Gridley.Marlin.2.0.6.Configuration.Files.zip)

### Steps to Reproduce

1. Download Marlin 2.0.6.
2. Apply Hardware settings to Configuration.h.
3. Compile and Upload Marlin 2.0.6 to Arduino Mega 2560 Controller using USB interface.
4. Reboot Controller.
5. Connect to Controller with USB interface with Simplify3D.
6. Monitor communications, watch ASCII text go by, then Simplify tries other communications protocols after not detecting "start". Simplify3D disconnects since a "connection" was never established.
7. Modify the Simplify3D / Tools / Firmware Configuration setting for Wait for startup command from "start" to "echo:start".
8. Connect to Controller with USB interface with Simplify3D.
9. Connection is established with updated startup command.

The Startup Command should not need updating to establish communications, as Marlin versions 2.0.5.4, or 1.1.6.

Had to modify the Simplify3D expected startup command to use Marlin 2.0.6.

ellensp, 2020-08-24:
Echo is a host command, to ensure the line is displayed in the host.  It was wrong before. now it is correct.

closed 2020-08-24 by ellensp

nathangridley, 2020-08-24:
> Echo is a host command, to ensure the line is displayed in the host. It was wrong before. now it is correct.

Here is the connection report:

Attempting connection at \\.\COM3...
READ: echo:start
  Connected to machine!
READ: Marlin 2.0.6
READ: 
READ: echo: Last Updated: 2020-07-28 | Author: (none, default config)
Last Updated: 2020-07-28 | Author: (none, default config)
READ: echo:Compiled: Aug 23 2020
Compiled: Aug 23 2020
READ: echo: Free Memory: 4673  PlannerBufferBytes: 1200
Free Memory: 4673  PlannerBufferBytes: 1200
SENT: T0
SENT: M105
READ: echo:V81 stored settings retrieved (657 bytes; crc 4828)
V81 stored settings retrieved (657 bytes; crc 4828)

Note that just about every line with echo (but not those "processed and responded to, like the start echo") is duplicated.  This behavior is new in 2.0.6, not in 2.0.5.4, or 1.1.6.  If this is correct behavior, then I should be able to disable the "echo back" behavior in Simplify3D, right?  In the "echo:start", Simplify3D responded with "Connected to machine!", after I edited for the first time ever, the "Wait for startup command" indicator string, which I never had to do before.