MarlinFirmware/Marlin #19130: [FR] Allow separate sensitivities for homing and probing with StallGuard2

eriktjardahl, 2020-08-24:
### Description

I have a Delta machine that is currently using StallGuard2 to both home and probe the bed, however the perfect sensitivity for homing and probing is not the same as the probing would like a higher sensitivity than homing. If there was an option to define different sensitivities for homing and probing I, and everyone else using it, would be able to better use the sensorless probing feature with reduced wear and risk of damage to our printers.

### Feature Workflow

1. Add definitions for second set of StallGuard2 sensitivity used with sensorless probing in marlin
2. Separate lcd menu options for homing and probing sensitivity

Kadah, 2020-08-25:
How's your repeatability with sensorless homing? I'm still testing and debugging my delta build, but so far I'm getting a deviation of +/- 0.3 at best between homings. That is with sensitivity already at max before instant false triggers. 

I would expect sensorless probe on a delta would have the added complexity of stall forces not being a constant across the bed or even between the axises. 0,0, same across all XYZ, 0,MAX_Y would be higher force on Z and less on XY, while 0,MIN_Y would be the opposite. My guess is that the leveling mesh would end up hyperbolic.

Such deviations aren't an issue on a cartesian as XY will likely be homed to a negative position (so crashing at 0,0 wouldn't be a thing) and a half millimeter shift on the XY print area won't make a difference.

eriktjardahl, 2020-08-25:
I haven't tested it all too much yet but it does seem to not be very repeatable as of yet. However that seems to be because I'm trying to balance the sensitivity between homing and probing as well as be from the cheap hardware I have. 

I did find that homing seemed to become more reliable and easier to tweak when I ramped up the speed from the standard snail speed meant for z-screws to 50mm/s.