MarlinFirmware/Marlin #19132: [BUG] Z height not smooth between manual bed points on a large bed

wrightwells, 2020-08-24:
### Bug Description
On a large bed 2400mm x 1220mm after running a manual bed leveling the Z hight jumps 5mm at approximatly three quarters along the Y axis when moving along the Y axis either via the controller of within gcode. 
However it transitions smoothly betweem points when a simple mesh is entered with a greater height difference between points.

### My Configurations
I have a RAMPS1.4 and have uncommented the manual mesh bed leveling and set it to a grid of 4x4
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5118254/Configuration_adv.zip)

### Steps to Reproduce

1. Create manual leveling mesh or run G29 S3 etc (sample below) to create the mesh
2. Run M420 S1 to use the mesh
3. Home X Y Z
4. Raise Z by 10
5. Move Y axis between 700 - 900 (issues seen)
6 Move X axis 50 
7. Move Y axis between 900 - 700 (issue seen)
8. repeat across X axis
 
**Expected behavior:** Smooth transition between mesh points

**Actual behavior:** 
Makes a rapid 5mm+ movement at all points along X axis between Y point 700 - 900 whilst moving along the Y axis.
However using test grid A (above) the transition is smooth.

#### Additional Information
Video showing jump in Z
[video error mp4.zip](https://github.com/MarlinFirmware/Marlin/files/5118253/video.error.mp4.zip)
Actual mesh that cuases the issue
[image]

Test mesh A that raises along the Y axis
[image]
Test mesh B that raises across the X axix
(not uploaded)

boelle, 2020-08-24:
what version?

does it also happen with  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

wrightwells, 2020-09-02:
I have changed boards to a skr1.3 (to remove that variable) and (had fun getting) bugfix-2.0.x branch installed.
attached are the config files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5161290/Configuration.zip)

I have included the gcode I have used to reload the mesh.
[meshBed.zip](https://github.com/MarlinFirmware/Marlin/files/5161304/meshBed.zip)

After zeroing all the axis, I then move the Y axis and at just over 900 and I get the z stepping up.

wrightwells, 2020-09-13:
So I changed my bed size from 
x_bed_size0 1220
y_bed_size 2440
To a smaller size 
x_bed_size0 600
y_bed_size 600

I loaded the same mesh data (to pretend the machine was smaller but with the same mesh data) and the machine does not setup up on the z axis whilst moving the Y axis.
To confirm everything I recompiled at 2440, 1220 bed size and repeated the process. It still steps up the z axis.

wrightwells, 2020-10-21:
Missed the notification that this was closing however it is still an outstanding bug.

sjasonsmith, 2020-11-18:
@wrightwells, I plan to put some effort into debugging reported leveling issues over the coming weeks.
Would you be able to update to the latest bugfix-2.0.x again, and verify that you can still produce the problem?

I'm sorry to ask you to do it again, but without hardware similar to yours it would be hard for me to know if it is still happening.

Please include your configuration files and any other files you may have modified, such as your board's pins file.

wrightwells, 2020-11-18:
@sjasonsmith Will do and thanks, I'll attempt to do this at the weekend.

wrightwells, 2020-11-20:
[Marlin-bugfix-2.0.x (18112020) marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/5573773/Marlin-bugfix-2.0.x.18112020.marlin.config.zip)
Here are the config files and the leveling mesh.
I have altered other setting by doing a diff from the MPCNC original , hopefully only relevant ones but I am very much a novice.
I tested this on the SKR1.3 board with DRV8825 stepper drivers, however it was also doing this originally on a ramps1.4 board. 

So I tested it again and moving the y axis between 810-820 causes the z to lift and drop.

This time I did not zero the axis, I just rebooted and moved the y axis.

Thanks

wrightwells, 2020-12-21:
Still open, posting to keep open

ontheronix, 2021-05-30:
I'm having similar quirks with 2.0.8.1 and a bed of 200x995mm when using mesh bed leveling through G29 S1 and S2. Halfway the nozze starts using different curves to move from point to point, and the last 5 points (of 25) fail completely: 
first time I went through the mesh bed leveling the the nozzle went approx. 5cm up and stayed there. After which the movements of the z-axis were reverted. Had to restart the printer. Second and third time the nozzle crashed into the bed, also at point 21 (see below). What strikes me is that the Z-count goes negative all of a sudden (see last lines). 

> SENT: M105
READ: echo:busy: processing
busy: processing
READ: MBL G29 point 1 of 25
READ: X:10.00 Y:10.00 Z:0.70 E:0.00 Count X:800 Y:800 Z:280
READ: ok
READ: ok T:158.20 /0.00 B:115.38 /0.00 C:61.77 /0.00 @:0 B@:0
SENT: M105
READ: ok T:158.19 /0.00 B:115.32 /0.00 C:61.76 /0.00 @:0 B@:0
SENT: M105
READ: ok T:157.21 /0.00 B:114.62 /0.00 C:61.76 /0.00 @:0 B@:0
SENT: G29 S2
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: MBL G29 point 2 of 25
READ: X:53.75 Y:10.00 Z:0.70 E:0.00 Count X:4300 Y:800 Z:280
READ: ok
READ: ok T:154.79 /0.00 B:113.18 /0.00 C:61.66 /0.00 @:0 B@:0
READ: ok T:154.79 /0.00 B:113.18 /0.00 C:61.66 /0.00 @:0 B@:0
SENT: G29 S2
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: MBL G29 point 3 of 25
READ: X:97.50 Y:10.00 Z:0.70 E:0.00 Count X:7800 Y:800 Z:280
READ: ok
READ: ok T:152.85 /0.00 B:112.06 /0.00 C:61.61 /0.00 @:0 B@:0
READ: ok T:152.85 /0.00 B:112.06 /0.00 C:61.61 /0.00 @:0 B@:0
SENT: G29 S2
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: G29 S2
READ: MBL G29 point 4 of 25
READ: X:141.25 Y:10.00 Z:0.70 E:0.00 Count X:11300 Y:800 Z:280
READ: ok
READ: ok T:151.16 /0.00 B:111.05 /0.00 C:61.54 /0.00 @:0 B@:0
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
SENT: G29 S2
READ: MBL G29 point 5 of 25
READ: X:185.00 Y:10.00 Z:0.70 E:0.00 Count X:14800 Y:800 Z:280
READ: ok
READ: ok T:149.36 /0.00 B:110.07 /0.00 C:61.41 /0.00 @:0 B@:0
READ: ok T:149.36 /0.00 B:110.07 /0.00 C:61.41 /0.00 @:0 B@:0
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
READ: MBL G29 point 6 of 25
READ: X:185.00 Y:254.75 Z:0.70 E:0.00 Count X:14800 Y:20380 Z:280
READ: ok
READ: ok T:147.08 /0.00 B:108.69 /0.00 C:61.26 /0.00 @:0 B@:0
READ: ok T:147.08 /0.00 B:108.69 /0.00 C:61.26 /0.00 @:0 B@:0
SENT: G29 S2
SENT: M105
READ: echo:busy: processing
busy: processing
READ: MBL G29 point 7 of 25
READ: X:141.25 Y:254.75 Z:0.70 E:0.00 Count X:11300 Y:20380 Z:280
[... 910 lines cut ...]
SENT: G29 S2
READ: >>> do_blocking_move_to  X10.00 Y744.25 Z0.70
READ: >  X10.00 Y989.00 Z5.00
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to  X10.00 Y989.00 Z5.00
READ: >>> do_blocking_move_to  X10.00 Y989.00 Z5.00
READ: >  X10.00 Y989.00 Z0.70
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to  X10.00 Y989.00 Z0.70
READ: MBL G29 point 21 of 25
READ: X:10.00 Y:989.00 Z:0.70 E:0.00 Count X:800 Y:79120 Z:-4505
READ: ok
READ: ok T:44.73 /0.00 B:51.58 /0.00 C:47.27 /0.00 @:0 B@:0
READ: ok T:44.73 /0.00 B:51.58 /0.00 C:47.27 /0.00 @:0 B@:0
SENT: M105
READ: ok T:44.72 /0.00 B:51.63 /0.00 C:47.22 /0.00 @:0 B@:0
WARNING: Device unplugged while connected to port
Disconnected.

Could it be some kind of overflow, like Repetier describes? 
`Note: Distortion correction uses 32 bit integer math. If the z change in steps multiplied by point distance in steps is larger then 2^31 computations will overflow and result is undetermined. In real live with typical corrections this is not hit, but when testing with some big z changes on high resolution printers it gets possible to exceed that limit!`

wrightwells, 2021-07-30:
Keep open

thinkyhead, 2021-07-31:
To help with diagnosis, first be sure to download the current `bugfix-2.0.x` branch. Then enable `DEBUG_LEVELING_FEATURE` and flash the firmware.

Then, get your mesh set up so that you'll definitely get weird behavior. Move the nozzle to a position where things are okay and send `G29 Q` to get a report of the current mesh Z offset. Move to a position where things are screwy and send `G29 Q` again. Compare the differences, and then report back what you are able to discover.

closed 2021-10-11 by github-actions[bot]