MarlinFirmware/Marlin #19137: Marlin Re-ARM detecting but not reading LCD SD card

Power3DPrinting, 2020-08-24:
### Bug Description

I am upgrading some old 3D printers with Re-ARM and the reprap discount smart controller LCD will detect but not read files saved to the SD card. Basically, the machine will home and move but I cannot give it gcode because everything but the LCD is inside the printer's case.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5119632/Configuration.zip)

### Steps to Reproduce

1. I followed chris's basement guide for wiring a reprap discount smart controller with Re-ARM: https://www.youtube.com/watch?v=-dChi2CR7Vc
2. I broke out the 5V lead from the ribbon connector on EXP1 to the 5V pin on J12 of the Re-ARM board
3. I broke out pins 6,7,8 as shown in the video to the Re-ARM board as shown
4. I enabled SD support and reprap_discount_smart_controller in configuration.h
5. I changed SDCARD_CONNECTION ONBOARD to SDCARD_CONNECTION LCD in Configuration_adv.h
6. I inserted a 16GB Fat32 sd card with a file named CT.gcode and it would detect a sd card had been inserted but I could not find file CT
7. I renamed the file as .g, .txt, and .nc with the same result
8. I tried formatting exFAT and NTFS but they would not be detected and caused the text to scramble
9. I need help, three printers with Re-ARM, same problem

**Expected behavior:** [What you expect to happen]
select gcode from sd card inserted into lcd sd slot

**Actual behavior:** [What actually happens]
no gcode is shown in sd card

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
https://www.youtube.com/watch?v=-dChi2CR7Vc   wiring

https://github.com/MarlinFirmware/Marlin/issues/12436   configuration changes

https://reprap.org/forum/read.php?146,238858   SD card and file formatting

boelle, 2020-08-24:
Same with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

Power3DPrinting, 2020-08-24:
> Same with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

I haven't given a try to the bugfix version but if you think it is worth a try I will write up a new configuration.

boelle, 2020-08-24:
its the most updated branch that get fixes at least once or twice each day

you can use a tool like winmerge to copy over old settings

Power3DPrinting, 2020-08-24:
Thanks, I'll go do that!

Power3DPrinting, 2020-08-25:
> its the most updated branch that get fixes at least once or twice each day
> 
> you can use a tool like winmerge to copy over old settings

It was very similar to 2.0.6 so it made it easy to transpose the configs but I didn't get any different result from the sd card reader. Nothing shows up in the menu :(

Power3DPrinting, 2020-08-25:
I am losing track of the trouble shooting I am doing if I don't write it down.

I tried SPI_SPEED SPI_HALF_SPEED, SPI_QUARTER_SPEED, and SPI_EIGHTH_SPEED  on each of my three machines, will not read from SD card

I tried a combination of 250000 baudrate and SD_CHECK_AND_RETRY, 115200 baudrate and SD_CHECK_AND_RETRY, and 115200 baudrate and //SD_CHECK_AND_RETRY. will not read from SD card

I tried a couple more SD cards with no difference

I have repeatedly homed successfully with each new firmware so the TMC2130's in standalone with their SPI pins: SDI, SCK, CS, and SDO pins facing upwards appears to be fine

I also tried moving the EXP1 5V pin from J12 to J3 with no difference so power draw seems fine

My best guess is that I did something wrong with the LCD adapter wiring to J12 such as another part of the LCD not getting 5V or 3.3V or the pins_RAMPS_RE_ARM.h have some mismatched pins and whatever I jumped from pins farthest from the AUX3 part of the LCD adapter from left to right pins 4,5,6 to J12 pins P1_10, P1_14, P1_17 respectively were either not enough, one of them are acting interference, or another is coming into conflict with SPI connection to the LCD.

I need sleep zzzzzzzz

Power3DPrinting, 2020-08-25:
#define LCD_BACKLIGHT_PIN              P0_16  //(16) J3-7 & AUX-4 - only used on DOGLCD controllers
    #define LCD_PINS_ENABLE                P0_18  // (51) (MOSI) J3-10 & AUX-3
    #define LCD_PINS_D4                    P0_15  // (52) (SCK)  J3-9 & AUX-3
    #if ENABLED(ULTIPANEL)
      #define LCD_PINS_D5                  P1_17  // (71) ENET_MDIO
      #define LCD_PINS_D6                  P1_14  // (73) ENET_RX_ER
      #define LCD_PINS_D7                  P1_10  // (75) ENET_RXD1
    #endif
  #endif

I was checking pins_RAMPS_RE_ARM.h and rewatching chris's guide and realized LCD pins 5,6,and 7 going to P1_10, P1_14, P1_17 do not match with his video. In his video he says pins 6,7,8 go to P1_10, P1_14, P1_17.

/**
 * Smart LCD adapter
 *
 * The Smart LCD adapter can be used for the two 10 pin LCD controllers such as
 * REPRAP_DISCOUNT_SMART_CONTROLLER. It can't be used for controllers that use
 * DOGLCD_A0, DOGLCD_CS, LCD_PINS_D5, LCD_PINS_D6 or LCD_PINS_D7. A custom cable
 * is needed to pick up 5V for the EXP1 connection.
 *
 * SD card on the LCD uses the same SPI signals as the LCD. This results in garbage/lines
 * on the LCD display during accesses of the SD card. The menus/code has been arranged so
 * that the garbage/lines are erased immediately after the SD card accesses are completed.
 */

I am too tired to think

  //#define SCK_PIN                        P0_15  // (52)  system defined J3-9 & AUX-3
  //#define MISO_PIN                       P0_17  // (50)  system defined J3-10 & AUX-3
  //#define MOSI_PIN                       P0_18  // (51)  system defined J3-10 & AUX-3
  //#define SS_PIN                         P1_23  // (53)  system defined J3-5 & AUX-3 (Sometimes called SDSS)

As shown there are four lcd pin thing... omg I need to sleep

Power3DPrinting, 2020-08-25:
I tried adding this to the smart discount controller in configuration.h, the text seems a bit more stable now but did not help with SD card reading. 

//
// RepRapDiscount Smart Controller.
// https://reprap.org/wiki/RepRapDiscount_Smart_Controller
//
// Note: Usually sold with a white PCB.
//
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_2 DELAY_NS(250)

I am currently running one of my machines tethered to my computer, it is returning an SD Init Fail and now it firmly says there is "no media" with the sd card inserted when trying to access the sd card. Running from USB is a little simpler uploading firmware but I really don't have any clue what else to do but try almost randomly turning settings on and off without result.

for now I tried turning on and off SD_CHECK_AND_RETRY without success, I also turned off EEPROM_AUTO_INIT without and success getting the printer to read from the SD card.

Power3DPrinting, 2020-08-26:
I have finally gotten SPI working with the TMC2130's and I was moving onto auto fan cooling the TMC drivers because they get hot and all I had spare are some 5V fans. I found out that all of the servo pins are protected from M42 and this got me thinking maybe there is a protected pin list somewhere that is interfering with SD card reading. I don't know where this might be because there isn't anything labeled as protected in pin_RAMPS_RE_ARM.h. help???

Power3DPrinting, 2020-08-26:
#define E0_AUTO_FAN_PIN P1_18

servo 3 pin D4 is not working with my 5V 0.1A 40mm fan and Re-ARM but did work with the Arduino Mega 2560 when called with M42 P4 S254.

I tried returning E0_AUTO_FAN_PIN to -1 and issuing M42 P4 S254 and M42 P1_18 S254 but could not get the fan to spin up.

Not sure how to disable protected pins because there isn't any array listed as such. I have a good guess these protected pins are also protecting the pins to the SD card reader and is why the reader will not work.

Power3DPrinting, 2020-08-26:
[image]

https://github.com/MarlinFirmware/Marlin/issues/11927

I found this similar thread and I can narrow down the issue is EXP1 related with a pin from D4 or RS from the LCD adapter to J3 on the Re-ARM being a potential missing link for SD card communication. As seen, I have D5, D6, D7, and 5V wired correctly, I can also confirm the encoder and beep pins are functional. I am guessing the EN/Enable pin is functional so that leaves me to wondering how D4 or RS are connected and if pins_RAMPS_RE_ARM.h match them.

AnHardt, 2020-08-26:
> ...because there isn't any array listed as such...

```cpp
bool pin_is_protected(const pin_t pin) {
  static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;
  LOOP_L_N(i, COUNT(sensitive_pins)) {
    pin_t sensitive_pin;
    memcpy_P(&sensitive_pin, &sensitive_pins[i], sizeof(pin_t));
    if (pin == sensitive_pin) return true;
  }
  return false;
}
```

Then search for [SENSITIVE_PINS](https://github.com/MarlinFirmware/Marlin/search?q=SENSITIVE_PINS&unscoped_q=SENSITIVE_PINS)
and maybe for [HAL_SENSITIVE_PINS](https://github.com/MarlinFirmware/Marlin/search?q=HAL_SENSITIVE_PINS&unscoped_q=HAL_SENSITIVE_PINS)

Power3DPrinting, 2020-08-26:
Thanks for the reply!!!

What folder/page is that in? I couldn't find it in a ctrl+f search of config, config_adv, and pins_RAMPS_RE_ARM.h.

Just another update on troubleshooting, I just tried using a full graphic display (I don't have enough for all my printers) but it appears the SD card reader is working with it but now I am getting a TMC connection error and only the A4988 controlled Z-motor will move. So there is maybe hope that I don't have to run these printers tethered to my computer so LCD SD cards can work, just not sure why not with the 2004 display :|

Power3DPrinting, 2020-08-26:
I have no idea what happened, I tried rolling back to the discount smart controller and the TMC connection error persists meaning something got shorted or damaged... great... I'll have to see if I can get things working on this machine again but I will have to move to another machine

Power3DPrinting, 2020-08-27:
Great news! All three of my printers and nine TMC2130's cause TMC CONNECTION ERR which means my resoldering and wiring checks mean I have done all that I can to the physical system and Marlin is frankly janked after my edits to get the full graphic LCD working.

I am going to try and start stripping features out of my config until things working I guess starting with turning off power loss recovery and eeprom even though they were working before. This is great. Here I was thinking I needed to buy new hardware but now I just have to bill myself 100 hrs trying to diagnose this Marlin fishy business.

Power3DPrinting, 2020-09-02:
Going back through the connections of the TMC2130's, some of the double wire dupont female pins needed to be tightened with some needle nose pliers or re-crimped altogether to fit in their plastic dupont holders. I also found 4 of 10 drivers were not working so I pooled the 6 working drivers into two machines and got them working with the Full Graphic LCD. I will refresh the solder of the non-functional TMC2130's to see if I can get enough for the third printer and I modeled a new mount for the Full Graphic Display. Not a clue how to get the RepRap Discount Smart Controller working, I browsed through pins_RAMPS_RE_ARM.h and saw some of the same LCD jumped pins were defined for TMC2130 CS pins but I am really short on time to test and reupload new firmware to see if it will finally let me read from the LCD SD reader.

closed 2020-10-10 by github-actions[bot]