MarlinFirmware/Marlin #19138: [FR] (DHT11 sensor codes)

aliys97, 2020-08-24:
I read the below link and added to my code but lots of errors showed in IDE while trying to compiling.

https://github.com/MarlinFirmware/Marlin/pull/5635/files/bc9aa8e9cd13cb614ca50e101d378166d19fd2ef

aliys97, 2020-08-24:
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::DHT(unsigned char, unsigned char, unsigned char)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::DHT(unsigned char, unsigned char, unsigned char)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::begin(unsigned char)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::convertCtoF(float)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::convertFtoC(float)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::computeHeatIndex(float, float, bool)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::expectPulse(bool)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::read(bool)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::readTemperature(bool, bool)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::readHumidity(bool)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\DHT-sensor-library-master\DHT.cpp.o (symbol from plugin): In function `DHT::DHT(unsigned char, unsigned char, unsigned char)':

(.text+0x0): multiple definition of `DHT::computeHeatIndex(bool)'

sketch\DHT.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Adafruit_Sensor-master\Adafruit_Sensor.cpp.o (symbol from plugin): In function `Adafruit_Sensor::printSensorDetails()':

(.text+0x0): multiple definition of `Adafruit_Sensor::printSensorDetails()'

sketch\Adafruit_Sensor.cpp.o (symbol from plugin):(.text+0x0): first defined here

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_25'

sketch\MarlinSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board Arduino Mega or Mega 2560.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

boelle, 2020-08-24:
try with vscode+platform.io extension

boelle, 2020-08-24:
and remember to use  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

boelle, 2020-08-24:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ellensp, 2020-08-25:
That code is for marlin v1 and is not compatible with marlin v2

aliys97, 2020-08-25:
here my  Configuration.h and Configuration_adv.h files as you requested.
[code.zip](https://github.com/MarlinFirmware/Marlin/files/5121500/code.zip)

aliys97, 2020-08-25:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

here my Configuration.h and Configuration_adv.h files as you requested.
[code.zip](https://github.com/MarlinFirmware/Marlin/files/5121504/code.zip)

aliys97, 2020-08-25:
> That code is for marlin v1 and is not compatible with marlin v2

my marlin version is 1.1x.
do you have any Idia?

aliys97, 2020-08-25:
Any idea?!

per1234, 2020-08-25:
Also posted to the Arduino forum:
https://forum.arduino.cc/index.php?topic=702121

closed 2020-08-27 by ellensp