MarlinFirmware/Marlin #19141: [BUG] Filament Change from menu with MK2_MULTIPLEXER SINGLE_NOZZLE Dual Extrusion

andreibobirica, 2020-08-25:
I have tested Marlin 2.0.6 and the newest bugfix branch
### Bug Description
Marlin has filament change functionality.
If you run it at 0 degrees of the HotEnd and not in print time, you will have problems.
These problems only happen with Two Extruders. If I set up a single extruder nothing happens.
In version 2.0.5.4 there were no such problems.

Marlin configuration must be
MK2_MULTIPLEXER
SINGLE_NOZZLE
DISTINCT_E
dual extruder
single driver

First of all by selecting what you want to do from the menu (LOAD / UNLOAD / CHANGE ... and for each extruder) the list of preset temp will appear shortly after (only if the temp is at 0).

At the end of the Preset list is the Custom Temp option.
The neighboring number (which should be the current temperature) changes continuously with each action, even going negative, without a sense.

The second problem is when selecting a preset.
The temperature is set but the WAIT TEMP menu does not appear.
Same thing also happens with the CUSTOM TEMP.

If you then go to the status screen, and return to the filament change menu as before, the wait temp screen will now start (because the temperature is already set).

In this facebook post on the official Marlin page, in the comments, there are videos showing all these details.
https://www.facebook.com/groups/1049718498464482/permalink/2595373927232257/

### My Configurations
https://github.com/andreibobirica/Marlin/tree/390ab7b6e41828fd9f51308456e72702f653d095

**Expected behavior:** 
-Close to the custom temp that there is the current temperature and not random numbers
-Once the Preset for the preheat has been selected, the Wait Temp screen starts and then the filament change

**Actual behavior:** [What actually happens]
-Close to the custom temp that there is random numbers.
-Once the Preset for the preheat has been selected set temp, and nothing also happen

#### Additional Information
Examining the code I realized that from version 2.0.5.4 to 2.0.6 inside the filament menu many things have changed.
All these problems are given by the introduction of multiple PRESETs and not just 2 as it was previously.
For the first problem (random number) I imagine it reads from an array the temperature of the extruder with a wrong index (despite being SINGLE_NOZZLE).
For the second problem I tried a lot to identify the problem and I think first of all it is in the portion of code in which it does:
IF (he needs to warm up)
MAKE A
ELSE IF (does not need heating)
MAKE B

The MAKE A portion of code is never executed, under any circumstances.

andreibobirica, 2020-09-02:
Does anyone have any news of this annoying bug?

sjasonsmith, 2020-09-11:
@andreibobirica could you provide pictures of the problem? I think that will help people understand the issue better.

sjasonsmith, 2020-09-11:
I know you have links to videos, but many people (myself included) don't follow links to anything while skimming issues for ones I can help with.

andreibobirica, 2020-09-11:
> I know you have links to videos, but many people (myself included) don't follow links to anything while skimming issues for ones I can help with.

Where can I upload a video? Little or nothing can be understood from the photos. 
[image]

sjasonsmith, 2020-09-11:
> -Close to the custom temp that there is random numbers.

I assumed this would be something that could be displayed, at least.

andreibobirica, 2020-09-11:
> > -Close to the custom temp that there is random numbers.
> 
> I assumed this would be something that could be displayed, at least.

Yes, it is highlighted in this photo, in the entry "prerisc.personal"

> [image]

andreibobirica, 2020-09-11:
As for the main problem:
Normally this piece of code is executed with an Extruder:
https://github.com/MarlinFirmware/Marlin/blob/cb9a34c6f75bed9d8782658d7680d77ea428f48d/Marlin/src/lcd/menu/menu_filament.cpp#L128

With more extruders, however, this is done:
https://github.com/MarlinFirmware/Marlin/blob/cb9a34c6f75bed9d8782658d7680d77ea428f48d/Marlin/src/lcd/menu/menu_filament.cpp#L134

In the next menu, in the list of heating presets, if clicked, the temperature waiting menu does not appear:
https://github.com/MarlinFirmware/Marlin/blob/cb9a34c6f75bed9d8782658d7680d77ea428f48d/Marlin/src/lcd/menu/menu_filament.cpp#L94

The problem is where I highlighted the code for you

closed 2020-10-19 by github-actions[bot]