MarlinFirmware/Marlin #19142: Stuttering during Auto Bed Leveling process

inismor, 2020-08-25:
### Bug Description

Auto Bed Leveling works as advertised. I have it set to probe twice and it eventually completes. During the process though, usually between a fast and slow probe, the process will just halt for a few seconds, sometimes as long as 15 seconds. After "sleeping" the process picks back up and goes on as if nothing happened.

I've implemented Marlin on several printers in the past. This time it's an Ender 3 in the middle of an SKR-1.4 Turbo install. BLTouch, Bilinear, 5x matrix of probe points. This is the first time I've ever seen this kind of behavior.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5120421/config.zip)

### Steps to Reproduce

1. Start ABL. Usually G29 in a start gcode script:

`G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S170 ;Only heat extruder to bare minimum before ABL
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S170 ; wait for extruder temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 ; home all
G29
M104 S[first_layer_temperature] ; now set extruder to final temp
M109 S[first_layer_temperature] ; wait for final temmp
G1 Z2 F240
G1 X2 Y10 F3000
G1 Z0.28 F240
G92 E0.0
G1 Y190 E15.0 F1500.0 ; intro line
G1 X2.3 F5000
G1 Y20 E30 F1200.0 ; intro line
G92 E0.0`

**Expected behavior:** Smooth probing of 25 points before printing

**Actual behavior:** Stuttering and pausing randomly during probes

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

[Stuttering ABL.m4v.zip](https://github.com/MarlinFirmware/Marlin/files/5121551/Stuttering.ABL.m4v.zip)

inismor, 2020-08-26:
I seem to have "solved" my own problem, but it wasn't my doing. Evidently, my hotend thermistor was on the outs. Today it blew up and took out my control board at the same time. New board, new thermistor, things now work as expected.

So the thermistor readings were having something to do with this issue, although I'll leave it to the coders to figure out why. It is odd that it only appeared during probing.

inismor, 2020-08-26:
I seem to have spoken too soon. I tried a number of G29s and they worked without incident. Then when I tried to ABL with the hotend and bed heated to temperature (170 and 60 respectively) the delays and stuttering came back. Not quite as bad as before, but still irritating to sit through.

ManuelMcLure, 2020-08-26:
This is normal behavior if you have both ``PROBING_HEATERS_OFF`` and ``WAIT_FOR_BED_HEATER`` enabled as you do in your configuration. Marlin will turn off the bed heater when probing and then turn on the bed heater and wait until the bed hits the set temperature again before doing the next probe.
With a BLTouch ``PROBING_HEATERS_OFF`` is not needed - it's more useful for inductive probes that can be affected by the current running through the bed as it heats.

inismor, 2020-08-26:
That was it. Funny—I thought for sure that I used my normal configuration for this printer, but looking through the config, I must have ticked `PROBING_HEATERS_OFF` as part of the BLTouch code I got from Antclabs. Lesson learned. Thanks.

closed 2020-08-26 by inismor

thinkyhead, 2020-08-27:
> With a BLTouch `PROBING_HEATERS_OFF` is not needed

Au contraire! These options do help give more reliable BLTouch readings. The electrical noise on the board generated by switching 12V / 24V circuits on and off has been noted to change the electrical characteristics of peripherals, including the BLTouch, and the heater-off and fan-off options should be used if it is not too inconvenient.

ManuelMcLure, 2020-08-27:
Huh - that has not been my experience. Then again, I do use a level shifter to ensure that the BLTouch gets a nice beefy 5V servo signal and I have a genuine Panucatt Re-ARM board so the quality is probably a bit above many boards out there.